mne-tools/mne-python #7442: ENH: Multi-line evoked topomap plot

fraimondo, 2020-03-13:
#### Describe the problem
When I analyze some ERPs, I want to plot evoked topomaps every few ms. The current maximum is 20, and they are plot in one single line, spanning all across the screen.

#### Describe your solution
It might be a good idea to plot them in several rows. Like this:

[image]

#### Additional context
I've been doing this for years by keeping a copy of the `plot_evoked_topomap` function with an extra parameter `max_cols`. I'm now tired of updating this function every time that MNE updates, so if you think its a good idea, I can do the PR.

Here's the proposed modification: https://github.com/fraimondo/mne-python/commit/d57e486151976620ffb82609f00affde4b39a4af

agramfort, 2020-03-13:
I would then mimic the API names of plt.subplots ie nrows, ncols
making nrows=1 the default and letting ncols default to 'auto' being
np.ceil(n_times // nrows) or something like that

fraimondo, 2020-03-13:
So I modified it an added `nrows` and `ncols` with the posibility of using `'auto'` in at least one of them: https://github.com/fraimondo/mne-python/commit/38f4b1cc20963b1593210d6f188e2086caaae49c

The only doubt I still have, is that I had to remove the limit of 20 times in `mne.viz.utils._process_times`.

Is it a desired feature? Shall I proceed with the PR? Maybe modify the test `test_plot_topomap_basic`?

agramfort, 2020-03-13:
Please do it with a PR so can comment on the diff

fraimondo, 2020-03-17:
done!

closed 2020-03-17 by fraimondo