mne-tools/mne-python #7447: Saving the plots when using raw.plot() and raw.plot_psd()

datalw, 2020-03-13:
I want to read a lot of EEG data and plot them in one loop. Instead of showing the plots I want to save them directly for later analysis. With savefig from matplotlib I couldn't make it work. 
Btw. I used the parameter title in raw.plot() to give a title, but it somehow does not show up.

drammock, 2020-03-13:
for `raw.plot()` the title parameter changes the *window title*, it doesn't add any text to the figure.  You can add a traditional figure title via matplotlib:

```python
fig = raw.plot()
fig.subplots_adjust(top=0.9)
fig.suptitle('some text', size='xx-large', weight='bold')
```
there's an example here: https://mne.tools/dev/auto_tutorials/preprocessing/plot_30_filtering_resampling.html#slow-drifts

as far as saving directly to disk, it should work:

```python
fig = raw.plot(show=False)
fig.savefig('some_filename.png')
```

Note that if you're in *ipython* instead of regular python interpreter, `show=False` doesn't work properly, but saving the figure does still work.

If you still can't get it to work, please re-open this issue and provide a [reproducible example](https://stackoverflow.com/help/mcve) using the MNE-Python sample data.

closed 2020-03-13 by drammock

datalw, 2020-03-16:
Thanks a lot! It solved my problems