mne-tools/mne-python #7449: BUG: Mac install problem

bloyl, 2020-03-14:
#### Describe the bug
On mac os, using a clean anaconda environment and a clean install (both stable and master) coreg doesn't respond to editing fiducials. It looks good though :)

I assume this is a version issue of either VTK or Mayavi but I"m not sure how to roll back to test. I tried using the environment from `https://raw.githubusercontent.com/mne-tools/mne-python/maint/0.19/environment.yml` but that didn't help. 

#### Steps to reproduce

```
curl --remote-name https://raw.githubusercontent.com/mne-tools/mne-python/master/environment.yml
conda env create -n mne-test environment.yml 
conda activate mne-test
pip install "PyQt5>=5.10"
mne coreg -d ~/mne_data/MNE-sample-data/subjects -s sample -f ~/mne_data/MNE-sample-data/MEG/sample/sample_audvis_raw.fif
```

click on edit and try to move the lpa, rpa or nasion.

#### Expected results
Fid points should be editable

#### Actual results
mouse clicks are caught and rotate the scene etc but points don't move

#### Additional information

```
(mne-test) a483e75df2f6:~ bloyl$ python -c "import mne; mne.sys_info()"
Platform:      Darwin-18.7.0-x86_64-i386-64bit
Python:        3.7.6 (default, Jan  8 2020, 13:42:34)  [Clang 4.0.1 (tags/RELEASE_401/final)]
Executable:    /opt/anaconda3/envs/mne-test/bin/python
CPU:           i386: 8 cores
Memory:        16.0 GB

mne:           0.19.2
numpy:         1.18.1 {blas=openblas, lapack=openblas}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=MacOSX}

sklearn:       0.22.1
numba:         0.48.0
nibabel:       3.0.2
cupy:          Not found
pandas:        1.0.1
dipy:          1.1.1
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.14.1}
pyvista:       0.23.1
vtk:           8.2.0
```

Of note I just reinstalled anaconda

```
(mne-test) a483e75df2f6:~ bloyl$ conda --version
conda 4.8.2
```

I'm using MacOS Mojave version `10.14.6 (18G3020)` if that ends up being important.

agramfort, 2020-03-14:
@bloyl I can replicate your issue with a fresh install too. WIth my old env
using vtk 8.1.1 and python 3.6 it works. Unfortunately vtk 8.1.1 does not
work with python 3.7 so it's hard to test. I just tried to install vtk
8.1.2 on my old env and it works there...

bloyl, 2020-03-21:
I haven't been able to make any progress on this.
While I was able to use info about @agramfort 's old env to create a new environment it still wasn't working. So I've been unable to find a environment to iterate forward and see what might be the root cause of the problem.

I don't mind trying to sort this out (ideally for the 0.20 release) but I don't really have a plan of attack. Does any one have any ideas/suggestions?

agramfort, 2020-03-21:
@larsoner can you replicate the pb?

>

larsoner, 2020-03-23:
I should be able to try tomorrow

larsoner, 2020-03-23:
Using:
```Console
mne coreg -s sample -d ~/mne_data/MNE-sample-data/subjects -f ~/mne_data/MNE-sample-data/MEG/sample/sample_audvis_raw.fif --trans ~/mne_data/MNE-sample-data/MEG/sample/sample_audvis_raw-trans.fif
```

On my preexisting environment everything is fine:
```
Platform:      Darwin-19.3.0-x86_64-i386-64bit
Python:        3.7.3 (default, Mar 27 2019, 16:54:48)  [Clang 4.0.1 (tags/RELEASE_401/final)]
Executable:    /Users/larsoner/anaconda3/bin/python
CPU:           i386: 4 cores
Memory:        8.0 GB

mne:           0.20.dev0
numpy:         1.18.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=Qt5Agg}

sklearn:       0.22.1
numba:         0.45.0
nibabel:       2.5.0
cupy:          Not found
pandas:        0.25.3
dipy:          1.0.0
mayavi:        4.7.2.dev0 {qt_api=pyqt5, PyQt5=5.13.0}
pyvista:       0.23.2
vtk:           8.2.0
```
If I `conda remove vtk` and `pip install vtk`, which gives VTK 8.1, it still works:
```
...
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.13.0}
vtk:           8.1.2
```
If I `pip install --upgrade pyqt5`, it breaks:
```
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.14.1}
```
Same on `pip install "pyqt5<5.14.1"`, but going back to `pip install "pyqt5<5.14"` it works.

@bloyl can you confirm that `pip install pyqt5<5.14` fixes your problem? If so we can update the install docs and also hopefully fix the upstream Mayavi picking issue.

larsoner, 2020-03-23:
BTW @bloyl is it possible you did not follow the updated `dev` install instructions? It has `!=5.14.1` in the additional OSX steps:

>  pip install "PyQt5>=5.10,!=5.14.1"

In any case I'll open a PR since I think 5.14.0 is broken, too.

bloyl, 2020-03-23:
`pyqt5-5.13.2` installed via `pip install pyqt5<5.14` Works! Thanks @larsoner

FYI I had been using `5.14.1` SInce i started out tying to install stable :)

PR #7498 (2020-03-24): MRG, FIX: Update installation instructions
Closes #7449.

Point picking breakage appears not to be our problem or even Mayavi's, but rather at the VTK level. Probably VTK 8.2 is not compatible with PyQt5 >= 5.14, so we'll need to wait for the VTK9 update to try again (and to get a full 3.8 CI running).
closed 2020-03-24 by agramfort