mne-tools/mne-python #7450: Single Nicolet file(.e) read.

MrNobodyCali, 2020-03-14:
I'm sorry but I have only a single .e file. The 'mne.io.read_raw_nicolet' function needs a '.data' and a '.head' files, but I don't really know what they are from. Can I have the only .e file as input and get the mne raw data objective?

agramfort, 2020-03-14:
I don't think we support the .e extension from nicolet. I've turned you're issue into a feature request.

do not hesitate to share some file ideally as small as possible so we can add a short test.

MrNobodyCali, 2020-03-14:
I think two tools for Matlab may be helpful.
https://github.com/fieldtrip/fieldtrip
The tool can read the header and data in a .e file.
https://github.com/ieeg-portal/Nicolet-Reader
The tool includes a .e file for test.

Really appreciate your help!

agramfort, 2020-03-14:
this reader https://github.com/ieeg-portal/Nicolet-Reader seems really
minimal
according to readme file.

I have not checked for fieldtrip

>

MrNobodyCali, 2020-03-15:
Actually I did test only by fieldtrip whose nicolet analytical files are based on Nicolet-Reader. I followed the instruction(http://www.fieldtriptoolbox.org/getting_started/nicolet/):
> hdr = ft_read_header(nicoletfile);
> data = ft_read_data(nicoletfile, 'header', hdr);

And my codes enter into the files: 'fieldtrip/fileio/private/read_nervus_header.m' and 'fieldtrip/fileio/private/read_nervus_data.m'. But Nicolet-Reader has a .e file as a sample file.

agramfort, 2020-03-15:
I fear you're going to write a reader for these files in mne or use
fieldtrip to convert the data to edf, brainvision or fieldtrip format that
mne can read.

>

MrNobodyCali, 2020-03-18:
Maybe I will convert the files to other formats. Thanks a lot!

Falach, 2021-04-19:
@MrNobodyCali 
I also got some .e files that I need to read somehow, did you manage to find a "pythonic" solution? :)

MrNobodyCali, 2021-04-29:
@Falach
Actually I used Matlab to analyze files before and I choose .edf files now. But I notice mne-python added a new file interface `mne.io.read_raw_nihon` (https://mne.tools/stable/generated/mne.io.read_raw_nihon.html). `.EEG` is the original file type from Nihon Kohden equipment and `.e` file is an output file type of this machine. Maybe you can fetch the original `.EEG` rather than `.e` to analyze. If you want to find a "Matlabic" solution, I'm glad to offer some help : )

Falach, 2021-05-03:
@MrNobodyCali 
Unfortunately, I currently don't have access to the machine so the .e files are all I have.
So if I'll try Matlab, I found this simple code to load it: https://www.fieldtriptoolbox.org/getting_started/nicolet/
Do you have maybe a script for converting it to something that I can load with mne?
Thanks!

sunnyrain0827, 2023-10-10:
> I don't think we support the .e extension from nicolet. I've turned you're issue into a feature request.
> 
> do not hesitate to share some file ideally as small as possible so we can add a short test.

Dear Alexandre,

I was wondering if there is any update to read .e files in Python? Thank you!

Best,

Yu