mne-tools/mne-python #7452: ValueError: Points have to be provided as one dimensional arrays of length 3.

andraderenew, 2020-03-15:
I am trying to analyze this tutorial dataset  with MNE-Python and I cannot get through the analysis because of the error below.

I run:
import mne
import matplotlib.pyplot as plt
import numpy as np
raw=mne.io.read_raw_bdf('/home/andraderenew/Downloads/EEG/faces_run1.bdf')
raw.load_data()
raw.set_eeg_reference()
 montage = mne.channels.read_custom_montage('/home/andraderenew/Downloads/EEG/electrode_locations_and_headshape.sfp')

>>> raw.set_montage(montage)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "</home/andraderenew/.local/lib/python3.7/site-packages/mne/externals/decorator.py:decorator-gen-30>", line 2, in set_montage
  File "/home/andraderenew/.local/lib/python3.7/site-packages/mne/utils/_logging.py", line 90, in wrapper
    return function(*args, **kwargs)
  File "/home/andraderenew/.local/lib/python3.7/site-packages/mne/channels/channels.py", line 488, in set_montage
    set_dig=set_dig, raise_if_subset=raise_if_subset)
  File "/home/andraderenew/.local/lib/python3.7/site-packages/mne/channels/montage.py", line 1545, in _set_montage
    _mnt = _get_montage_in_head(montage)
  File "/home/andraderenew/.local/lib/python3.7/site-packages/mne/channels/montage.py", line 1356, in _get_montage_in_head
    return transform_to_head(montage.copy())
  File "/home/andraderenew/.local/lib/python3.7/site-packages/mne/channels/montage.py", line 935, in transform_to_head
    native_head_t = get_ras_to_neuromag_trans(nasion, lpa, rpa)
  File "/home/andraderenew/.local/lib/python3.7/site-packages/mne/transforms.py", line 650, in get_ras_to_neuromag_trans
    raise ValueError("Points have to be provided as one dimensional "
ValueError: Points have to be provided as one dimensional arrays of length 3.

agramfort, 2020-03-15:
can you share the file electrode_locations_and_headshape.sfp ?

>

andraderenew, 2020-03-15:
[electrode_locations_and_headshape.sfp.zip](https://github.com/mne-tools/mne-python/files/4334964/electrode_locations_and_headshape.sfp.zip)
 I had to compress because github won´t allow for .sfp format

agramfort, 2020-03-15:
ok so our reader for custom montage and sfp is too much hard coded for
template files like
https://github.com/mne-tools/mne-python/blob/master/mne/channels/data/montages/GSN-HydroCel-32.sfp
for example it assumes Fid start with Fid... and they are called
('FidNz', 'FidT9', 'FidT10')
it also ignores headshape points.

I would suggest to add a read_dig_sfp function that does what you expect.
This new read_dig_sfp function should also try to reuse the _read_sfp function
by allowing to pass the names of the fiducial points.

do you want to try to do it?

hoechenberger, 2020-03-19:
I think this could easily be fixed and make it into 0.20

larsoner, 2020-03-19:
Are you on it @hoechenberger or do you need help?

hoechenberger, 2020-03-19:
I have not yet looked into this, could do tomorrow if that's okay with @agramfort?

agramfort, 2020-03-19:
yes ok but it needs first a plan based on a few files. I don't want to
overfit this single sfp example

>

andraderenew, 2020-03-19:
I am sorry no to be able to work on this because I have no time these days

larsoner, 2020-03-25:
Moving the milestone to 0.21

larsoner, 2020-08-31:
This no longer crashes on `master` but it does have the problem that `headshape` only gets a single value

larsoner, 2020-08-31:
@andraderenew is `headshape` as a name a standard SFP file thing, or something specific to your lab?

PR #8178 (2020-09-01): MRG, BUG: Fix montage plotting, deduplication, and preload check
Fix a few small bugs before release. Pooled together because the changes are all pretty small and hopefully easy to understand, so I figured I'd try to save some CI cycles. Can split back out if necessary.

Closes #8025
Closes #7859
Closes #7452
closed 2020-09-01 by larsoner