mne-tools/mne-python #7463: RFC: result of connectivity=None to clustering functions may be misleading

drammock, 2020-03-17:
Summarizing a discussion I just had offline with @larsoner:

In `mne.stats.spatio_temporal_cluster{_1samp}_test`, the input array `X` takes shape (n_observations, n_times, n_vertices). Normally, the vertices come from an STC object, and a (sparse) matrix encoding the connectivity is easily obtained from the associated source space.

However, it is allowed to pass `connectivity=None`, which will default to using `scipy.ndimage.generate_binary_structure(2, 1)` to define a 2D regular lattice connectivity. The problem with that is that the 2D lattice connectivity may or may not be appropriate, because (again) the input shape is (n_observations, n_times, n_vertices), so it's only useful if your n_vertices dimension is in fact spatially one-dimensional.  But when would a neuroscience application be dealing with a series of spatial points that actually have one-dimensional connectivity?

It seems like supporting `connectivity=None` is a case of YAGNI, and might lead to unexpected results for users who, by analogy with the usage with STCs, think "OK, I have this 2D array, I need to ravel it, but I don't need to compute connectivity because it says the default is to use regular lattice connectivity, which is what I want".  

I think `permutation_cluster{_1samp}_test` has the same problem: it [flattens out the last dimension](https://github.com/mne-tools/mne-python/blob/master/mne/stats/cluster_level.py#L837-L838) prior to eventually [running the default lattice connectivity code](https://github.com/mne-tools/mne-python/blob/master/mne/stats/cluster_level.py#L492-L493).

My recommendation is to simply remove `connectivity=None` as a valid option to those functions. @agramfort WDYT?

crossref #4859

drammock, 2020-03-17:
By the way, we could add (to the `Notes` section of those functions, or to a tutorial/example) advice about how to compute regular lattice connectivity for yourself (`sklearn.feature_extraction.image.grid_to_graph` seems to be the easiest way)

agramfort, 2020-03-17:
(n_observations, n_times, n_vertices) could be (n_observations, n_times,
n_frequencies)
where a grid would make sense. I think we had an example about this at some
point.

still think it's a YAGNI?

did you get feedback from users doing a mess without noticing?

>

drammock, 2020-03-17:
> (n_observations, n_times, n_vertices) could be (n_observations, n_times, n_frequencies) where a grid would make sense.

Good point, that seems like a valid use case.

> did you get feedback from users doing a mess without noticing?

I noticed this myself when trying to do a simulation with a 2D numpy array in place of an stc.

I'll leave this open for now and push a small docstring improvement to close it.

agramfort, 2020-03-17:
works for me

>

drammock, 2020-03-17:
In the process of updating the docs, I re-realized that there is a problem here: if I pass a series of TF images to `permutation_cluster_1samp_test`, then my input array `X` is shape (n_obs, n_times, n_freqs). If I also pass `connectivity=None`, expecting a regular lattice connectivity structure to be imposed on my TF image, that's not what I'll get, because [the time and freq dimensions get unrolled](https://github.com/mne-tools/mne-python/blob/master/mne/stats/cluster_level.py#L837-L839) *before* the lattice connectivity is computed (code path is [here](https://github.com/mne-tools/mne-python/blob/master/mne/stats/cluster_level.py#L837-L839) to [here](https://github.com/mne-tools/mne-python/blob/master/mne/stats/cluster_level.py#L429-L431) to [here](https://github.com/mne-tools/mne-python/blob/master/mne/stats/cluster_level.py#L474-L475) to [here](https://github.com/mne-tools/mne-python/blob/master/mne/stats/cluster_level.py#L492-L493)).

So I think there is a legitimate problem with how `connectivity=None` works, that can't be fixed with a better docstring.

drammock, 2020-03-17:
```python
import numpy as np
from sklearn.feature_extraction.image import grid_to_graph
from scipy.ndimage import label

foo = np.array([[[0, 0, 1], [1, 0, 0]]])
# 1 obs, 2 times, 3 verts/freqs. This should have 2 clusters

# what I think is currently happening:
# step 1: https://github.com/mne-tools/mne-python/blob/master/mne/stats/cluster_level.py#L837-L838
bar = foo.reshape(foo.shape[0], -1)
# step 2: https://github.com/mne-tools/mne-python/blob/master/mne/stats/cluster_level.py#L492-L493
labels, n_labels = label(bar)

# this yields ONE label, which connects foo[0,0,-1] to foo[1,0,0].
# that is wrong, and is a result of the reshaping.

# how it should be done (perhaps, may be a more efficient way)
connectivity = grid_to_graph(*foo.shape)  # print(connectivity.toarray())  to view it
# ...then reshape, and use connectivity as if it had been passed in by the user
```

agramfort, 2020-03-18:
then we are likely to have a bug. I found the example I wrote years ago:

https://mne.tools/dev/auto_tutorials/stats-sensor-space/plot_stats_cluster_1samp_test_time_frequency.html#sphx-glr-auto-tutorials-stats-sensor-space-plot-stats-cluster-1samp-test-time-frequency-py

it's actually using  permutation_cluster_1samp_test
<https://mne.tools/dev/generated/mne.stats.permutation_cluster_1samp_test.html#mne.stats.permutation_cluster_1samp_test>
which
is what I would recommend to use for time-frequency data. It's not spatio
temporal. So if you use spatio_temporal_xxx function the last dimension
must be time and connectivity is replicated over time axis.

larsoner, 2020-03-18:
I'm not sure how that example (which uses `connectivity=None`) could have worked if the flow is as @drammock suggests. @drammock do you want to dig into it? It seems like at the very least we need to add some unit tests to capture these different possibilities and assert equivalences where they should exist. Alex's example suggests that the `grid_to_graph` should give the same results (number of clusters and p values) as a `connectivity=None` in `permutation_cluster_1samp_test` for example.

If you are tired of digging into it I can also look into it @drammock

drammock, 2020-03-18:
@larsoner I just saw your comment.  As it turns out, I ended up digging into the unit tests today anyway, and discovered that I was mistaken.  The current implementation does what I expect it to do.  So updating the docstrings is all that *needs* to happen.

To me, my mistake indicates that the various clustering code paths are somewhat obfuscated, and a refactor for maintainability might be a good idea. Maybe when we tackle the clustering API overhaul we can sneak in a refactor at that time.

agramfort, 2020-03-18:
tell us when it's good to go

>

PR #7470 (2020-03-19): MRG, DOC: Clustering docstrings and connectivity=None
In preparation for addressing #7463, this cleans up the docstrings for the 4 public-facing clustering functions:

- `mne.stats.permutation_cluster_test`
- `mne.stats.permutation_cluster_1samp_test`
- `mne.stats.spatio_temporal_cluster_test`
- `mne.stats.spatio_temporal_cluster_1samp_test`

remaining tasks:

- [x] figure out whether to change behavior of `connectivity=None` (see #7463 for discussion) *after extensive testing: I was wrong, everything works as expected with the current implementation.* closes #7463
- [x] revise entry for `conn [... cut]
closed 2020-03-19 by larsoner