mne-tools/mne-python #7466: mne.preprocessing.find_bad_channels_maxwell()'s crosstalk_file kwarg cannot be pathlib.Path

hoechenberger, 2020-03-17:
#### Describe the bug
`mne.preprocessing.find_bad_channels_maxwell()` features a `crosstalk_file` kwarg. Unlike in many (all?) other places in MNE, this argument cannot be a `pathlib.Path` object, and is required to be a string.

#### Steps to reproduce
```python
import pathlib
import mne

sample_data_folder = pathlib.Path(mne.datasets.sample.data_path())
sample_data_raw_file = (sample_data_folder / 'MEG' / 'sample' / 
                        'sample_audvis_raw.fif')
fine_cal_file = sample_data_folder / 'SSS' / 'sss_cal_mgh.dat'
crosstalk_file = sample_data_folder / 'SSS' / 'ct_sparse_mgh.fif'

raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False)
raw.crop(tmax=60).load_data()

mne.preprocessing.find_bad_channels_maxwell(raw, cross_talk=crosstalk_file,
                                            calibration=fine_cal_file) 
```

#### Expected results
```
Scanning for bad channels in 12 intervals (5.0 sec) ...
    Bad MEG channels being reconstructed: ['MEG 2443']
    Processing 204 gradiometers and 102 magnetometers
    Low-pass filtering data at 40.0 Hz
    Static bad channels: []
[done]
[]
```

#### Actual results
```python
~/Development/mne-python/mne/preprocessing/maxwell.py in _maxwell_filter(***failed resolving arguments***)
    316     #
    317     if cross_talk is not None:
--> 318         sss_ctc = _read_ctc(cross_talk)
    319         ctc_chs = sss_ctc['proj_items_chs']
    320         meg_ch_names = [info['ch_names'][p] for p in meg_picks]

~/Development/mne-python/mne/io/proc_history.py in _read_ctc(fname)
    167     """Read cross-talk correction matrix."""
    168     if not isinstance(fname, str) or not op.isfile(fname):
--> 169         raise ValueError('fname must be a file that exists, not %s' % fname)
    170     f, tree, _ = fiff_open(fname)
    171     with f as fid:

ValueError: fname must be a file that exists, not /Users/hoechenberger/mne_data/MNE-sample-data/SSS/ct_sparse_mgh.fif
```

#### Additional information
It works as expected when converting `crosstalk_file` to a string before passing it to  `find_bad_channels_maxwell()`.

agramfort, 2020-03-17:
PR welcome with the fix

>

larsoner, 2020-03-17:
I'll roll it into my open PR

PR #7460 (2020-03-17): MRG, FIX: Offload filtering step for find_bad_channels_maxwell
Moves the filtering step out of `find_bad_channels_maxwell`. This makes our outputs differ from MaxFilter a bit more, but:

1. It is better in principle (and practice) to filter all the data, then process in chunks, rather than filter each chunk independently (more edge artifacts)
2. By moving the filtering out of the function, it allows the user to choose the filtering tradeoffs
3. Based on inspecting the results with @hoechenberger , it seems like we do at least as well on this PR as we do in `master`

This PR also make a couple  [... cut]
closed 2020-03-17 by drammock