mne-tools/mne-python #7467: Errors during opening certain EDF files

datalw, 2020-03-17:
### with `mne.io.read_raw_edf(file, preload=False)` 

I got the following error:

```
EDF file detected
Setting channel info structure...
Creating raw.info structure...

<ipython-input-15-1c792ee67613>:21: RuntimeWarning: 6 channel names are too long, have been truncated to 15 characters:
['Sensor-E:EDA (Ha', '[A] EEG (1-45 Hz', '[B] EEG (1-45 Hz', '[C] EEG (1-45 Hz', '[D] EEG (1-45 Hz', '[G] HRV-LF / HRV']
  raw.append(mne.io.read_raw_edf(file, preload=False))

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-15-1c792ee67613> in <module>()
     19 for file in file_list:
     20     file = os.path.join(path,file)
---> 21     raw.append(mne.io.read_raw_edf(file, preload=False))
     22     #raw.append(mne.io.read_raw_edf(file, preload=True, stim_channel=None))

C:\ProgramData\Anaconda3\lib\site-packages\mne\io\edf\edf.py in read_raw_edf(input_fname, montage, eog, misc, stim_channel, exclude, preload, verbose)
   1171     return RawEDF(input_fname=input_fname, montage=montage, eog=eog, misc=misc,
   1172                   stim_channel=stim_channel, exclude=exclude, preload=preload,
-> 1173                   verbose=verbose)
   1174 
   1175 

<C:\ProgramData\Anaconda3\lib\site-packages\mne\externals\decorator.py:decorator-gen-157> in __init__(self, input_fname, montage, eog, misc, stim_channel, exclude, preload, verbose)

C:\ProgramData\Anaconda3\lib\site-packages\mne\utils\_logging.py in wrapper(*args, **kwargs)
     88             with use_log_level(verbose_level):
     89                 return function(*args, **kwargs)
---> 90         return function(*args, **kwargs)
     91     return FunctionMaker.create(
     92         function, 'return decfunc(%(signature)s)',

C:\ProgramData\Anaconda3\lib\site-packages\mne\io\edf\edf.py in __init__(self, input_fname, montage, eog, misc, stim_channel, exclude, preload, verbose)
    130             # Read TAL data exploiting the header info (no regexp)
    131             tal_data = self._read_segment_file([], [], 0, 0, int(self.n_times),
--> 132                                                None, None)
    133             onset, duration, desc = _read_annotations_edf(tal_data[0])
    134 

<C:\ProgramData\Anaconda3\lib\site-packages\mne\externals\decorator.py:decorator-gen-158> in _read_segment_file(self, data, idx, fi, start, stop, cals, mult)

C:\ProgramData\Anaconda3\lib\site-packages\mne\utils\_logging.py in wrapper(*args, **kwargs)
     88             with use_log_level(verbose_level):
     89                 return function(*args, **kwargs)
---> 90         return function(*args, **kwargs)
     91     return FunctionMaker.create(
     92         function, 'return decfunc(%(signature)s)',

C:\ProgramData\Anaconda3\lib\site-packages\mne\io\edf\edf.py in _read_segment_file(self, data, idx, fi, start, stop, cals, mult)
    143         return _read_segment_file(data, idx, fi, start, stop,
    144                                   self._raw_extras[fi], self.info['chs'],
--> 145                                   self._filenames[fi])
    146 
    147 

C:\ProgramData\Anaconda3\lib\site-packages\mne\io\edf\edf.py in _read_segment_file(data, idx, fi, start, stop, raw_extras, chs, filenames)
    319                             npad=0, axis=-1)
    320                 assert ch_data.shape == (len(ch_data), buf_len)
--> 321                 data[ii, d_sidx:d_eidx] = ch_data.ravel()[r_sidx:r_eidx]
    322 
    323     # only try to read the stim channel if it's not None and it's

TypeError: list indices must be integers or slices, not tuple

```
[... 85 lines cut ...]
C:\ProgramData\Anaconda3\lib\site-packages\mne\utils\_logging.py in wrapper(*args, **kwargs)
     88             with use_log_level(verbose_level):
     89                 return function(*args, **kwargs)
---> 90         return function(*args, **kwargs)
     91     return FunctionMaker.create(
     92         function, 'return decfunc(%(signature)s)',

C:\ProgramData\Anaconda3\lib\site-packages\mne\io\edf\edf.py in _read_segment_file(self, data, idx, fi, start, stop, cals, mult)
    143         return _read_segment_file(data, idx, fi, start, stop,
    144                                   self._raw_extras[fi], self.info['chs'],
--> 145                                   self._filenames[fi])
    146 
    147 

C:\ProgramData\Anaconda3\lib\site-packages\mne\io\edf\edf.py in _read_segment_file(data, idx, fi, start, stop, raw_extras, chs, filenames)
    319                             npad=0, axis=-1)
    320                 assert ch_data.shape == (len(ch_data), buf_len)
--> 321                 data[ii, d_sidx:d_eidx] = ch_data.ravel()[r_sidx:r_eidx]
    322 
    323     # only try to read the stim channel if it's not None and it's

IndexError: index 11 is out of bounds for axis 0 with size 10
```

Can anyone help me with this?

agramfort, 2020-03-17:
can you share a problematic file?

>

cbrnr, 2020-03-17:
In addition, could you please check if the problematic file passes the validation in EDFbrowser (https://www.teuniz.net/edfbrowser/EDFbrowser%20manual.html#EDF_and_BDF_compatibility_checker)?

datalw, 2020-03-17:
> 
> 
> In addition, could you please check if the problematic file passes the validation in EDFbrowser (https://www.teuniz.net/edfbrowser/EDFbrowser%20manual.html#EDF_and_BDF_compatibility_checker)?

This is really a nice useful tool. I checked the file with this tool, unfortunately the file cannot be opened either. The problem would be the file and I'll have the file fixed first then. Thanks a lot!

datalw, 2020-03-30:
> 
> 
> In addition, could you please check if the problematic file passes the validation in EDFbrowser (https://www.teuniz.net/edfbrowser/EDFbrowser%20manual.html#EDF_and_BDF_compatibility_checker)?

The file was fixed and EDFbrowser can open this file now, but I still get the same error while opening it in MNE. 
Since I do not wish to make this file public, shall I share this privately e.g. via email?

cbrnr, 2020-03-30:
What did you do to fix the file? Sure, just send me a link to the file (since I assume it will be too large to attach directly).

datalw, 2020-03-31:
> 
> 
> What did you do to fix the file? Sure, just send me a link to the file (since I assume it will be too large to attach directly).

I did not fix the file by myself. I got another exported file from the same recording. The file is here: https://drive.google.com/drive/folders/1j32lv4g4brHuqOYnchr6cPZIDeh94S0x?usp=sharing

cbrnr, 2020-03-31:
Thanks, I'll take a look.

cbrnr, 2020-03-31:
This file works with 0.20 - can you try with that latest version?

datalw, 2020-04-01:
> 
> 
> This file works with 0.20 - can you try with that latest version?

I updated my MNE to 0.20 and could read the file. Thanks a lot.

closed 2020-04-01 by datalw