mne-tools/mne-python #7471: Consistently using `set_eeg_reference([])`

sappelhoff, 2020-03-18:
Based on this the thread on referencing in the [mailing list](https://mail.nmr.mgh.harvard.edu/pipermail/mne_analysis/2020-March/006577.html), I got the idea that the past recommendation to use `set_eeg_reference([])` to prevent MNE from adding an automatic average reference is no longer needed.

Yet throughout the codebase, there are still references to this old behavior (a list follows).

https://github.com/mne-tools/mne-python/blob/25f6e5e097d8492bfcc3abb6e77804d372f68a9d/examples/preprocessing/plot_rereference_eeg.py#L47-L50

https://github.com/mne-tools/mne-python/blob/25f6e5e097d8492bfcc3abb6e77804d372f68a9d/tutorials/evoked/plot_eeg_erp.py#L72-L75

https://github.com/mne-tools/mne-python/blob/25f6e5e097d8492bfcc3abb6e77804d372f68a9d/tutorials/evoked/plot_eeg_erp.py#L93

... and there are many more :-)

Is it worth it to go through the code and remove these wrong docs? 

cc @dominikwelke @larsoner 

PS: Through my browser (Firefox), I get an encryption warning for the MNE mailing list. Just fyi.

cbrnr, 2020-03-18:
I never understood this behavior anyway, and it would be great if this could be clarified!

agramfort, 2020-03-18:
cleanup PR is very welcome !

>

larsoner, 2020-03-18:
Yes we should get this into 0.20, @sappelhoff do you have time for it this week? If not I or maybe @drammock could look

sappelhoff, 2020-03-18:
Cool, I can do this today. Good deed for the day :)

dominikwelke, 2020-03-18:
nice, @sappelhoff ! I also planned to open a PR after finishing another project.. too slow :)

PR #7474 (2020-03-19): MRG: [DOC] MNE does no longer auto apply average ref
closes #7471 

These are the files I could catch that did still assume MNE-Python to auto apply an average ref projection. Also, the docstr of `set_eeg_reference` was still assuming that.

I wasn't sure with `tutorials/preprocessing/plot_55_setting_eeg_reference.py` --> see my inline "todo" comment there marked with `XXX`

PS: What was the PR that removed the auto average ref behavior of MNE?
closed 2020-03-19 by larsoner