mne-tools/mne-python #7472: Create canonical template channel locations

cbrnr, 2020-03-18:
Currently there are some issues with plotting montages, especially with those consisting of template locations (as opposed to realistic locations), see e.g. #7141 and #7455.

At the moment, MNE ships 24 montage files, and it is not obvious which of these contain template locations and which contain realistic locations. Moreover, among these montage files there are several files containing template locations, which is really confusing.

Ideally, there should be only one template containing all possible locations. This is something we have already discussed in #5269. Unfortunately, the discussion went into a different direction and the issue was closed without solving the original problem.

Therefore, I'd like to bring up this issue again. I suggest that we get rid of the various template location files and unify them in one central place. We could use a simple text file where we could also support multiple labels per location to accommodate different naming schemes. We could compute all locations with the tools provided by @sappelhoff, and we could either include his calculation tools or just the resulting text file in MNE.

agramfort, 2020-03-18:
disclairmer neither me or @larsoner or @massich with who I did most of the
refactoring of DigMontage
have strong experience with EEG data requiring a template. The history is
that we took files from fieldtrip
and aimed at offering the same options with MNE. Ideally we should as well
or better than fieldtrip or eeglab
with little effort. Now I don't have any strong opinion of how to achieve
this. I just have the impression that
depending of eeg system it's easier to have a file per EEG cap. my 2c

>

cbrnr, 2020-03-18:
In my experience, topomaps are often used to visualize (channel-based) results. In such cases, it is more important that the maps look pretty (symmetrical), because exact locations are usually not available. This is what template locations are mainly used for.

Most caps use some variation of the extended 10-20, 10-10 or 10-05 systems, which are all based on the same underlying subdivision of certain distances on a sphere. However, sometimes they use different names for the same locations. Therefore, I think having only one template with a large number of channels (e.g. 345 as @sappelhoff proposed) would be sufficient to create all kinds of different montages (if we allow for duplicate labels).

Right now the whole montage system is rather messy. It is nice that there are some realistic montages, but there should be only one template montage as opposed to many (partially or completely) overlapping files.

sappelhoff, 2020-03-18:
> Right now the whole montage system is rather messy. It is nice that there are some realistic montages, but there should be only one template montage as opposed to many (partially or completely) overlapping files.

I agree.

Some manufacturers (e.g., Biosemi) however do use their own system (not a standard), and thus the template positions of the 10-05 system do not apply.

cbrnr, 2020-03-18:
> Some manufacturers (e.g., Biosemi) however do use their own system (not a standard), and thus the template positions of the 10-05 system do not apply.

True, but in this case we can have more than one template locations file if they are based on different divisions of the (spherical) head. Also, BioSemi does not exclusively use its own non-standard locations - we have a BioSemi system in our lab and our caps are standard 10-20 (basically what MNE already ships with the `biosemi64` montage).

Can you create BioSemi custom locations too?

mmagnuski, 2020-03-18:
I'm actually fine with the current state of affairs with respect to channel locations (apart from #7141). As long as these separate files correspond to channel locations shipped by the producer, I'm ok with duplication/overlap/inconsistencies.

agramfort, 2020-03-18:
https://mne.tools/dev/auto_tutorials/intro/plot_40_sensor_locations.html?highlight=montage
is
our new entry point
to document how to work with sensor locations.

we have this
https://mne.tools/dev/auto_tutorials/io/plot_20_reading_eeg_data.html?highlight=montage
and
https://mne.tools/dev/auto_examples/visualization/plot_montage.html?highlight=montage

I don't want to propose to start doing MEP (MNE Enhancement Proposals)
but a more detailed plan would help me follow how to progress **starting**
from what we have now.

>

cbrnr, 2020-03-18:
I didn't know that there is a separate `Layout` class. So isn't this already a separation between realistic (`Montage`) vs. template (`Layout`) locations? I don't quite understand the distinction, because it seems you can only assign a montage to a `raw` object but not a layout.

cbrnr, 2020-03-18:
> I'm actually fine with the current state of affairs with respect to channel locations (apart from #7141). As long as these separate files correspond to channel locations shipped by the producer, I'm ok with duplication/overlap/inconsistencies.

Many/most people will be needing 10-20/10-10/10-05 locations, so I think it is confusing to have a selection of different manufacturer montages containing basically the same channels.

larsoner, 2020-03-18:
Layout is really only ever meant to be used with `plot_topo`, never shown in relationship to a head circle (because in general the geometry is incorrect / inconsistent). This has been discussed a lot [over the last 3 years](https://github.com/mne-tools/mne-python/issues/3893#issuecomment-271925975) and this is what we settled on.

agramfort, 2020-03-18:
we have decided that layout should be removed for topomaps and just
used for axes grids as used in evoked.plot_topo

I think...

cbrnr, 2020-03-18:
OK, so we should basically always use montages.

larsoner, 2020-03-18:
Yeah that's the TL;DR :)

cbrnr, 2020-03-18:
@sappelhoff how can I read [your 10/10 TSV file](https://github.com/sappelhoff/eeg_positions/blob/master/data/standard_1010.tsv)? I tried `mne.channels.read_custom_montage`, but the format (extension) is not supported. I tried renaming to `.txt`, but then the function complains about missing fiducials.

sappelhoff, 2020-03-18:
It's not "MNE-READY", I always read it with pandas and I have my own simple plotting function to display the positions: https://github.com/sappelhoff/eeg_positions/blob/8d91d63c4996a81aabc24ea14dae51e8e1833ca9/eeg_positions/utils.py#L302

pull requests welcome. The main contribution of my repository is figuring out how manufacturers produce their "template" positions and having that algorithm as a python function: https://github.com/sappelhoff/eeg_positions/blob/8d91d63c4996a81aabc24ea14dae51e8e1833ca9/eeg_positions/utils.py#L10

mmagnuski, 2020-03-18:
> Many/most people will be needing 10-20/10-10/10-05 locations, so I think it is confusing to have a selection of different manufacturer montages containing basically the same channels.

It might be confusing, I understand that. I never personally found it confusing though. Are you sure there are no location differences between these files?

agramfort, 2020-03-18:
would a hangout meeting work for tomorrow around 3pm Paris time?

cbrnr, 2020-03-19:
> would a hangout meeting work for tomorrow around 3pm Paris time?

Yes, I should be available. I have another meeting at 2pm, so I might be a bit late if we don't finish on time.

agramfort, 2020-03-19:
sent an cal invitation with hangout link

sappelhoff, 2020-03-19:
@agramfort @cbrnr 

Here is code to read my computed standard locations from the [eeg_postitions](https://github.com/sappelhoff/eeg_positions/blob/master/data/standard_1005.tsv) repo into an MNE montage (SEE "DETAILS" BELOW THE PLOT). Note: Scroll down, there is lots of data pasted here for convenience.

Results in this plot:

[image]

<Details>

```Python
import pandas as pd
import numpy as np
import mne

DATA = """
label	x	y	z
Nz	0.0000	1.0000	0.0000
Iz	0.0000	-1.0000	0.0000
Cz	0.0000	0.0000	1.0000
T9	-1.0000	0.0000	0.0000
T10	1.0000	0.0000	0.0000
NFpz	0.0000	0.9877	0.1564
Fpz	0.0000	0.9511	0.3090
AFpz	0.0000	0.8910	0.4540
AFz	0.0000	0.8090	0.5878
AFFz	0.0000	0.7071	0.7071
Fz	0.0000	0.5878	0.8090
FFCz	0.0000	0.4540	0.8910
FCz	0.0000	0.3090	0.9511
FCCz	0.0000	0.1564	0.9877
CCPz	0.0000	-0.1564	0.9877
CPz	0.0000	-0.3090	0.9511
CPPz	0.0000	-0.4540	0.8910
Pz	0.0000	-0.5878	0.8090
PPOz	0.0000	-0.7071	0.7071
POz	0.0000	-0.8090	0.5878
POOz	0.0000	-0.8910	0.4540
Oz	0.0000	-0.9511	0.3090
OIz	0.0000	-0.9877	0.1564
N2h	0.1564	0.9877	0.0000
N2	0.3090	0.9511	0.0000
AFp10	0.4540	0.8910	0.0000
AF10	0.5878	0.8090	0.0000
AFF10	0.7071	0.7071	0.0000
F10	0.8090	0.5878	0.0000
FFT10	0.8910	0.4540	0.0000
FT10	0.9511	0.3090	0.0000
FTT10	0.9877	0.1564	0.0000
TTP10	0.9877	-0.1564	0.0000
TP10	0.9511	-0.3090	0.0000
TPP10	0.8910	-0.4540	0.0000
P10	0.8090	-0.5878	0.0000
PPO10	0.7071	-0.7071	0.0000
PO10	0.5878	-0.8090	0.0000
POO10	0.4540	-0.8910	0.0000
I2	0.3090	-0.9511	0.0000
I2h	0.1564	-0.9877	0.0000
N1h	-0.1564	0.9877	0.0000
N1	-0.3090	0.9511	0.0000
AFp9	-0.4540	0.8910	0.0000
AF9	-0.5878	0.8090	0.0000
AFF9	-0.7071	0.7071	0.0000
F9	-0.8090	0.5878	0.0000
FFT9	-0.8910	0.4540	0.0000
FT9	-0.9511	0.3090	0.0000
FTT9	-0.9877	0.1564	0.0000
TTP9	-0.9877	-0.1564	0.0000
TP9	-0.9511	-0.3090	0.0000
TPP9	-0.8910	-0.4540	0.0000
[... 295 lines cut ...]
fname = 'data1005.csv'
with open(fname, 'w') as fout:
    fout.write(DATA)

df = pd.read_csv(fname, sep='\t')

# Turn data into digmontage
ch_pos = df.set_index('label').to_dict('index')
for key, val in ch_pos.items():
    ch_pos[key] = np.asarray(list(val.values()))

data = mne.utils.Bunch(
    nasion=ch_pos['Nz'],
    lpa=ch_pos['T9'],
    rpa=ch_pos['T10'],
    ch_pos=ch_pos,
    coord_frame='unknown',
    hsp=None, hpi=None,
)

montage = mne.channels.make_dig_montage(**data)

montage.plot()

```

sappelhoff, 2020-03-19:
For fun, I also plotted @robertoostenveld's [electrode positions](https://robertoostenveld.nl/electrodes/plotting_1005.txt) from his [blog post](https://robertoostenveld.nl/electrode/) using the same Python code as in https://github.com/mne-tools/mne-python/issues/7472#issuecomment-601253011.

(3 adjustments: add column names to Robert's data, and in Robert's data, `lpa=LPA` and `rpa=RPA`, instead of for my case `lpa=T9`, `rpa=T10`)

Note how both electrode position datasets line up (compare to https://github.com/mne-tools/mne-python/issues/7472#issuecomment-601253011), although in Robert's assumption the equator is going through Fpz, Oz, T7 and T8 ... and mine is one contour line below: Nz, Iz, T9, T10

Interesting to see:
- Robert does not include the electrode right on the eyes ... I just left them in my dataset, because THEORETICALLY, they are valid ... although in practice it's impossible to put electrodes there (at least in living creatures) :-)
- In the plot of Robert's data, T9 is overplotted with LPA and T10 is overplotted with RPA --> showing again, how the data line up.

For me this is interesting, because it seems to show that for MNE plotting purposes the assumption of where the equator of the sphere goes DOES NOT MATTER.

[image]

cbrnr, 2020-03-19:
Very nice @sappelhoff! Your montage seems a bit cleaner to me so I'd prefer if we used it as the new default EEG montage. I can integrate this into `make_standard_montage` if you want. I'll also try this montage with my topomaps and see how it looks like.

Open issues:

- How do we name the montage? The best name would be `standard_1005`, but this is already taken. I'd still vote for this name because it is the most intuitive one. The other standard montages should be renamed to something else.
- Alternatively, we could use the name `template_1005`, but this is less obvious.
- We should clearly promote this new montage as the standard EEG template locations and clarify what the other montages really contain.

agramfort, 2020-03-19:
+1 :

- to integrate these files in make_standard_montage
- name it template_1005

> - deprecate standard_1005 and standard_1020 saying to use template_1005
- update the doc pages accordingly

question : can we use the same file for fsaverage template locations? we do
need
a realistic_1005 file?

how does this sound?
anything else to suggest?

mmagnuski, 2020-03-20:
Hi, sorry I couldn't join your live conversation.

> For me this is interesting, because it seems to show that for MNE plotting purposes the assumption of where the equator of the sphere goes DOES NOT MATTER.

It should matter - the channels at the equator should be plotted at the head outline (as long as the sphere center and radius are correct), the channels below equator would be drawn outside of the head circle (the same as in topomaps). Could you check how it works with #7455? Or maybe post a snippet so I could quickly test?

I actually have a preferrence to have the equator where the head circumference approximately is (so I'd say at the level of `Fpz, Oz, T7 and T8 ...`). It just seems graphically logical given how head circle is drawn (which seems not to work in your case, however). Head circle -> head circumference -> equator.

robertoostenveld, 2020-03-20:
Just to add some background. In FieldTrip we mainly make use of https://github.com/fieldtrip/fieldtrip/blob/master/private/elproj.m to compute projections of 3D electrode positions to a 2D plane. There is always distortion, as in cartography. Furthermore, we tend to keep the 2D layout (for plotting) separate from the 3D electrode positions, since both have different requirements. See http://www.fieldtriptoolbox.org/template/. 

The one @sappelhoff made above is identical how I would do it in first instance. However, since you want to _improve_ things, you may want to consider taking it outside the circle. Right now LPA and T9 appear to be above the ear, whereas in reality they are (vertical halfway) just in front of the ears.  You could consider something like this http://www.fieldtriptoolbox.org/template/layout/#with-a-more-realistic-display-of-temporal-sensors-1. This is also something that EEGLAB uses, especially for the EGI geodesic sensor nets, which go very far down on the face.

sappelhoff, 2020-03-20:
@mmagnuski the snippet was under the clickable "details" under the first plot. But see here:

- [Robert data (Fpz, Oz, T7, T8 equator)](https://robertoostenveld.nl/electrodes/plotting_1005.txt)
    - note: you need to add labels to the data first row: `label x y z`
- [Stefan data (Nz, Iz, T9, T10 equator)](https://raw.githubusercontent.com/sappelhoff/eeg_positions/master/data/standard_1005.tsv)

```Python
import pandas as pd
import numpy as np
import mne

fname = 'XXX'  # set this to either Roberts or my data
df = pd.read_csv(fname, sep='\t')

# Turn data into digmontage
ch_pos = df.set_index('label').to_dict('index')
for key, val in ch_pos.items():
    ch_pos[key] = np.asarray(list(val.values()))

data = mne.utils.Bunch(
    nasion=ch_pos['Nz'],  # Fpz in Robert's data
    lpa=ch_pos['T9'],  # LPA in Robert's data
    rpa=ch_pos['T10'],  # RPA in Robert's data
    ch_pos=ch_pos,
    coord_frame='unknown',
    hsp=None, hpi=None,
)

montage = mne.channels.make_dig_montage(**data)

montage.plot()
```

> I actually have a preferrence to have the equator where the head circumference approximately is (so I'd say at the level of Fpz, Oz, T7 and T8 ...)

Okay, but it's actually common practice to measure the head circumference from the nasion to the inion :thinking:

robertoostenveld, 2020-03-20:
The head circumference is not measured over Nasion, LPA, Inion and RPA (the palpable anatomical landmarks), but slightly higher. If you were to measure over the landmarks, the measuring tape would be over the eyes and over the ears. Although I personally would not measure precisely where to put the tape to determine the head circumference, I would say that it is approximately 10% above each anatomical landmark, which means it goes over Fpz, T7, Oz and T8. I.e. in https://www.sciencedirect.com/science/article/pii/S1388245700005277#FIG1 <https://www.sciencedirect.com/science/article/pii/S1388245700005277#FIG1> it is the innermost circle. 

Have a look at this screen capture https://youtu.be/ha-tY69k2ZI <https://youtu.be/ha-tY69k2ZI>.

cbrnr, 2020-03-20:
People will go crazy if we show electrodes outside the head radius. Maybe we need to show more circles like in your figure?

cbrnr, 2020-03-20:
OK, back to my IC topomaps. Here's how they looked with `standard_1020`:

[image]

And here's how they look now with @sappelhoff's 1005 montage:

[image]

The frontmost channel is Fz, whereas the backmost channel is POz. Which of these visualizations do you prefer? Which is more accurate?

mmagnuski, 2020-03-20:
Thanks for the tips @robertoostenveld !

@cbrnr
> People will go crazy if we show electrodes outside the head radius. Maybe we need to show more circles like in your figure?

But we do show electrodes outside the head radius in topomaps now (as EEGLAB also does) and I didn't notice people going crazy in these situations. :) (plotting montage and plotting topomaps has the same underlying mechanism. As @robertoostenveld mentions - plotting outside the head circle gives a more realistic picture of where the channels are with respect to head circumference, ears etc.)

@cbrnr 
I think it might be easier to judge using all montage channels and with the sphere radius fix in place (#7455).

@sappelhoff
Thanks for the code, I'll try it out!

cbrnr, 2020-03-20:
> But we do show electrodes outside the head radius in topomaps now (as EEGLAB also does) and I didn't notice people going crazy in these situations. :)

#7141 and myself. With going crazy I meant writing issues and bug reports because topomaps suddenly look different and not what people expect.

And I don't say realistic on purpose, because a sphere projected to a circle can never be realistic, so I think it's better to try to provide something that looks reasonable and what people are used to.

> I think it might be easier to judge using all montage channels and with the sphere radius fix in place (#7455).

But if you agree that the montage by @sappelhoff is realistic, are you saying that the fix modifies the locations so they look what people expect?

IMO the previous topomap is much better than the new one.

I adapted the code a little bit so that Pandas is not required:

```python
import numpy as np
from mne.channels import make_dig_montage

fname = 'standard_1005.tsv'

labels = np.loadtxt(fname, skiprows=1, usecols=0, dtype='U6')
coords = np.loadtxt(fname, skiprows=1, usecols=(1, 2, 3))
ch_pos = {label: coord for label, coord in zip(labels, coords)}

montage = make_dig_montage(ch_pos=ch_pos,
                           nasion=ch_pos['Nz'],
                           lpa=ch_pos['T9'],
                           rpa=ch_pos['T10'])

montage.plot()
```

mmagnuski, 2020-03-20:
@cbrnr
Bear in mind that previously all channels were stretched to the whole head circle irrespective of their locations (so even if you select only frontal channels they would fill the whole head circle). And there were many issues about it here, so I wouldn't treat it as a good baseline (and definitely not something that one would expect or find intuitive). If you want to show the topomaps in a systematic (not necessarily *realistic*) manner (as was decided a few months ago and as @larsoner implemented - with a sphere with respect to which channels are plotted) then you need the head outline (currently - the sphere radius) to have some spatial meaning. And unless you want it to represent the neck level - you'll find some channels outside of the head circle in at least some montages (EGI for example). What does the head circle represent in your preferred topomaps - I don't mean <in the plots you posted> but: what would you like it to represent?

Before using mne I used EEGLAB for a few years and I always found the topomaps ("topoplots" in non-mne-esqe) there intuitive (I also never heard students or co-workers complaining about EEGLAB topomaps; maybe apart form jet colormap being the default :) ). Once you know what the head circle represents its easier to interpret the channel positions. This is what I meant by *more realistic*, maybe a better word would be more interpretable (also the position of the ears is less misleading then).
Thanks for the simplified code, I'll try it out in the evening/weekend!

larsoner, 2020-03-20:
+1 for keeping our current interpretation of the head circle as defining channels above/below some equator (and hence the possibility of having channels below this equator outside the circle)

mmagnuski, 2020-03-20:
BTW - a long term plan to maybe make everyone happy could be to extend Layout functionality and use it in topomaps / montage plots so that everyone could define what they want to be plotted where (with the current behaviour the default if `layout=None`) - when building their custom layout (and allow for other head shapes as in FT: http://www.fieldtriptoolbox.org/template/layout/#with-a-more-realistic-display-of-temporal-sensors-1). But that is a distant future possibility - and only if someone is motivated enough to work on this.

cbrnr, 2020-03-20:
I'm fine with having the circle correspond to some equator. The current behavior is definitely much better than the old one, which stretched the available channels to the whole circle. I think what's still bugging me is that interpolation is always done at least up to the circle - what is the parameter that sets the extrapolation limit? `ica.plot_components` doesn't have an `extrapolate` parameter, and `topomap_args=dict(extrapolate="local")` doesn't have any effect.

agramfort, 2020-03-20:
>
>  ica.plot_components doesn't have an extrapolate parameter, and
> topomap_args=dict(extrapolate="local") doesn't have any effect.
>

very likely to be a bug

drammock, 2020-03-20:
Would people be happy with showing 2 circles, like [the brain products cap diagram](https://github.com/mne-tools/mne-python/issues/7141#issuecomment-581849666) does and like [the figure in Robert's paper](https://www-sciencedirect-com.offcampus.lib.washington.edu/science/article/pii/S1388245700005277#FIG2) does?  Outer circle goes through LPA/RPA/NAS, inner circle through T7/T8/Oz? Perhaps making the inner one optional, like `inner_circle=False` (default), with a docstring like "Whether to show the inner circle, corresponding to the approximate location of the contour of largest head circumference"?

hoechenberger, 2020-03-20:
> Would people be happy with showing 2 circles, like [the brain products cap diagram](https://github.com/mne-tools/mne-python/issues/7141#issuecomment-581849666) does and like [the figure in Robert's [...]

Sorry I'm late to the party, but this is _exactly_ what I would wish for. I like the Brain Products / MES easyCap designs best. IMHO they're easy to understand and, when I was a student, they actually helped me "get" the 10/20, 10/10, 10/5 systems.

For reference, I have attached here a picture extracted from a manual of a slightly newer cap with 64 electrodes, where the "inner circle" is completely filled with electrodes, and I have taken the liberty to add Nz, Iz, T9 and T10 (LPA and RPA, respectively).

[image]

This is a screenshot of the same cap, but with 128 electrodes:
[image]

mmagnuski, 2020-03-21:
@drammock @hoechenberger 
Do you mean adding it to topomaps too or just the montage plots?

hoechenberger, 2020-03-21:
@mmagnuski Both (so montage and topoplots would be consistent) – unless there's scientific reason _not_ to have this correspondence..?

drammock, 2020-03-21:
> @mmagnuski Both, topo and montage plots (so they would be consistent)
agreed. Any time we show a head circle it should be an option.

> unless there's scientific reason _not_ to have this correspondence..?
As long as the docs say "**approximate** location of the contour of largest head circumference" I think we're fine, regardless of whether we're dealing with idealized or actual measured sensor positions.

larsoner, 2020-03-25:
I don't think we'll have time to get the second contour into 0.20 unless someone really cranks on it (aiming to release Friday), so I'll push the milestone and we can backport to 0.20.1 if people really want it there

cbrnr, 2020-04-21:
Could we add an option to make the inner circle the extrapolation border? That way, I'd almost have my original topomaps back.

cbrnr, 2021-01-15:
I'd like to revive this issue. #8753 reminded me that I'd really like to have just one canonical template for 10-20-based montages. If there are other incompatible layouts (such as BioSemi ABC), we should of course provide a template for those as well. But currently, we have (1) a lot of overlapping/duplicated montage files and (2) realistic as opposed to template locations in many montages (including 1020).

Could we try to agree on what to do next? There seem to be people who are OK with the current state (@mmagnuski), whereas e.g. @sappelhoff and @cbrnr would prefer to clean things up by computing locations ourselves. I don't know about what others think, so please let me know here (@larsoner, @agramfort, @drammock, @hoechenberger).

hoechenberger, 2021-01-15:
I haven't taken the time to re-read the entire thread (yet), but in any case I'd like to pitch in again my proposal to follow the easyCap viz, as mentioned in https://github.com/mne-tools/mne-python/issues/7472#issuecomment-601838313

mmagnuski, 2021-01-15:
@cbrnr I previously said that I'm ok with how things are **except** #7141. That issue is not resolved yet, but there seemed to be an agreement around @larsoner's proposition last time it was discussed:

> if we do so we should probably just get rid of standard_1020 altogether and have standard_1020_sphere and standard_1020_realistic as the new options to force people to be explicit about their choice.

Judging by that issue, me, @cbrnr and @drammock (apart from @larsoner) were +1 for such a solution. I did not notice any voices against this idea.

mmagnuski, 2021-01-15:
I'm also fine with computing the 10-20 channel positions if it allowed to specify the sphere circumference placement (I prefer it to be at the Oz level, not below).

cbrnr, 2021-01-15:
Sorry @mmagnuski, I misunderstood you. But why do we need "realistic" locations? Generic montages are used are used to schematically show topoplots. In all other cases, you'd need subject-specific locations.

larsoner, 2021-01-15:
I'm inclined to internally use @sappelhoff's code with an option to compute spherical or realistic, and use a few files just for testing that it works correctly. Then the user-facing API about `_sphere` and `_realistic` can use that code. To me it seems like the cleanest solution.

Then if people want to add more locations like in  #8753 they do it by defining where those are in standard coords (az/el presumably?) and we probably get spherical vs realistic for free.

> I'm also fine with computing the 10-20 channel positions if it allowed to specify the sphere circumference placement (I prefer it to be at the Oz level, not below).

The nice thing about the compute-on-the-fly solution is that something like this seems pretty easy to implement. No need for extra files just some slight definition/tweak in the code...

>  Generic montages are used are used to schematically show topoplots. In all other cases, you'd need subject-specific locations.

If you want to be correct you do, yeah. And if you really want to be correct, you need an MRI. But people don't always go this maximum-accuracy route. People often want to do source localization even when they don't have subject-specific locations (or MRIs). It was the motivation for making this example

https://mne.tools/dev/auto_tutorials/source-modeling/plot_eeg_no_mri.html#sphx-glr-auto-tutorials-source-modeling-plot-eeg-no-mri-py

It's something people do, and as long as they know and we make it clear to them that the localization is quite suspect, then we should probably support it. And if we're going to support it, we might as well make the electrode positions more realistic if we can.

cbrnr, 2021-01-15:
Do you think these "realistic" locations really improve source modeling? They are likely as bad as the template locations, and it doesn't matter if these or template locations are warped to a template head model. IMO we can keep both if you think they are useful, but I'd rather not expose the realistic ones in the `plot_*` functions that just visualize the locations. I think it will be confusing for users, and "realistic" sounds better than "template" so most people will probably choose the former (worse looking) for their plots.

sappelhoff, 2021-01-15:
> I'm also fine with computing the 10-20 channel positions if it allowed to specify the sphere circumference placement (I prefer it to be at the Oz level, not below).

I am working on that now, it shouldn't be too hard to change my code to allow people to select where they want the equator to lie (Fpz, T8, Oz, T7 ... or Nz, T10, Iz, T9)

larsoner, 2021-01-15:
> Do you think these "realistic" locations really improve source modeling?

If they more realistically replicate where the electrodes were on the subject then I'd expect them to do something at least somewhat beneficial. Thinking of it as a kind of "middle ground" between where they actually were in reality and where the spherical positions would put them (which should be less correct?) then it should be better, assuming there is some reasonably smooth monotonic "localization error" function going between them, which is what I'd assume. But this is really not my area... I'm mostly assuming that someone added them for a reason, and this is at least one explanation for why it might be the case.

They already exist in `master`, so assuming @sappelhoff's code supports computing locations that way then it seems reasonable to continue to support it, barring some evidence that it has no/insufficient utility for people (or has enough drawbacks to balance out the potential utility).

cbrnr, 2021-01-15:
You are right, maybe these realistic locations are the template locations warped to a standardized (MNI) head? Maybe @robertoostenveld knows the details (because these montage files seem to be taken from FieldTrip)? In any case, it seems like we should promote template locations for plotting/visualization and recommend the realistic ones only for source localization purposes (provided we can get the details on how these were created and should be used).

sappelhoff, 2021-01-15:
>  Maybe [at]robertoostenveld knows the details (because these montage files seem to be taken from FieldTrip)?

yes, he knows about that (he programmed it for fieldtrip and is also the author of the 10-05 paper, and the associated blog post)

in https://github.com/mne-tools/mne-python/issues/7472#issuecomment-601611263 he also links to this fun video for realistic 10-20 positions: https://www.youtube.com/watch?v=ha-tY69k2ZI&feature=youtu.be

as far as I remember, the code for that is here: https://github.com/fieldtrip/fieldtrip/blob/a8f1a17c1bb59e2a3ec0288efec31cd15dfcadbf/ft_electrodeplacement.m

My repo on standard coordinate computation does currently not work with realistic head models. Only with spheres :-)

cbrnr, 2021-01-15:
Do you have a TL;DR?

drammock, 2021-01-15:
I'm working on one :)

drammock, 2021-01-15:
Summarizing what I see as the key points, in hopes of coming to some kind of consensus plan:

1. Some people want nice-looking diagrams of templatic electrode position based on a spherical head. We used to provide this, but defaults have changed and it's no longer possible/easy.
2. Some people want to do source modeling in the absence of digitized electrode locations or MRI structural scans. Using template electrode locations that are based on a non-spherical head are (probably? almost surely?) better than sphere-based templates for that purpose. We currently support this, and it is the default (for both 2D diagrams and inverse modeling).
3. There is a unified code path for digitized and templatic electrode locations that we shouldn't abandon. Therefore to support both sphere- and non-sphere templates, we need to provide both sphere- and non-sphere inputs, either as text files or by computing on the fly.
4. @sappelhoff's code currently doesn't support on-the-fly computing of the non-sphere template locations.

I propose that step 1 is introducing separate text files for sphere and non-sphere templates (cf. [here](https://github.com/mne-tools/mne-python/issues/7141#issuecomment-599708579)).  Step 2 would be getting rid of many/most of the text files in favor of computing things on-the-fly with some future version of @sappelhoff's code that can also support the non-sphere template.

larsoner, 2021-01-15:
Sounds good to me, thanks for summarizing @drammock . Any volunteer for (1) (@cbrnr or @mmagnuski perhaps)? Then @sappelhoff could maybe adapt his code for (2)?

sappelhoff, 2021-01-15:
> Then @sappelhoff could maybe adapt his code for (2)?

I am not sure whether my code is easily adapted to non-spherical (even irregular?!) objects unfortunately. The center piece of my code is a simple algorithm (that I took a long time and many questions to find), which is based on finding a point on an arc at a fraction of that arc. With the assumption that the arc is given by three points on a sphere, and the fraction is measured from one point to the last ... fraction=0 is point 1, fraction=0.5 is point 2, fraction=1 is point 3.

Now doing this for something other than a sphere required a completely different approach as far as I can tell ... not sure though: https://github.com/fieldtrip/fieldtrip/blob/a8f1a17c1bb59e2a3ec0288efec31cd15dfcadbf/ft_electrodeplacement.m

Having that said, I like Dan's proposal. Even without on-the-fly computation of *realistic* template positions, we could already make the first step and supply on-the-fly computation of *spherical* template positions.

> An orthogonal issue is where the "equator" gets drawn. Personally I prefer the brainvision-style diagrams with inner and outer rings, and making one or both of them optional, as discussed here. 

+1

> I prefer this to what @sappelhoff says he's working on (allowing user to pick where the equator falls) because it can be implemented at the drawing end rather than at the computing-sensor-locations end, and (I think?) that makes it easier.

given that I already spent my evening on this, I'll implement it in https://github.com/sappelhoff/eeg_positions anyhow ... but what does make it into MNE and what doesn't is another question :-) (here I would agree with you that less user choice may be better in this case)

hoechenberger, 2021-01-15:
@sappelhoff 
> I am working on that now, it shouldn't be too hard to change my code to allow people to select where they want the equator to lie (Fpz, T8, Oz, T7 ... or Nz, T10, Iz, T9)

🚀🥳📈💪

sappelhoff, 2021-02-13:
fyi, I just released version `2.0.0` of `eeg_positions`: https://github.com/sappelhoff/eeg_positions

It's now a proper pypi package with tests, docs, api, and all the nice things :-)

[28 more replies]