mne-tools/mne-python #7479: EEG noise reduction with ASR

BCIstorm, 2020-03-19:
I'm using Python for signal processing and would like to use the Artifact Subspace Reconstruction (ASR) algorithm ([https://sccn.ucsd.edu/wiki/Artifact_Subspace_Reconstruction_(ASR)]) to detect and remove EEG noise.
Is there a Python implementation of the ASR algorithm?
Thanks

agramfort, 2020-03-19:
@BCIstorm this is not available per se in MNE but we have bits and pieces.

we have OTP https://mne.tools/stable/auto_examples/preprocessing/plot_otp.html
there is autoreject https://autoreject.github.io/
etc.

if you absolutely want this feature you'll need to give a try to implementing it. I am not closing this issue as It can be seen as a feature request

jonathanjfshaw, 2020-04-22:
There is a WIP implementation here:
https://github.com/bertrandlalo/timeflux_rasr

nbara, 2020-04-23:
@jonathanjfshaw I wasn't aware of this one. 

I also have a working (AFAICT), but poorly documented version on my computer. It does both ASR and the riemannian variant (in my experience they give very similar results). I will try to clean it up and upload it to github in the next couple weeks.

nbara, 2020-05-01:
As promised, I uploaded my code to my github page [(see here)](https://github.com/nbara/python-meegkit/blob/master/meegkit/asr.py). 

It's still a bit rough around the edges and the riemannian variant is experimental*. To be honest I haven't found ASR super helpful. It sure does look compelling visually, but at least in my case it didn't make a big difference in my decoding results. Still I'm happy if it helps anyone. 

<details>
* because I had to fiddle around with pymanopt to create a non-linear eigenspace decomposition, which is not 100% tested. 
<details>

agramfort, 2020-05-01:
@nbara can you share a self contained example using a fetched mne dataset ?

>

nbara, 2020-05-02:
Sure, do you have any idea which dataset I should use? The ASR paper recommends at least 30s of clean data to calibrate. But ideally it should also contain high amplitude artifacts to get a nice before/after effect.

agramfort, 2020-05-02:
none I can suggest like that but maybe some eegbci subject? unless it's
public hence quite clean data

>

doctorgt, 2020-11-18:
Hi nbara, 

your implementation of ASR is really interesting. However, I still do not understand the documentation you wrote for it in the MEEGKIT, even when I saw the example of its use. Could you help me a little? I want to learn how to use it like in EEGLAB, input the raw eeg I want to clean and get the raw cleaned from artifacts (deleted from the data, no substituting it).

DiGyt, 2021-05-28:
For anyone who's interested in integrating the ASR method into MNE-Python, please give us a sign by commenting/reacting to this thread.

This will help me and the devs to judge whether including and maintaining ASR in MNE is worth the effort.

doctorgt, 2021-05-28:
I really would like to have ASR included in MNE-Python. I am trying to translate from EEGLAB to Python, but since I do not know MATLAB is almost impossible to me.

nbara, 2021-05-28:
I am a bit ambivalent. As stated in #9302, I feel it's mostly an online method, for which MNE-python is not necessarily well equipped. Also, I have yet to see a proper comparison between ASR and other denoising techniques in terms of SNR or decoding performance. In my experience ASR **looks** great upon visual inspection, but I am not 100% convinced. I would be happy to be proven wrong though.

@doctorgt Just out of curiosity, do you use ASR offline? Have you made any quantitative comparison with/without ASR and found a clear benefit?

DiGyt, 2021-05-29:
> I am a bit ambivalent. As stated in #9302, I feel it's mostly an online method, for which MNE-python is not necessarily well equipped. Also, I have yet to see a proper comparison between ASR and [...]

Completely agree. I created a less systematic (but not exhaustive) notebook that compares ASR performance with other cleaning/preprocessing techniques on three EEG datasets. Looks like ASR doesn't exceedingly improve ERPs / SNR assumed from the average signal, but it seems to have some edge over other cleaning techniques when it comes to different decoding tasks...

I will share the notebook here after it's cleaned and commented...

DiGyt, 2021-06-14:
Soo, here is a link to my comparison of a few automated cleaning methods.

[https://digyt.github.io/automated_EEG_cleaning_comparison/](https://digyt.github.io/automated_EEG_cleaning_comparison/)

It's far from exhaustive and perfectly valid, as this would imply more variable data and more computing power (definitely going beyond an online Google Colab Script). Other limitations are named in the notebook.

However, as far as this allows me to draw conclusions, I would say the following about Artifact Subspace Reconstruction:
- Artifact Subspace Reconstruction seems to work surprisingly bad if you want to investigate "ERP-like" signal-to-noise ratio, *but* it improves decoding accurary for tasks which are not based on the average signal. This should be a very welcome contrast as most other automated cleaning techniques are more or less explicitly designed towards minimizing deviation from the average signal/ERP.
- The presented algorithm for Artifact Subspace Reconstruction is working on raw data and has the potential to fully interpolate data (of course this doesn't necessarily improve the signal), both factors that make it more flexible related to algorithms that require epochs or pseudo-epochs to work. Of course exclusion can be a better choice than interpolation, but an option for simply marking/excluding bad segments can easily be added.
- A secondary argument is that the riemannian variant (which can be now implemented as pyriemann seems to have no more compatibility issues with the newest scipy) is reported to perform better than the standard (/euclidean) version. Again one also could argue that this is already implemented as the standard cleaning plugin in EEGLab and used by some people.

So, in brief I would say that ASR is no "superhuman" and universally applicable cleaning algorithm, but from my point of view it definitely has its use cases.

DiGyt, 2022-10-24:
**Note for people who want to use ASR with MNE-Python:**

Integration of ASR into the MNE incubator is under discussion in #9302.

Until then, a MATLAB-like and MNE-Python-compatible version of vanilla ASR can be found here:
https://github.com/DiGyt/asrpy

ASR, rASR, and other denoising algorithms can be found in @nbara's beautiful Python-MEEGkit:
https://github.com/nbara/python-meegkit

larsoner, 2022-10-24:
Closing for #9302

closed 2022-10-24 by larsoner