mne-tools/mne-python #7482: CSP questions

zhoudapeng1234, 2020-03-20:
I don't understand some steps in the CSP calculation process, maybe someone could answer these questions:

1. Why there is no  "whitening transformation" in the fit process? It is different from the calculate steps in the references 《Optimizing Spatial Filters for Robust EEG Single-Trial Analysis》and 《Multiclass common spatial patterns and information theoretic feature extraction》.

2. Why the eigen_values need to subtract 0.5 before being sorted?
[image]

agramfort, 2020-03-20:
ping @alexandrebarachant @cbrnr

cbrnr, 2020-04-21:
@mbillingr I think you wrote some of the CSP code - can you answer these two questions?

mbillingr, 2020-04-21:
Amazing how the most "obvious and self-explanatory" code turns into "WTF did I think?" after a couple of years...

1. I can't answer the first question; it has been too long that I have read the papers. Do the papers say anything about why the whitening transform is required?

2. This is an obscure way of rearranging the sorted eigenvalues. I believe it was meant to separate the patterns associated with the two classes.

larsoner, 2021-02-19:
Looks like (2) was fixed by #8151, the whitening issue I'm not sure about

catalinamagalvan, 2021-03-18:
About (1), the calculation steps are here very similar to the ones in 《Optimizing Spatial Filters for Robust EEG Single-Trial Analysis》. The only difference I see is that in this function the generalized eigenvalue problem solved is:

`eigen_values, eigen_vectors = linalg.eigh(covs[0], covs.sum(0))`

instead of

`eigen_values, eigen_vectors = linalg.eigh(covs[0], covs[1])`

that would be strictly the same of the papers, but both are equivalent.

larsoner, 2023-10-01:
I think the questions have been answered and if not, nowadays https://mne.discourse.group/ is the place to go to discuss them (and then maybe open a PR to improve our docs)

closed 2023-10-01 by larsoner