mne-tools/mne-python #7487: RFC: get_channel_types polysemy

drammock, 2020-03-20:
I've stumbled on a not-so-nice aspect of the API when working on #7486:

1. `mne.io.pick.get_channel_types` ([source](https://github.com/mne-tools/mne-python/blob/master/mne/io/pick.py#L18-L61)) takes no input, gets all known channel types as dict keys (values are FIFF constants for `kind` and `coil_type`).
2. `mne.io.pick._get_channel_types` ([source](https://github.com/mne-tools/mne-python/blob/master/mne/io/pick.py#L1124-L1133)) takes an `Info`, picks, boolean `unique`, and boolean whether to restrict to data channels only. Returns a list or set depending on `unique`.
3. `<Raw/Epochs/Evoked>.get_channel_types` ([source](https://github.com/mne-tools/mne-python/blob/master/mne/channels/channels.py#L210-L219)) method (via the `ContainsMixin`). Returns list of channel types.

Number 1 is called once in a test, and nowhere else in the codebase. I'd like to propose renaming it from `get_channel_types` to `get_all_channel_types`, `show_all_channel_types`, or `get_channel_types_dict`. I prefer the first option. This is a public function so it would be an API change w/ deprecation cycle. If acceptable, I could roll it into #7486 or do it in a separate PR.

larsoner, 2020-03-21:
> Number 1 is called once in a test, and nowhere else in the codebase. I'd like to propose renaming it from get_channel_types to get_all_channel_types, show_all_channel_types, or get_channel_types_dict. I prefer the first option.

That was for @cbrnr in #4842 so perhaps he should comment. But I agree that `inst.get_channel_types` is probably more widely used and that the overlap is problematic, so +1 for renaming `mne.io.pick.get_channel_types` to `get_all_channel_type_constants` or something similar (a variant of names you suggested that makes it clearer it's giving a mapping between ch types and their constants, in this case via a dict)

agramfort, 2020-03-21:
+1 to rename to get_all_channel_type_constants

assuming @cbnr is fine with that

cbrnr, 2020-03-21:
I'm fine with renaming the function, but I'd prefer a slightly shorter name if possible. Also, I haven't checked the return values of `mne.io.pick.get_channel_types` and `mne.io.pick._get_channel_types`, but can/should we de-duplicate code by calling the private function inside the public one?

agramfort, 2020-03-21:
this is now what the PR from Dan does. Except using a short function name

>

drammock, 2020-03-21:
> Also, I haven't checked the return values of `mne.io.pick.get_channel_types` and `mne.io.pick._get_channel_types`, but can/should we de-duplicate code by calling the private function inside the public one?

Not possible, they do completely different things, which is why I proposed to rename one of them.

What I have done is deduplicate the *instance methods* and the private function, which were doing (roughly) the same thing.

commit 9e785061 (2020-03-22): MRG, MAINT: standardize the way we get channel types (#7486)

* MAINT: standardize the way we get channel types

* fix unused imports

* fix missing comma

* closes #7487 (rename get_channel_types -> get_channel_type_constants)

* fix tests

* allow non-integer picks

* fix docdict key

* fix test

* proper deprecation

* add docstring to new test
closed 2020-03-22 by agramfort