mne-tools/mne-python #7492: Cannot read FIF file with invalid date

hoechenberger, 2020-03-22:
#### Describe the bug
When trying to read a `.fif` file that contains an invalid date field (in my case, an invalid birthday), an exception is raised. Previously (0.19.2, if I'm not mistaken), the file would be read, and I could manually check the date to see if it was reasonable:

```python
        subject_info = raw.info['subject_info']
        birthday = subject_info['birthday']
        if (not 1900 < birthday[0] < 2100 or
                not 1 <= birthday[1] <= 12 or
                not 1 <= birthday[2] <= 31):
            del subject_info['birthday']
```
I cannot use this approach anymore as reading the file fails in the first place.

#### Steps to reproduce
Try to read a file with an invalid data (measurement date or, for that matter, birthdaty) via
`mne.io.read_raw_fif()`.

(I can provide such a file if need be)

#### Expected results
Previously, the file would be loaded, with the `birthday` entry in the `info` dict being obviously nonsensical.

#### Actual results
An exception is raised.

```python
Traceback (most recent call last):
  File "/home/parietal/rhochenb/Development/Cam-CAN/Code/convert_camcan.py", line 87, in <module>
    raw = mne.io.read_raw_fif(raw_fname)
  File "/home/parietal/rhochenb/Development/mne-python/mne/io/fiff/raw.py", line 452, in read_raw_fif
    return Raw(fname=fname, allow_maxshield=allow_maxshield,
  File "<decorator-gen-169>", line 21, in __init__
  File "/home/parietal/rhochenb/Development/mne-python/mne/io/fiff/raw.py", line 79, in __init__
    self._read_raw_file(next_fname, allow_maxshield,
  File "<decorator-gen-170>", line 21, in _read_raw_file
  File "/home/parietal/rhochenb/Development/mne-python/mne/io/fiff/raw.py", line 151, in _read_raw_file
    info, meas = read_meas_info(fid, tree, clean_bads=True)
  File "<decorator-gen-24>", line 21, in read_meas_info
  File "/home/parietal/rhochenb/Development/mne-python/mne/io/meas_info.py", line 1244, in read_meas_info
    tag = read_tag(fid, pos)
  File "/home/parietal/rhochenb/Development/mne-python/mne/io/tag.py", line 451, in read_tag
    tag.data = fun(fid, tag, shape, rlims)
  File "/home/parietal/rhochenb/Development/mne-python/mne/io/tag.py", line 371, in _read_julian
    return _julian_to_cal(int(np.frombuffer(fid.read(4), dtype=">i4")))
  File "/home/parietal/rhochenb/Development/mne-python/mne/utils/numerics.py", line 991, in _julian_to_cal
    tmp_date = _julian_to_dt(jd)
  File "/home/parietal/rhochenb/Development/mne-python/mne/utils/numerics.py", line 922, in _julian_to_dt
    dt = timedelta(days=(jd - jd_t0))
OverflowError: days=2145043059; must have magnitude <= 999999999
```

#### Additional information
Ideally, "invalid" dates would not prevent the file from being loaded; the invalid dates should probably simply be dropped from the `info` dict, and a warning should be emitted. But loading the file should not fail with an exception.

agramfort, 2020-03-22:
this is for a cam-can file. Yes as many people have these data I agree
it should not crash
but I would see with cam-can authors how did this happen? Which file
is problematic?

hoechenberger, 2020-03-22:
On the Parietal storage, it's the file located at `/storage/store/data/camcan/camcan47/cc700/meg/pipeline/release004/data/aamod_meg_get_fif_00001/CC120120/rest/rest_raw.fif'`

agramfort, 2020-03-22:
the mne-c code confirms the mess with the birthday:

mne_show_fiff --in
/storage/store/data/camcan/camcan47/cc700/meg/pipeline/release004/data/aamod_meg_get_fif_00001/CC120120/rest/rest_raw.fif
--verbose | grep birthday
 404 = birthday           3.6.5874898

vs for a correct file:

mne_show_fiff --in
/storage/store/data/camcan/camcan47/cc700/meg/pipeline/release004/data/aamod_meg_get_fif_00001/CC120137/rest/rest_raw.fif
--verbose | grep birthday
 404 = birthday           1.1.1999

closed 2020-03-23 by agramfort