mne-tools/mne-python #7493: mne.Report.parse_folder doesn't recognize '_meg.fif' files

dmalt, 2020-03-22:
#### Describe the bug
`Report.parse_folder` wouldn't recognize `*meg.fif` files, altough the documentation example
[Getting started with mne.Report](https://mne.tools/stable/auto_tutorials/misc/plot_report.html#sphx-glr-auto-tutorials-misc-plot-report-py) suggests it should.

#### Steps to reproduce
1. Create the raw data object and save it with a filename ending with `meg.fif`
2. Use `Report.parse_folder` to create report for this file
```python
from mne import create_info
from mne import Report
from mne.io import RawArray
import numpy as np

n_ch = 5
info = create_info(n_ch, sfreq=200)
data = np.random.randn(n_ch, 1000)
RawArray(data, info).save("test_meg.fif")

report = Report()
report.parse_folder('.', pattern='*meg.fif', render_bem=False)
report.save("report.html")
```

#### Expected results
This is how it looks if I save the raw object as 'test_raw.fif' instead of 'test_meg.fif'

[image]

#### Actual results
What I actually get is an empty report.
[image]

Here's the console output of the script above:
```
Creating RawArray with float64 data, n_channels=5, n_times=1000
    Range : 0 ... 999 =      0.000 ...     4.995 secs
Ready.
Writing /home/dmalt/Code/python/playground/parse_folder/test_meg.fif
Closing /home/dmalt/Code/python/playground/parse_folder/test_meg.fif [done]
Embedding : jquery.js
Embedding : jquery-ui.min.js
Embedding : bootstrap.min.js
Embedding : jquery-ui.min.css
Embedding : bootstrap.min.css
Iterating over 0 potential files (this may take some time)
Saving report to location /home/dmalt/Code/python/playground/parse_folder/report.html
Rendering : Table of Contents
```

#### Additional information
```
Platform:      Linux-5.3.0-42-generic-x86_64-with-glibc2.10
Python:        3.8.1 (default, Jan  8 2020, 22:29:32)  [GCC 7.3.0]
Executable:    /home/dmalt/.miniconda3/envs/mne_dev/bin/python
CPU:           x86_64: 8 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.20.dev0
numpy:         1.18.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=Qt5Agg}

sklearn:       Not found
numba:         Not found
nibabel:       Not found
cupy:          Not found
pandas:        Not found
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found
```

dmalt, 2020-03-22:
Looking at the source code at `mne/report.py` suggests it should be an easy fix. I can create a PR for this. I'm not sure though if this is just a documentation problem and you didn't intend to support the BIDS naming for raw files yet.

agramfort, 2020-03-22:
PR welcome

>

PR #7505 (2020-03-24): MRG, FIX: Add support for BIDS-style raw files to reports
#### Reference issue
Fixes #7493 

#### What does this implement/fix?
`mne.Report.parse_folder` used to ignore `meg.fif` raw files although in the documentation example
[Getting started with mne.Report](https://mne.tools/dev/auto_tutorials/misc/plot_report.html#sphx-glr-auto-tutorials-misc-plot-report-py) `_meg.fif` is listed within the supported formats.
This PR fixes file fetching in `report.py` so it actually loads `meg.fif`.
closed 2020-03-24 by jasmainak