mne-tools/mne-python #7502: GPU Installation Fail

Vatsal-Malaviya, 2020-03-24:
1. 'MNE_USE_CUDA' is not recognized as an internal or external command,
operable program or batch file.
[image]

2. 'mne.utils.set_config' is not recognized as an internal or external command,
operable program or batch file.
[image]

Note: 
conda install cupy; was successful
MNE-python installation successful
My notebook has Nvidea RTX 2060

drammock, 2020-03-24:
The first one didn't work because Bash died not allow spaces around the equals sign when assigning values to environment variables. The second one didn't work because you're trying to run a python command in the Bash shell.

closed 2020-03-24 by drammock

Vatsal-Malaviya, 2020-03-24:
Issue still prevails 
[image]

Vatsal-Malaviya, 2020-03-24:
> The first one didn't work because Bash does not allow spaces around the equals sign when assigning values to environment variables. The second one didn't work because you're trying to run a python command in the Bash shell.

Issue still prevails
[image]

larsoner, 2020-03-24:
Is that actually even a Bash shell? It looks like a windows command prompt based on the `>` end-line character

Vatsal-Malaviya, 2020-03-24:
It is anaconda prompt snip, I thought the given code snippet was even applicable for Windows machine

larsoner, 2020-03-24:
All of our code samples are Bash, for windows command prompt shell users -- or users of other shells generally -- you'll need to look up how to set environment variables in your given shell

Vatsal-Malaviya, 2020-03-24:
I am still a newbie, it will be very helpful if you could guide me in a simple way

Vatsal-Malaviya, 2020-03-25:
> I am still a newbie, it will be very helpful if you could guide me in a simple way

Can you direct me to the correct syntax to create a new environment variable(MNE_USE_CUDA)?
If that's what you are trying to say in your last message to create the environment variable first and then implement the above code.

agramfort, 2020-03-25:
maybe this can help http://www.dowdandassociates.com/blog/content/howto-set-an-environment-variable-in-windows-command-line-and-registry/

none of the mne dev work on windows sorry

GuillaumeFavelier, 2020-03-25:
Windows is not my main OS for dev but here is what I do to set variables on `Windows 10`:

On `Settings`:

[image]

In the search fileld: `Edit environment variables`:

[image]

Then `New`:

[image]

[image]

And finally `OK`:

[image]

Hoping it could be useful.

drammock, 2020-03-25:
@Vatsal-Malaviya what @GuillaumeFavelier suggested is one way to set environment variables in Windows.  But perhaps easier in your case is to run the second command you tried: `mne.set_config('MNE_USE_CUDA', 'true')`, but run it *in a python interpreter* instead of in the windows command shell.  In other words:

```
(mne) C:\Users\Vatsal\Desktop> python
>>> import mne
>>> mne.set_config('MNE_USE_CUDA', 'true')
```