mne-tools/mne-python #7503: pick_channels or pick doesn't work when preload=False

lkorczowski, 2020-03-24:
Hi,

I don't know if it is a bug but this behaviour is weird and has been discussed in several issues (which looked sometimes resolved), e.g. #4398, #4381, #1814, #1766

I looked in theses issues and I link to some comments below to show that the behaviour doesn't work as expected.

#### Describe the bug

```
RAW  = mne.io.read_raw_edf(filename, preload=False)  # prepare loading
picks_chan = ['1', '1-2','1-F','2','2-F']                                  # subset of EMG electrodes
RAW  = RAW.pick_channels(picks_chan).load_data().     # pick channels and load into memory
```
return 
```
RuntimeError: By default, MNE does not load data into main memory to conserve resources. adding, dropping, or reordering channels requires raw data to be loaded. Use preload=True (or string) in the constructor or raw.load_data().
```

#### Steps to reproduce
If you have any EDF, just
```
filename =                                                                        # your EDF file
RAW  = mne.io.read_raw_edf(filename, preload=False)  # prepare loading
picks_chan =                                                                     # subset of electrodes you want
RAW  = RAW.pick_channels(picks_chan).load_data().     # pick channels and load into memory
```

#### Expected results
I believe that any of the following examples should work (as well as the one mentioned earlier). As when you have several hundred of electrodes, you just don't want to preload everything before selecting electrodes.
```
picks_chan = ['1', '1-2','1-F','2','2-F']                                             # subset of EMG electrodes
RAW  = mne.io.read_raw_edf(filename, preload=False, pick=)  # preload with specific elec
```

```
RAW  = mne.io.read_raw_edf(filename, preload=False)  # prepare loading
picks_chan = ['1', '1-2','1-F','2','2-F']                                  # subset of EMG electrodes
RAW  = RAW.pick(picks=picks_chan).load_data().     # pick channels and load into memory
```

#### Actual results
```
---------------------------------------------------------------------------

RuntimeError                              Traceback (most recent call last)

<ipython-input-9-087f8efbd675> in <module>
      3 RAW  = mne.io.read_raw_edf(filename, preload=False)  # prepare loading
      4 picks_chan = ['1', '1-2','1-F','2','2-F']            # subset of EMG electrodes
----> 5 RAW  = RAW.pick_channels(picks_chan)       # pick channels and load
      6 

~/anaconda3/envs/tinnsleep-env/lib/python3.8/site-packages/mne/channels/channels.py in pick_channels(self, ch_names)
    702         .. versionadded:: 0.9.0
    703         """
--> 704         return self._pick_drop_channels(
    705             pick_channels(self.info['ch_names'], ch_names))
    706 

~/anaconda3/envs/tinnsleep-env/lib/python3.8/site-packages/mne/channels/channels.py in _pick_drop_channels(self, idx)
    810         from ..time_frequency import AverageTFR, EpochsTFR
    811 
--> 812         _check_preload(self, 'adding, dropping, or reordering channels')
    813 
    814         if getattr(self, 'picks', None) is not None:

~/anaconda3/envs/tinnsleep-env/lib/python3.8/site-packages/mne/utils/check.py in _check_preload(inst, msg)
    179         name = "epochs" if isinstance(inst, BaseEpochs) else 'raw'
    180         if not inst.preload:
--> 181             raise RuntimeError(
    182                 "By default, MNE does not load data into main memory to "
    183                 "conserve resources. " + msg + ' requires %s data to be '

RuntimeError: By default, MNE does not load data into main memory to conserve resources. adding, dropping, or reordering channels requires raw data to be loaded. Use preload=True (or string) in the constructor or raw.load_data().
```

#### Additional information

Proposition of a `pick`parameter in io.read:
https://github.com/mne-tools/mne-python/issues/1766#issuecomment-71865283

The weirdest part is that you can get the data without preloading with `RAW[picks_chan]` but you are loosing the raw structure (so additional steps are required).
https://github.com/mne-tools/mne-python/issues/4381#issuecomment-315356945

I hope that this issue will be helpful, let me know how can I help.

agramfort, 2020-03-24:
it's not a bug. It's on purpose. You cannot drop channels without having
them in memory.
At least what you ask is not supported.

>

larsoner, 2020-03-24:
Indeed we plan to do it someday, subscribe to #1766 for updates. Closing as dup of #1766

closed 2020-03-24 by larsoner