mne-tools/mne-python #7504: mne.find_event()'s uint_cast needs to be more flexible

hoechenberger, 2020-03-24:
#### Describe the bug
I am currently working with Cam-CAN data, which were acquired using a Neuromag MEG system. According to the `mne.find_events()` docs, there may be issues when reading events created on such a system if `STI016` was active. In the date I have, there is no `STI016`; however, my stimulus channel of interest, `STI101`, yields spurious events, even when passing the recommended `uint_cast=True` parameter. I found that this param causes a type cast to `uint16`; yet, it seems my data requires to be cast to `uint8`.

#### Steps to reproduce
I have prepared a raw FIF file that only contains the `STI101` channel:
https://drive.google.com/open?id=1rpgvxNSyWti6_VoAaFaCqsKgYAQ2wWUh

```
import numpy as np
import matplotlib.pyplot as plt
import mne

infile = '/tmp/STI101_raw.fif'
sti101_raw = mne.io.read_raw_fif(infile, verbose=False)

events = mne.find_events(sti101_raw, verbose=False)
print(np.unique(events[:, 2]))

events_unit_cast = mne.find_events(sti101_raw, verbose=False,
                                   uint_cast=True)
print(np.unique(events_unit_cast[:, 2]))

# Handle the data manually.
data = sti101_raw.get_data().squeeze()
print(np.unique(data))
print(np.unique(data.astype(np.uint8)))

#### Expected results
```
[    1     2     3     4     5     6     7 ]
[    1     2     3     4     5     6     7 ]
[-3.2768e+04 -3.2764e+04  0.0000e+00  1.0000e+00  2.0000e+00  3.0000e+00
  4.0000e+00  5.0000e+00  6.0000e+00  7.0000e+00]
[0 1 2 3 4 5 6 7]
```

#### Actual results
```
[    1     2     3     4     5     6     7 32768]
[    1     2     3     4     5     6     7 32768 32772]
[-3.2768e+04 -3.2764e+04  0.0000e+00  1.0000e+00  2.0000e+00  3.0000e+00
  4.0000e+00  5.0000e+00  6.0000e+00  7.0000e+00]
[0 1 2 3 4 5 6 7]
```
As you can see, only when casting to `uint8`, I get the correct results (last line)

agramfort, 2020-03-24:
did you look at mask parameter?
eg

mne.find_events(raw, uint_cast=True, mask=5)

?

>

hoechenberger, 2020-03-24:
Trying that now. Will report back.

hoechenberger, 2020-03-24:
So I did a few tests where I systematically changes the `mask`, `uint_cast` and `min_duration` parameters:

```python
import numpy as np
import matplotlib.pyplot as plt
import mne

infile = '/tmp/STI101_raw.fif'
sti101_raw = mne.io.read_raw_fif(infile, verbose=False, preload=True)

tests = [dict(name='default', params=dict()),
         dict(name='mask', params=dict(mask=127)),
         dict(name='mask_and_cast', params=dict(mask=127, uint_cast=True)),
         dict(name='min_dur', params=dict(min_duration=0.002)),
         dict(name='mask_and_min_dur',
              params=dict(mask=127, min_duration=0.002)),
         dict(name='mask_and_cast_and_min_dur',
              params=dict(mask=127, uint_cast=True, min_duration=0.002))]

for test in tests:
    print(f'\n==> {test["name"]}:')
    events = mne.find_events(sti101_raw, **test['params'])
    event_counts = np.unique(events[:, 2], return_counts=True)[1]
    print(f'Counts: {event_counts}')
```

Output:
```
==> default:
258 events found
Event IDs: [    1     2     3     4     5     6     7 32768]
Counts: [ 39  39  38   4   5   1   2 130]

==> mask:
136 events found
Event IDs: [  1   2   3   4   5   6   7 124]
Counts: [39 39 38  4  5  1  2  8]

==> mask_and_cast:
136 events found
Event IDs: [1 2 3 4 5 6 7]
Counts: [39 39 38 12  5  1  2]

==> min_dur:
254 events found
Event IDs: [    1     2     3     4     5 32768]
Counts: [ 39  39  38   4   4 130]

==> mask_and_min_dur:
132 events found
Event IDs: [  1   2   3   4   5 124]
Counts: [39 39 38  4  4  8]

==> mask_and_cast_and_min_dur:
132 events found
Event IDs: [1 2 3 4 5]
Counts: [39 39 38 12  4]
```

As you can see, the count of events 1–3 remains stable, which is already good. Yet, some are missing: according to the paper, it should be 40 each.

Event counts for events 4–7 change with the used parameters. Disclaimer: expected would be four events of 4 and 5, each. Events >5 should not occur, but apparently they're due to "event summation". However, this in itself seems to be a different issue.

My current conclusion is that using a `mask` and `min_dir` delivers only the events that would be expected (based on the paradigm description), yet the count of events for event 4 is way off now.

hoechenberger, 2020-03-24:
If I don't use `STI101`, but instead read all `STI001:009` channels manually and sum their signals, then create a virtual new stim channel and run `mne.find_events()` on it, I no longer observe the issue I described initially. So there's something peculiar about the `STI101` channel, which is supposed to represent the sum across the other trigger lines.

agramfort, 2020-03-24:
it's very possible that there is a bug in the generation of STI101 during
acquisition on this dataset.

you should ask cam-can people

cc @dengemann

>

hoechenberger, 2020-03-25:
> it's very possible that there is a bug in the generation of STI101 during acquisition on this dataset.

Could be, I will be in touch with the Cam-CAN people.

For now, I'm using this workaround which seems to work nicely by avoiding `STI101` (albeit it's in _serious_ need of some refactoring):

```python
        stim_data = (raw
                     .copy()
                     .load_data()
                     .pick_channels(stim_chs)
                     .get_data())
        stim_data /= 5  # Signal is always +5V

        # First channel codes for bit 1, second for bit 2, etc.
        for stim_ch_idx, _ in enumerate(stim_chs):
            stim_data[stim_ch_idx, :] *= 2**stim_ch_idx

        # Create a virtual channel which is the sum of the individual channels.
        stim_data = stim_data.sum(axis=0, keepdims=True)
        info = mne.create_info(['STI_VIRTUAL'], raw.info['sfreq'], ['stim'])
        stim_raw = mne.io.RawArray(stim_data, info, first_samp=raw.first_samp)

        events = mne.find_events(stim_raw, stim_channel='STI_VIRTUAL',
                                 min_duration=0.002)
```

agramfort, 2020-03-25:
but on the cam-can side it tells me that bids should write fixed events so
we keep original files
even if they have acquisition issues

>

hoechenberger, 2020-03-25:
> but on the cam-can side it tells me that bids should write fixed events so we keep original files even if they have acquisition issues

Yes I don't touch any of the raw data, if that's what you mean. The raw data is converted to BIDS as-is, aside from the event annotations I add.

hoechenberger, 2020-03-25:
Closing for now as this seems to be an issue with the data itself, nothing that should require fixing in MNE itself.

hoechenberger, 2020-04-30:
tl;dr: This problem does _not_ seem to be limited to a few spurious recordings in the Cam-CAN dataset, but also exists with data acquired in other imaging centers, including NeuroSpin.

I've encountered this (or a related) problem with data acquired as NeuroSpin. Have not yet looked into it in greater detail (i.e., it _could_ also be a different problem, but my gut feeling says it's the exact same). I used my workaround again: manually construct a virtual stim channel based on the individual stim channels in the dataset, and ignore STI101. I'm inclined to wrap this into a helper function and expose it via an `find_events` kwarg.

Re-opening. cc @agramfort

agramfort, 2020-04-30:
go for it. Indeed it's a frequent pb.

>

u01ai11, 2021-03-24:
Not that this impacts the solution, but you might be interested in where these spurious events are coming from. It may be an issue with set-ups which have devices (e.g. button boxes) share channels with software triggers -- which impacts the calculation of STI101. I think this is done to preserve parallel port compatibility for different presentation software. For instance the Cambridge Neuromag setup has this quirk detailed here: https://imaging.mrc-cbu.cam.ac.uk/meg/Triggers .