mne-tools/mne-python #7515: ENH: Add dipole projection in dipole plots

larsoner, 2020-03-25:
In things like:

https://mne.tools/dev/auto_tutorials/source-modeling/plot_visualize_stc.html#dipole-fits

it's difficult to discern the orientation. To help with this, we should project the orientation vector onto each of the three 2D planes that we show.

larsoner, 2020-04-09:
@hoechenberger if you are going to dig into the dipole plotting code, this would be a good place to start to understand what is going on

hoechenberger, 2020-04-09:
@larsoner Can do! Incidentally I also just published a proof-of-concept dipole -> evoked simulator built with the help of @agramfort, https://mybinder.org/v2/gh/hoechenberger/dipoles_demo/master?urlpath=voila%2Frender%2Findex.ipynb

larsoner, 2020-04-27:
Added this to #2995, and assigned that one to you @hoechenberger . Feel free to revisit the suggestions there with @agramfort at some point to see what in that list is still relevant (probably most of them). Though I *think* we took care of the head bound issue so I'll tick that box...

closed 2020-04-27 by larsoner