mne-tools/mne-python #7526: Website examples after #7520

GuillaumeFavelier, 2020-03-27:
I inspected the website frontpage after #7520 and I noticed 3 issues:

1) Taking screenshot with time viewer is not consistent with we had previously
https://18971-1301584-gh.circle-artifacts.com/0/dev/auto_examples/visualization/plot_publication_figure.html#sphx-glr-auto-examples-visualization-plot-publication-figure-py

[image]

2) The titles overlap
https://18971-1301584-gh.circle-artifacts.com/0/dev/auto_examples/inverse/plot_psf_ctf_vertices_lcmv.html#sphx-glr-auto-examples-inverse-plot-psf-ctf-vertices-lcmv-py

[image]

3) The figure is still registered after `close()`
https://18971-1301584-gh.circle-artifacts.com/0/dev/auto_examples/visualization/plot_publication_figure.html#sphx-glr-auto-examples-visualization-plot-publication-figure-py

[image]

larsoner, 2020-03-27:
Can you make a PR to fix these?

GuillaumeFavelier, 2020-03-27:
I have something locally, I'll take care of 1)

closed 2020-04-07 by larsoner