mne-tools/mne-python #7528: ENH: Add montage.rename_channels

larsoner, 2020-03-27:
It should just be a simple dict remapping, I think. It seems like a nicer thing to do than ask people to rename their raw data channels (and possibly rename them back).

agramfort, 2020-03-27:
I don't get it

>

larsoner, 2020-03-27:
In things like:
```
decoding/plot_decoding_csp_eeg.py:raw.rename_channels(lambda x: x.strip('.'))
time_frequency/plot_time_frequency_erds.py:raw.rename_channels(lambda x: x.strip('.'))  # remove dots from channel names
visualization/plot_roi_erpimage_by_rt.py:raw.rename_channels(mapping)
```
we rename the channels in the `raw` instance just so that we can successfully call `apply_montage`. It seems cleaner (and reasonable) to want to keep the original channel names in `raw` as they are, and be able to do `montage.rename_channels(...)` instead of `raw.rename_channels(...)` before calling `raw.set_montage`. If you do want to keep your original channel names, this allows you to do it, otherwise (in `master`) you need another call to `raw.rename_channels({val: key for key, val in mapping})` or so.

There have been several times I have hacked this sort of thing together personally. For example, if I want to replace digitized EEG locations with montage ones (for example for data simulation or testing) and I have the mapping from my EEGXXX to 10-20 channels, it would be nicer to do `read_dig_montage('standard_1020').rename_channels(...)` knowing my mapping than it is to rename my `raw` data to use the `standard_1020` names.

Since raw data is more "permanent" and montages transient / added to the `raw` data, it seems to make sense to make the `montage` the more flexible/changeable entity.

agramfort, 2020-03-27:
got it.

Yes it seems useful indeed.

>

PR #7685 (2020-04-27): MRG, ENH: Add DigMontage.rename_channels
Closes #7528
closed 2020-04-27 by agramfort