mne-tools/mne-python #7540: Error when computing PSDs with eLORETA

fleurgaudfernau, 2020-03-31:
#### Describe the bug
When trying to compute psd over epochs using eLORETA method, the noise_norm matrix used to compute the psd is set to None, hence generating an error.

#### Steps to reproduce
```
import matplotlib.pyplot as plt
import mne
from mne.datasets import sample
from mne.minimum_norm import read_inverse_operator, compute_source_psd_epochs

data_path = sample.data_path()
fname_inv = data_path + '/MEG/sample/sample_audvis-meg-oct-6-meg-inv.fif'
fname_raw = data_path + '/MEG/sample/sample_audvis_raw.fif'
fname_event = data_path + '/MEG/sample/sample_audvis_raw-eve.fif'
label_name = 'Aud-lh'
fname_label = data_path + '/MEG/sample/labels/%s.label' % label_name
subjects_dir = data_path + '/subjects'

event_id, tmin, tmax = 1, -0.2, 0.5
snr = 1.0  # use smaller SNR for raw data
lambda2 = 1.0 / snr ** 2
method = "eLORETA" 

# Load data
inverse_operator = read_inverse_operator(fname_inv)
raw = mne.io.read_raw_fif(fname_raw)
events = mne.read_events(fname_event)

# Set up pick list
include = []
raw.info['bads'] += ['EEG 053']  # bads + 1 more

# pick MEG channels
picks = mne.pick_types(raw.info, meg=True, eeg=False, stim=False, eog=True,
                       include=include, exclude='bads')
# Read epochs
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, picks=picks,
                    baseline=(None, 0), reject=dict(mag=4e-12, grad=4000e-13,
                                                    eog=150e-6))

# define frequencies of interest
fmin, fmax = 0., 70.
bandwidth = 4.  # bandwidth of the windows in Hz

n_epochs_use = 10
stcs = compute_source_psd_epochs(epochs[:n_epochs_use], inverse_operator,
                                 lambda2=lambda2,
                                 method=method, fmin=fmin, fmax=fmax,
                                 bandwidth=bandwidth, label=None,
                                 return_generator=True, verbose=True)

# compute average PSD over the first 10 epochs
psd_avg = 0.
for i, stc in enumerate(stcs):
    psd_avg += stc.data
```

#### Actual results
[image]

#### Additional information
Platform:      Windows-10-10.0.18362-SP0
Python:        3.7.5 (default, Oct 31 2019, 15:18:51) [MSC v.1916 64 bit (AMD64)]
Executable:    C:\Users\XXX\Anaconda3\python.exe
CPU:           Intel64 Family 6 Model 142 Stepping 10, GenuineIntel: 8 cores
Memory:        7.9 GB

mne:           0.20.0
numpy:         1.17.4 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.3.2
matplotlib:    3.1.1 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.22
numba:         0.46.0
nibabel:       2.5.1
cupy:          Not found
pandas:        0.25.3
dipy:          1.0.0
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.9.2}
pyvista:       0.22.4
vtk:           8.1.2

larsoner, 2020-03-31:
@fleurgaudfernau I think I found the problem, can you check to see if #7541 fixes the problem?

fleurgaudfernau, 2020-03-31:
It works now, thank you!

PR #7541 (2020-04-02): MRG, FIX: Fix eLORETA in TF-inverses
Fixes a problem mentioned on the listserv.

Closes #7540
closed 2020-04-02 by larsoner