mne-tools/mne-python #7543: Plot Time_Viewer Sliders not working

jgeller112, 2020-03-31:
When I plot STCs I cannot change any of the slider information. 

Software versions
Python 3.7.4 64bit [Clang 4.0.1 (tags/RELEASE_401/final)]
IPython 7.8.0
OS Darwin 18.7.0 x86_64 i386 64bit
pandas 0.25.1
numpy 1.17.2
seaborn 0.9.0

drammock, 2020-03-31:
### more information needed

Please include the full output of `$ python -c "import mne; mne.sys_info()"` (run from a terminal shell) or

```python
import mne
mne.sys_info()
```

### provide reproducible example code

Please provide a [minimal, complete, and verifiable example](https://stackoverflow.com/help/mcve) that demonstrates the problem you are experiencing. If at all possible, your example should use [MNE-Python's "sample" dataset](https://mne.tools/dev/overview/datasets_index.html#sample) or one of the other built-in datasets.

run from within a Python interpreter.

jgeller112, 2020-03-31:
```
Platform:      Darwin-18.7.0-x86_64-i386-64bit
Python:        3.7.4 (default, Aug 13 2019, 15:17:50)  [Clang 4.0.1 (tags/RELEASE_401/final)]
Executable:    /opt/anaconda3/bin/python
CPU:           i386: 6 cores
Memory:        40.0 GB

mne:           0.20.0
numpy:         1.17.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.3.1
matplotlib:    3.1.3 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.21.3
numba:         0.45.1
nibabel:       3.0.2
cupy:          Not found
pandas:        0.25.1
dipy:          Not found
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.9.2}
pyvista:       0.24.0
vtk:           8.1.2
<ipython-input-335-728c2c85df2a>:1: RuntimeWarning: macOS users should use PyQt5 >= 5.10 for GUIs, got 5.9.2. Please upgrade e.g. with:

    pip install "PyQt5>=5.10"

  mne.sys_info()

```

```
import os

from mne import read_source_estimate
from mne.datasets import sample

print(__doc__)

# Paths to example data
sample_dir_raw = sample.data_path()
sample_dir = os.path.join(sample_dir_raw, 'MEG', 'sample')
subjects_dir = os.path.join(sample_dir_raw, 'subjects')

fname_stc = os.path.join(sample_dir, 'sample_audvis-meg')

```

```
stc = read_source_estimate(fname_stc, subject='sample')

# Define plotting parameters
surfer_kwargs = dict(
    hemi='lh', subjects_dir=subjects_dir,
    clim=dict(kind='value', lims=[8, 12, 15]), views='lateral',
    initial_time=0.09, time_unit='s', size=(800, 800),
    smoothing_steps=5)

# Plot surface
brain = stc.plot(**surfer_kwargs)

# Add title
brain.add_text(0.1, 0.9, 'SourceEstimate', 'title', font_size=16)

```

When visualizing I cant move the sliders--I can send a video?

jgeller112, 2020-03-31:
https://www.youtube.com/watch?v=stGlwd79y7A

drammock, 2020-03-31:
hmm.  I can't reproduce (on Linux)...  @GuillaumeFavelier can you look?

agramfort, 2020-04-01:
I had this issue recently

you should install the Qt version that is recommended
and I had to install vtk 8.2 from conda.

>

larsoner, 2020-04-01:
Indeed the warning tells you to install a newer PyQt. The best command is actually:
```
pip install "PyQt5>=5.10,<5.14"
```

larsoner, 2020-04-01:
(pushed commits to `master` and `maint/0.20` to fix the warning)

larsoner, 2020-04-01:
Should we change the warning to an error? It will make it so that when people do `mne.sys_info` it will raise an error, but maybe it will be more visible that way?

agramfort, 2020-04-01:
I would keep it a warning

>

drammock, 2020-04-01:
Agree it should not be an error. I would also consider adding some really obvious change to the output of sys_info, like a row of 20 "!" marks or something, that called attention to the pyqt line. WDYT?

larsoner, 2020-04-01:
Fine by me

jgeller112, 2020-04-01:
Updated everything

```
Platform:      Darwin-18.7.0-x86_64-i386-64bit
Python:        3.7.6 (default, Jan  8 2020, 13:42:34)  [Clang 4.0.1 (tags/RELEASE_401/final)]
Executable:    /Users/hang/opt/anaconda3/bin/python
CPU:           i386: 6 cores
Memory:        40.0 GB

mne:           0.20.0
numpy:         1.18.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.22.1
numba:         0.48.0
nibabel:       3.0.2
cupy:          Not found
pandas:        1.0.1
dipy:          1.1.1
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.13.2}
pyvista:       0.24.0
vtk:           8.1.2

```
Now it does not use Pyvista to plot but mayavi

jgeller112, 2020-04-01:
Updated everything and it still does not work. 

```
Platform:      Darwin-18.7.0-x86_64-i386-64bit
Python:        3.7.6 (default, Jan  8 2020, 13:42:34)  [Clang 4.0.1 (tags/RELEASE_401/final)]
Executable:    /Users/hang/opt/anaconda3/bin/python
CPU:           i386: 6 cores
Memory:        40.0 GB

mne:           0.20.0
numpy:         1.18.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.22.1
numba:         0.48.0
nibabel:       3.0.2
cupy:          Not found
pandas:        1.0.1
dipy:          1.1.1
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.13.2}
pyvista:       0.24.0
vtk:           8.1.2
```

larsoner, 2020-04-01:
Actually on my macOS machine with VTK 8.1.2 or 8.2.0 I get segfaults when I click anywhere in the window with `time_viewer=True` like:
```
Thread 0 Crashed:
0   libsystem_kernel.dylib        	0x00007fff6e70c7fa __pthread_kill + 10
1   libsystem_c.dylib             	0x00007fff6e6233a2 raise + 26
2   libsystem_platform.dylib      	0x00007fff6e7be42d _sigtramp + 29
3   ???                           	000000000000000000 0 + 0
4   libvtkInteraction-8.1.1.dylib 	0x000000063fec1469 vtkSliderRepresentation2D::StartWidgetInteraction(double*) + 313
5   libvtkInteraction-8.1.1.dylib 	0x000000063fec5ac7 vtkSliderWidget::SelectAction(vtkAbstractWidget*) + 135
6   libvtkCommon-8.1.1.dylib      	0x000000062c0db281 vtkCallbackCommand::Execute(vtkObject*, unsigned long, void*) + 33
7   libvtkCommon-8.1.1.dylib      	0x000000062c245dfc vtkSubjectHelper::InvokeEvent(unsigned long, void*, vtkObject*) + 1004
8   libvtkRenderingKitPython37D-8.1.1.dylib	0x000000063ab149a8 PyvtkRenderWindowInteractor_LeftButtonPressEvent(_object*, _object*) + 136
9   python                        	0x000000010ed0c18e _PyMethodDef_RawFastCallKeywords + 350
10  python                        	0x000000010ee48e42 call_function + 306
11  python                        	0x000000010ee46aec _PyEval_EvalFrameDefault + 46092
12  python                        	0x000000010ed0b8d5 function_code_fastcall + 117
13  python                        	0x000000010ed0ece2 method_call + 130
14  python                        	0x000000010ed0c752 PyObject_Call + 130
15  sip.cpython-37m-darwin.so     	0x0000000121747d40 sip_api_call_procedure_method + 208
16  QtOpenGL.so                   	0x000000064103c1c7 sipQGLWidget::mousePressEvent(QMouseEvent*) + 103
17  QtWidgets                     	0x000000012265d058 QWidget::event(QEvent*) + 424
18  QtOpenGL.so                   	0x000000064103b9a0 sipQGLWidget::event(QEvent*) + 192
...
```
@agramfort can you replicate? At some point today or tomorrow I can try to `git bisect` to see where the problem showed up.

larsoner, 2020-04-01:
... actually that segfault only happens when I have a single time point, e.g., `stc.crop(0.1, 0.1)`. So that's a separate problem.

I can indeed replicate the clicking problem with VTK 8.1.2. I can rotate but not click the sliders. But then if I do:
```
pip uninstall -y vtk pyvista; conda install vtk; pip install --no-cache pyvista
```
it installs VTK 8.2 and things work. @jgeller112 can you try this line to see if it helps?

This suggests that we also need a warning about VTK if you're on macOS and have < 8.2.

jgeller112, 2020-04-01:
I get this error: 
```
ERROR: Cannot uninstall 'vtk'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.

```

larsoner, 2020-04-01:
Try `conda uninstall vtk` rather than `pip uninstall vtk` I guess.

jgeller112, 2020-04-01:
Now I get this completely different error:

Traceback (most recent call last):

  File "<ipython-input-46-286209441ec8>", line 1, in <module>
    brain = b.plot(subjects_dir=subjects_dir)

  File "/Users/hang/opt/anaconda3/lib/python3.7/site-packages/mne/source_estimate.py", line 1539, in plot
    verbose=verbose)

  File "<decorator-gen-129>", line 21, in plot_source_estimates

  File "/Users/hang/opt/anaconda3/lib/python3.7/site-packages/mne/viz/_3d.py", line 1796, in plot_source_estimates
    views=views)

  File "/Users/hang/opt/anaconda3/lib/python3.7/site-packages/mne/viz/_brain/_brain.py", line 210, in __init__
    fig=figure)

  File "/Users/hang/opt/anaconda3/lib/python3.7/site-packages/mne/viz/backends/renderer.py", line 33, in _get_renderer
    return backend._Renderer(*args, **kwargs)

  File "/Users/hang/opt/anaconda3/lib/python3.7/site-packages/mne/viz/backends/_pyvista.py", line 160, in __init__
    self.plotter = self.figure.build()

  File "/Users/hang/opt/anaconda3/lib/python3.7/site-packages/mne/viz/backends/_pyvista.py", line 73, in build
    plotter = self.plotter_class(**self.store)

  File "/Users/hang/opt/anaconda3/lib/python3.7/site-packages/pyvista/plotting/qt_plotting.py", line 671, in __init__
    super(BackgroundPlotter, self).__init__(parent=self.frame,

TypeError: super(type, obj): obj must be an instance or subtype of type

jgeller112, 2020-04-01:
```
Platform:      Darwin-18.7.0-x86_64-i386-64bit
Python:        3.7.6 (default, Jan  8 2020, 13:42:34)  [Clang 4.0.1 (tags/RELEASE_401/final)]
Executable:    /Users/hang/opt/anaconda3/bin/python
CPU:           i386: 6 cores
Memory:        40.0 GB

mne:           0.20.0
numpy:         1.18.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.22.1
numba:         0.48.0
nibabel:       3.0.2
cupy:          Not found
pandas:        1.0.1
dipy:          1.1.1
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.13.2}
pyvista:       0.24.0
vtk:           8.1.2
Exception ignored in: <function BackgroundPlotter.__del__ at 0x7fa483c52440>
Traceback (most recent call last):
  File "/Users/hang/opt/anaconda3/lib/python3.7/site-packages/pyvista/plotting/qt_plotting.py", line 838, in __del__
    if not self._closed:
RuntimeError: super-class __init__() of type BackgroundPlotter was never called

```

larsoner, 2020-04-01:
Your VTK is still 8.1.2. The `conda install vtk` should have installed `8.2`. Maybe you need a `conda update conda` first?

larsoner, 2020-04-01:
... also after you do get `8.2` installed, don't forget to then run:
```
pip uninstall -y pyvista; pip install --no-cache pyvista
```

jgeller112, 2020-04-01:
```
Platform:      Darwin-18.7.0-x86_64-i386-64bit
Python:        3.7.6 (default, Jan  8 2020, 13:42:34)  [Clang 4.0.1 (tags/RELEASE_401/final)]
Executable:    /Users/hang/opt/anaconda3/bin/python
CPU:           i386: 6 cores
Memory:        40.0 GB

mne:           0.20.0
numpy:         1.18.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.22.1
numba:         0.48.0
nibabel:       3.0.2
cupy:          Not found
pandas:        1.0.1
dipy:          1.1.1
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.13.2}
pyvista:       0.24.0
vtk:           8.2.0
```

I was able to update, but again I get this error: 

Exception ignored in: <function BackgroundPlotter.__del__ at 0x7fa483c52440>
Traceback (most recent call last):
  File "/Users/hang/opt/anaconda3/lib/python3.7/site-packages/pyvista/plotting/qt_plotting.py", line 838, in __del__
    if not self._closed:
RuntimeError: super-class __init__() of type BackgroundPlotter was never called

jgeller112, 2020-04-01:
File "<ipython-input-25-635b71f79a05>", line 1, in <module>
    brain = grand_avg.plot(subjects_dir=subjects_dir)

  File "/Users/hang/opt/anaconda3/lib/python3.7/site-packages/mne/source_estimate.py", line 1539, in plot
    verbose=verbose)

  File "<decorator-gen-129>", line 21, in plot_source_estimates

  File "/Users/hang/opt/anaconda3/lib/python3.7/site-packages/mne/viz/_3d.py", line 1796, in plot_source_estimates
    views=views)

  File "/Users/hang/opt/anaconda3/lib/python3.7/site-packages/mne/viz/_brain/_brain.py", line 210, in __init__
    fig=figure)

  File "/Users/hang/opt/anaconda3/lib/python3.7/site-packages/mne/viz/backends/renderer.py", line 33, in _get_renderer
    return backend._Renderer(*args, **kwargs)

  File "/Users/hang/opt/anaconda3/lib/python3.7/site-packages/mne/viz/backends/_pyvista.py", line 160, in __init__
    self.plotter = self.figure.build()

  File "/Users/hang/opt/anaconda3/lib/python3.7/site-packages/mne/viz/backends/_pyvista.py", line 73, in build
    plotter = self.plotter_class(**self.store)

  File "/Users/hang/opt/anaconda3/lib/python3.7/site-packages/pyvista/plotting/qt_plotting.py", line 671, in __init__
    super(BackgroundPlotter, self).__init__(parent=self.frame,

TypeError: super(type, obj): obj must be an instance or subtype of type

jgeller112, 2020-04-01:
I managed to get it working. 

Thanks!

larsoner, 2020-04-01:
What did you do to fix it? It might help the next person who has the problem

jgeller112, 2020-04-01:
```
conda uninstall vtk 
pip uninstall -y pyvista; conda install vtk; pip install --no-cache pyvista
```
I had to restart my Python session after this.

larsoner, 2020-04-01:
Thanks @jgeller112 

New installations following our instructions on macOS should end up with VTK8.2 and proper PyQt5, but still maybe we should add an entry to the troubleshooting page for this for people with existing environments. WDYT @drammock ?

drammock, 2020-04-01:
I'm on it

PR #7547 (2020-04-01): MRG, DOC: macOS VTK troubleshooting & improvements to get_help page
closes #7543
closed 2020-04-01 by drammock