mne-tools/mne-python #7548: Reading .mff file - keyerror: 'EEG'

rahulkakaiya, 2020-04-02:
Python noob. Please help me with this error whilst reading in an .mff file!!! Thank you

INPUT:
```
raw_path = r'desktop/file.mff'
raw_data = mne.io.read_raw_egi(raw_path)
```

OUTPUT:
```
Reading EGI MFF Header from /Users/rahulkakaiya/Desktop/neuro-analysis/Rahul-1-Danielle.mff...
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-14-6153fc6ec773> in <module>
      1 raw_path = r'/Users/rahulkakaiya/Desktop/neuro-analysis/Rahul-1-Danielle.mff'
----> 2 raw_data = mne.io.read_raw_egi(raw_path)

<decorator-gen-167> in read_raw_egi(input_fname, eog, misc, include, exclude, preload, channel_naming, verbose)

/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/mne/io/egi/egi.py in read_raw_egi(input_fname, eog, misc, include, exclude, preload, channel_naming, verbose)
    151     if input_fname.endswith('.mff'):
    152         return _read_raw_egi_mff(input_fname, eog, misc, include,
--> 153                                  exclude, preload, channel_naming, verbose)
    154     return RawEGI(input_fname, eog, misc, include, exclude, preload,
    155                   channel_naming, verbose)

<decorator-gen-165> in _read_raw_egi_mff(input_fname, eog, misc, include, exclude, preload, channel_naming, verbose)

/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/mne/io/egi/egimff.py in _read_raw_egi_mff(input_fname, eog, misc, include, exclude, preload, channel_naming, verbose)
    275     """
    276     return RawMff(input_fname, eog, misc, include, exclude,
--> 277                   preload, channel_naming, verbose)
    278 
    279 

<decorator-gen-166> in __init__(self, input_fname, eog, misc, include, exclude, preload, channel_naming, verbose)

/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/mne/io/egi/egimff.py in __init__(self, input_fname, eog, misc, include, exclude, preload, channel_naming, verbose)
    287         """Init the RawMff class."""
    288         logger.info('Reading EGI MFF Header from %s...' % input_fname)
--> 289         egi_info = _read_header(input_fname)
    290         if eog is None:
    291             eog = []

/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/mne/io/egi/egimff.py in _read_header(input_fname)
    152         Main headers set.
    153     """
--> 154     mff_hdr = _read_mff_header(input_fname)
    155     with open(input_fname + '/signal1.bin', 'rb') as fid:
    156         version = np.fromfile(fid, np.int32, 1)[0]

/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/mne/io/egi/egimff.py in _read_mff_header(filepath)
     28     """
     29     all_files = _get_signalfname(filepath)
---> 30     eeg_file = all_files['EEG']['signal']
     31     eeg_info_file = all_files['EEG']['info']
     32 

KeyError: 'EEG'
```

cbrnr, 2020-04-02:
I don't know the EGI format, but can you confirm that you are trying to load a valid EGI file? That is, have you tried opening it with other programs (e.g. Biosig)?

larsoner, 2020-04-02:
@rahulkakaiya can you try the branch from #7548 and see if it gives you a more informative error message?

PR #7553 (2020-04-03): MRG, ENH: Better error message when EEG data is missing
Closes #7548
closed 2020-04-03 by agramfort