mne-tools/mne-python #7549: BUG Writing forward obtained with make_forward_dipole

agramfort, 2020-04-02:
```
import pathlib
from os import path as op

import numpy as np
import mne

data_path = pathlib.Path(mne.datasets.sample.data_path())
subjects_dir = data_path / 'subjects'
subject = 'sample'
meg_path = data_path / 'MEG' / subject

data_path = mne.datasets.sample.data_path()
subjects_dir = op.join(data_path, 'subjects')
fname_ave = op.join(data_path, 'MEG', 'sample', 'sample_audvis-ave.fif')
fname_bem = op.join(subjects_dir, 'sample', 'bem',
                    'sample-5120-5120-5120-bem-sol.fif')
bem = mne.read_bem_solution(fname_bem, verbose=False)
fname_trans = op.join(data_path, 'MEG', 'sample',
                      'sample_audvis_raw-trans.fif')
info = mne.io.read_info(fname_ave, verbose=False)
info['bads'] = []
info['projs'] = []
picks = mne.pick_types(info, meg=True, eeg=True)
info = mne.pick_info(info, picks)

x_dip, y_dip, z_dip = 20, 20, 40
ori = np.array([[1, 0, 0]])
dip_pos = (x_dip, y_dip, z_dip)
pos = np.array([dip_pos]) / 1e3
dip = mne.Dipole(times=[0], pos=pos, amplitude=[10e-9], ori=ori, gof=[100])
fwd = mne.make_forward_dipole(dip, fname_bem, info, fname_trans)
mne.write_forward_solution('tmp-fwd.fif', fwd, overwrite=True)
```

gives

```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-3-8057eaeb0edb> in <module>
     30 dip = mne.Dipole(times=[0], pos=pos, amplitude=[10e-9], ori=ori, gof=[100])
     31 fwd = mne.make_forward_dipole(dip, fname_bem, info, fname_trans)
---> 32 mne.write_forward_solution('tmp-fwd.fif', fwd, overwrite=True)

<decorator-gen-258> in write_forward_solution(fname, fwd, overwrite, verbose)

~/work/src/mne-python/mne/forward/forward.py in write_forward_solution(fname, fwd, overwrite, verbose)
    739     start_block(fid, FIFF.FIFFB_MNE_ENV)
    740     write_id(fid, FIFF.FIFF_BLOCK_ID)
--> 741     data = fwd['info'].get('working_dir', None)
    742     if data is not None:
    743         write_string(fid, FIFF.FIFF_MNE_ENV_WORKING_DIR, data)

TypeError: tuple indices must be integers or slices, not str
```

cc @hoechenberger

larsoner, 2020-04-02:
https://mne.tools/stable/generated/mne.make_forward_dipole.html

`make_forward_dipole` returns a tuple of `fwd`, `stc`

agramfort, 2020-04-02:
now the bug report is clear

larsoner, 2020-04-02:
Yes, now it's a bug and not a usage error :)

PR #7550 (2020-04-02): MRG, FIX: Fix make_forward_dipole completion
Closes #7549
closed 2020-04-02 by larsoner