mne-tools/mne-python #7564: issue with volumetric source estimate morphing

fleurgaudfernau, 2020-04-06:
#### Describe the bug
I have mixed source estimates that I wish to morph to the fsaverage brain. To do so, I separated the mixed source estimates into a surface source estimate and a volume source estimate. When trying to morph the lastest (using morph_apply), I get the following error:
ValueError: vertices do not match between morph (77) and stc (33) for the volume:
(also, I believe there is a small mistake in morph_apply and what it means is actually that vertices do not match between morph (33) and stc (77) for the volume.)

I believe the error is coming from the fact that when computing the source morph, mne only considers the first of the two volume source spaces in cereb_used, hence generating a morph with 33 vertices instead of 77 vertices, but I do not know how to fix it.

#### Steps to reproduce
```
import mne
import os
import os.path as op
import numpy as np
from mne.datasets import sample, fetch_fsaverage
from mne.minimum_norm import apply_inverse, read_inverse_operator, make_inverse_operator

data_path = mne.datasets.sample.data_path()
subject = 'sample'
data_dir = op.join(data_path, 'MEG', subject)
subjects_dir = op.join(data_path, 'subjects')
bem_dir = op.join(subjects_dir, subject, 'bem')

fname_mixed_src = op.join(bem_dir, '%s-oct-6-mixed-src.fif' % subject)
fname_aseg = op.join(subjects_dir, subject, 'mri', 'aseg.mgz')
fname_model = op.join(bem_dir, '%s-5120-bem.fif' % subject)
fname_bem = op.join(bem_dir, '%s-5120-bem-sol.fif' % subject)
fname_evoked = data_dir + '/sample_audvis-ave.fif'
fname_trans = data_dir + '/sample_audvis_raw-trans.fif'
fname_fwd = data_dir + '/sample_audvis-meg-oct-6-mixed-fwd.fif'
fname_cov = data_dir + '/sample_audvis-shrunk-cov.fif'

#create a mixed source space
labels_vol = ['Left-Cerebellum-Cortex',
              'Right-Cerebellum-Cortex']
src = mne.setup_source_space(subject, spacing='oct5',
                             add_dist=False, subjects_dir=subjects_dir)

vol_src = mne.setup_volume_source_space(
    subject, mri=fname_aseg, pos=10.0, bem=fname_model,
    volume_label=labels_vol, subjects_dir=subjects_dir,
    add_interpolator=False,  # just for speed, usually this should be True
    verbose=True)

#Generate the mixed source space
src += vol_src

fwd = mne.make_forward_solution(
    fname_evoked, fname_trans, src, fname_bem,
    mindist=5.0,  # ignore sources<=5mm from innerskull
    meg=True, eeg=False, n_jobs=1)

leadfield = fwd['sol']['data']
src_fwd = fwd['src']
n = sum(src_fwd[i]['nuse'] for i in range(len(src_fwd)))

condition = 'Left Auditory'
evoked = mne.read_evokeds(fname_evoked, condition=condition,
                          baseline=(None, 0))
noise_cov = mne.read_cov(fname_cov)

snr = 3.0            # use smaller SNR for raw data
inv_method = 'MNE'  # sLORETA, MNE, dSPM
#parc = 'aparc'       # the parcellation to use, e.g., 'aparc' 'aparc.a2009s'
lambda2 = 1.0 / snr ** 2
inverse_operator = make_inverse_operator(evoked.info, fwd, noise_cov,
                                         depth=None, fixed=False)

stc = apply_inverse(evoked, inverse_operator, lambda2, inv_method,
                    pick_ori=None)

print(src)
print(stc.vertices)

#morphing
#recuperating volume source estimate from mixed source estimate stc
nb_sources_cortex = stc.vertices[0].shape[0] + stc.vertices[1].shape[0]
vertices_vol = np.concatenate((stc.vertices[2], stc.vertices[3]))
stc_vol = mne.VolSourceEstimate(stc.data[nb_sources_cortex:], vertices_vol, 
                                    tmin=stc.tmin, tstep=stc.tstep,
                                    subject = stc.subject)
#Maybe not the cleanest way to isolate a volume source estimate from a mixed source estimate?

#recuperating the source space I want to morph to
src_fsaverage_vol = subjects_dir + '/fsaverage/bem/fsaverage-vol-5-src.fif'
src_vol  = mne.read_source_spaces(src_fsaverage_vol)

#recuperating the source space to morph from
cereb_used = mne.SourceSpaces(inverse_operator['src'][2:])
#the source space (inverse_op['src']) is made of 4 source spaces: 2 surface source spaces and 2 volume source spaces
#- here I recuperate the 2 volume source spaces

morph = mne.compute_source_morph(src = cereb_used, 
                                    subject_from=stc_vol.subject,
                                    subject_to = 'fsaverage', 
                                    subjects_dir=subjects_dir,
                                    niter_affine=[10, 10, 5], niter_sdr=[10, 10, 5],  # just for speed
                                    src_to=src_vol, verbose=True)

stc_vol = morph.apply(stc_vol)
```

#### Actual results
ValueError: vertices do not match between morph (77) and stc (33) for the volume:
[ 619  620  634  635  636  651  860  874  875  876  888  903  906  918
  935 1129 1131 1143 1144 1145 1146 1158 1159 1160 1174 1188 1189 1204
 1416 1430 1431 1445 1716  608  609  623  624  625  638  639  640  652
  653  654  668  863  864  877  879  880  892  896  907  911  922  923
  924  925 1133 1147 1148 1149 1150 1162 1164 1165 1177 1180 1195 1209
 1417 1433 1434 1448 1449 1464 1717]
[ 619  620  634  635  636  651  860  874  875  876  888  903  906  918
  935 1129 1131 1143 1144 1145 1146 1158 1159 1160 1174 1188 1189 1204
 1416 1430 1431 1445 1716]

#### Additional information
Platform: Windows-10-10.0.18362-SP0
Python: 3.7.5 (default, Oct 31 2019, 15:18:51) [MSC v.1916 64 bit (AMD64)]
Executable: C:\Users\XXX\Anaconda3\python.exe
CPU: Intel64 Family 6 Model 142 Stepping 10, GenuineIntel: 8 cores
Memory: 7.9 GB

mne: 0.20.0
numpy: 1.17.4 {blas=mkl_rt, lapack=mkl_rt}
scipy: 1.3.2
matplotlib: 3.1.1 {backend=module://ipykernel.pylab.backend_inline}

sklearn: 0.22
numba: 0.46.0
nibabel: 2.5.1
cupy: Not found
pandas: 0.25.3
dipy: 1.0.0
mayavi: 4.7.1 {qt_api=pyqt5, PyQt5=5.9.2}
pyvista: 0.22.4
vtk: 8.1.2

larsoner, 2020-04-09:
@fleurgaudfernau just to get a bigger picture here. Would it work better if you could:

1. Compute a single morph for your surface+vol source space
2. Do `morph.apply(stc_mixed)` and get a mixed STC back
3. Extract the vol or surface STC from the mixed STC

? It seems like it would accomplish what you are trying to do without a bunch of messing around with `src` indexing.

If that sounds like a reasonable API, I'll probably work toward that. I think doing so correctly would fix your use case along the way.

fleurgaudfernau, 2020-04-10:
@larsoner Yes, that would be perfect!
(It was actually what I intended to do in the first place, but I've noticed that MNE doesn't have a lot of functionalities for Mixed Source Estimates and it is often simpler to separate them into surface and volume.)

Also, is it right to extract volume and surface STC from a mixed STC the way I do?
```
#morphing
#recuperating volume source estimate from mixed source estimate stc
nb_sources_cortex = stc.vertices[0].shape[0] + stc.vertices[1].shape[0]
vertices_vol = np.concatenate((stc.vertices[2], stc.vertices[3]))
stc_vol = mne.VolSourceEstimate(stc.data[nb_sources_cortex:], vertices_vol, 
                                    tmin=stc.tmin, tstep=stc.tstep,
                                    subject = stc.subject)
```
It would be very convenient (and reduce errors) to have a function doing it directly.

larsoner, 2020-04-10:
> Also, is it right to extract volume and surface STC from a mixed STC the way I do?

It looks reasonable except that I'm not sure that `VolSourceEstimate` is actually properly equipped to handle multiple lists of vertices, usually it's just a single one. Things might break for example when you go to plot with it, not sure. But these are the sort of bugs we'd want to work out in `MixedSourceEstimate.volume()` for example.

larsoner, 2020-04-23:
Reopening this one until we have tests for `SourceMorph` with mixed source estimates

PR #7597 (2020-04-21): MRG, ENH: Improve mixed source estimate support
- [x] Redefine "mixed" source estimate as two cortical surface source spaces, plus others
- [x] Add MixedVectorSourceEstimate
- [x] Add tests for generating all source space types in inversion (mixed not tested, mixed vector not tested)
- [x] Add MixedVectorSourceEstimate.surface() and .volume() to allow splitting (someday later maybe we'll add putting them back together but not for now)
- [x] Add round-trip I/O tests for Mixed(Vector)SourceEstimate
- [x] Add support for MixedSourceEstimate to `extract_label_time_course`
- [x] Deprecate `MixedSou [... cut]
PR #7734 (2020-05-29): MRG, ENH: Add mixed source estimate support to compute_source_morph
Closes #7564.
Closes #7007.

@fleurgaudfernau can you see if it works for your use case? For example this code now runs fine:

<details>
<summary>Starter code</summary>

```Python
import mne
import os.path as op
import numpy as np

data_path = mne.datasets.testing.data_path()
subject = 'sample'
data_dir = op.join(data_path, 'MEG', subject)
subjects_dir = op.join(data_path, 'subjects')
bem_dir = op.join(subjects_dir, subject, 'bem')

fname_src = op.join(bem_dir, 'sample-oct-4-src.fif')
fname_aseg = op.join(subjects_dir, subject [... cut]
closed 2020-05-29 by agramfort