mne-tools/mne-python #7570: mne.bem.make_watershed_bem() Coordinate System Error

mh105, 2020-04-07:
**READ THIS FIRST:** If you are having trouble getting MNE-Python to work with
your own data, you should probably ask for help on our
[Gitter channel](https://gitter.im/mne-tools/mne-python) or on our
[mailing list](https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis)
before opening an issue; you will get help faster that way. See
[the FAQ entry on filing bug reports](https://mne.tools/dev/overview/faq.html#i-think-i-found-a-bug-what-do-i-do)
for more guidance. If you're pretty sure your problem is a bug in MNE-Python,
**delete this section** and fill in the headings below, replacing the
placeholder text with your own responses. Surround any code samples with triple
backticks above and below the code block, or put code in a
[public gist](https://gist.github.com) and link to it in the issue.

#### Describe the bug
Incorrect extraction of volume info by the _extract_volume_info method in make_watershed_bem() in bem.py. If you look at the extracted BEM surfaces using the ancient mne freeview_bem_surfaces method from terminal that calls Freeview, you will see that the BEM surfaces have flipped yz coordinates. This is due to incorrect use of the header['Mdc'].T matrix. 

#### Steps to reproduce
Sorry I can't provide a snippet since this bug requires examining the BEM surfaces outside MNE-Python. But if you just run mne.bem.make_watershed_bem() on any Freesurfer T1 recon, and then use freeview to check the surfaces. For instance, you can use the MNE-C method mne freeview_bem_surfaces -s subject to check (or you can manually do it by freeview -f surfacefile) in the watershed folder, you will see the yz dimensions are flipped. 

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Expected results
Well.. the BEM surface should have the same volume-geometry as the T1.mgz from Freesurfer recon. It is not in version 0.20.0. 

#### Actual results
It's flipped in yz dimensions. 

#### Additional information
Ok here's some historical information: this is an inherited bug since long time ago when initially there was the bug on the line "write_surface(s, rr, tris, volume_info=volume_info)" that should have been "write_surface(surf_ws_out, rr, tris, volume_info=volume_info)". I don't know at which version this is fixed but it is correct in 0.20.0. However the _extract_volume_info method is wrong. header['Mdc'].T is not [xras, yras, bras]. Instead, it is [rxyz, axyz, sxyz]. So you need to transpose the matrix before assigning the ras direction vectors in vol_info. 

Here is an easy fix: change line 1229 from 
vol_info['xras'], vol_info['yras'], vol_info['zras'] = header['Mdc'].T
to 
vol_info['xras'], vol_info['yras'], vol_info['zras'] = list(zip(*header['Mdc'].T))
and you will fix it. 

In theory if everyone's using freesurfer correctly then the outputted surfaces will already be in the right Freesurfer coordinates. So just like you don't update volume_geometry in mne.bem.make_flash_bem(), you don't need to update watershed surfaces in mne.bem.make_watershed_bem() either. I could see the initial thought was probably to accommodate T1.mgz beyond the default output volume from recon-all pipeline, so I guess it makes more sense to fix this transpose error than to remove the volume-geometry update codes all together. Just as a further suggestion, there is a useful freesurfer command mris_convert --vol-geom volume function that handles all these different structural scan type issues. So you might just want to call mris_convert in a subprocess.run() call that will save you a lot of trouble and avoid having to define your own _extract_volume_info() function, that is dependent on how nibable reads header files. Cheers.

agramfort, 2020-04-07:
I just did `mne freeview_bem_surfaces -s sample` and it looks fine. Can you point to a dataset for which it's broken? It also works fine on somato dataset.

mh105, 2020-04-07:
Did you re-run mne.bem.make_watershed_bem() on sample or were you viewing the generated watershed BEM surfaces? I believe the BEM surfaces generated before v0.19 were correct because of the error on the line write_surface(s, rr, tris, volume_info=volume_info) that failed to update volume geometry on the generated surface files. Since they are in the correct coordinates after generated by freesurfer watershed, not updating actually gave the right volume geometry.

larsoner, 2020-04-07:
> Sorry I can't provide a snippet since this bug requires examining the BEM surfaces outside MNE-Python. 

Even something scripted that requires Freeview is fine. For example I just tried:
```
$ cd ~/mne_data/MNE-sample-data/subjects
$ export SUBJECTS_DIR=$PWD
$ export SUBJECT=sample
$ rm -Rf sample/bem/*.surf sample/bem/watershed
$ mne watershed_bem -s sample -d $PWD --verbose
$ freeview -v sample/mri/T1.mgz -f sample/bem/inner_skull.surf
```
And got:

[image]

Can you make a PR to fix the behavior? Then we can test locally and see that it works. If you want to go the extra mile, you could look into modifying `test_watershed_bem` in `mne/commands/tests/test_commands.py` to examine the volume info of the output surface. There is already a `read_surface` call at the end, I think if you modify it to do `read_metadata=True` you could come up with some check of the xras yras and/or zras that should fail on `master` but pass on this PR.

mh105, 2020-04-07:
Maybe I'm missing something, but isn't the surface shown in your plot upside-down?? That's precisely the issue I had.

larsoner, 2020-04-07:
Yes, I mean to say that I can replicate the issue :)

Are you up for making a PR to fix it?

mh105, 2020-04-07:
Oh, I see. I read the "Even something scripted that requires Freeview is fine" as the surface looks fine. Sure... I can do that, but which fix do you want? 
1) Remove volume geometry update all together from mne.bem.make_watershed_bem()
2) Do the transpose fix of _extract_volume_info method 
3) Do a subprocess call of mris_convert --vol-geom to update volume geometry
Any of the three will fix this issue.

bloyl, 2020-04-07:
As a user who just generated a ton of bems using 0.20, is this a
consistency with freeview/freesurfer visualization bug or will it also
effect MNE coregistration/source modeling/forward generation as well?

mh105, 2020-04-07:
I'll let the experts confirm this, but I think it does. It's not a freeview visualization bug but an error on volume geometry of BEM surfaces. Your forward model will be wrong since the source space is flipped upside-down from your BEM surfaces.

larsoner, 2020-04-07:
> Actually now I think about it more carefully, your forward model should be fine. I believe at no point is MNE using the volume geometry explicitly in computations, so the coordinates should have been kept intact.

That's what I expect. @bloyl as long as `plot_alignment` looks correct, then things should be correct. We try to keep the transformations there and in the forward code consistent, and indeed we never use the volume information.

> 2. Do the transpose fix of _extract_volume_info method

This is the fix I would do. It's the least invasive (better than option 3 in this sense) yet keeps things the most correct / complete (better than option 1 in this sense).

mh105, 2020-04-07:
> This is the fix I would do. It's the least invasive (better than option 3 in this sense) yet keeps things the most correct / complete (better than option 1 in this sense).

Sounds good. Will push later this week.

yh-luo, 2020-04-15:
> Oh, I see. I read the "Even something scripted that requires Freeview is fine" as the surface looks fine. Sure... I can do that, but which fix do you want?
>
>     1. Remove volume geometry update all [...]

Hi @mh105 , I noticed that `list(zip(*header['Mdc'].T))` is the same as `header['Mdc']`:
```
>> list(zip(*header['Mdc'].T))
[(-1.0, 8.56817e-08, 1.4901168e-08),
 (1.15484006e-07, 3.1432176e-08, -1.0),
 (-1.9185245e-07, 1.0, 1.2805678e-08)]

>> header['Mdc']
array([[-1.00000000e+00,  8.56816982e-08,  1.49011683e-08],
       [ 1.15484006e-07,  3.14321760e-08, -1.00000000e+00],
       [-1.91852450e-07,  1.00000000e+00,  1.28056783e-08]], dtype=float32)
```

So `vol_info['xras'], vol_info['yras'], vol_info['zras'] = header['Mdc']` seems to fix the problem?
[image]

mh105, 2020-04-15:
> Hi @mh105 , I noticed that `list(zip(*header['Mdc'].T))` is the same as `header['Mdc']`:

Very true. Let me just do that instead of transposing back.

yh-luo, 2020-04-18:
> > Hi @mh105 , I noticed that `list(zip(*header['Mdc'].T))` is the same as `header['Mdc']`:
> 
> Very true. Let me just do that instead of transposing back.

After digging in the file history, the bug was likely caused by misunderstanding of `header_dtd['Mdc']` in [`nibabel/freesurfer/mghformat.py`](https://github.com/nipy/nibabel/blob/e2f50b4cf11482e6f018195aa6b89ccf12a3141e/nibabel/freesurfer/mghformat.py).

### Details
`header_dtd['Mdc']` was originally considered to be the cosines transform according to [FreeSurfer WiKi](https://surfer.nmr.mgh.harvard.edu/fswiki/FsTutorial/MghFormat).
That might be the reason why `_extract_volume_info` acquires the RAS coordinates by transposing `header_dtd['Mdc']`. Because of the bug described in #7139, the output surfaces were not updated by `_extract_volume_info`. So they were correct and the bug went unnoticed. 
After several commits, the annotation of `header_dtd` then indicated that `Mdc` is actually the _TRANSPOSE_ of direction cosine matrix, which are the RAS coordinates (hopefully not wrong, I'm not familiar with the coordinate systems). Because #7139 was fixed in v20, the bug in `_extract_volume_info` then results in the coordinate issue.

Technically, I caused this error by #7144... I made some [changes](https://github.com/mne-tools/mne-python/compare/master...yh-luo:fix_volume_info) to deal with the it due to guilty. If you could not make a PR please let me know. Would fix it happily :sweat_smile:

mh105, 2020-04-18:
Oh I see. Yeah go ahead with PR. I was able to create the PR but I realized you already fixed it with the changes. You have better documentations on this as well =) Cheers

PR #7658 (2020-04-22): FIX: Fix RAS coordinates in watershed bem surfaces
#### Reference issue
Fixes #7570.

#### What does this implement/fix?
In v20, the RAS coordinates of the surfaces created by `mne.bem.make_water_shed_bem` were wrong (refer to #7570 for details). This PR fixes the problem and adds tests to `mne/commands/test_commands/test_watershed_bem` to check the RAS coordinates of the watershed BEM surfaces.

#### Additional information
The bug was found thanks to @mh105, replicated by @larsoner.
closed 2020-04-22 by larsoner