mne-tools/mne-python #7571: Segfault with raw.plot() on macOS

cbrnr, 2020-04-07:
Consider the following example:

```python
import numpy as np
import mne
info = mne.create_info(10, 500, "eeg")
raw = mne.io.RawArray(np.random.normal(scale=1e-5, size=(10, 10000)), info)
raw.plot()
```

When I run this script interactively (i.e. from IPython with `%run plot_example.py`) everything works fine. However, when I run the script with `python3 plot_example.py`, I get a segfault:

```
[1]    44467 segmentation fault  python3 plot_example.py
```

I use Python 3.8.2 installed via Homebrew, but the same happens with Anaconda Python 3.7.6 and 3.8.2. Is it just me or can someone reproduce this?

agramfort, 2020-04-07:
I don't have the pb

>

cbrnr, 2020-04-07:
Are you on a Mac? Which Python version do you use? Anaconda or Homebrew?

cbrnr, 2020-04-07:
Actually, the problem occurs only for `raw.plot(block=False)` (the default). Everything works if I run `raw.plot(block=True)` instead.

cbrnr, 2020-04-07:
Furthermore, I'm using the default `macosx` backend. I don't get the segfault with the `qt5agg` backend.

agramfort, 2020-04-07:
I confirm the segfault :

$MPLBACKEND=macosx python example.py
Segmentation fault: 11

>

larsoner, 2020-04-07:
The macOS backend always seems to have problems. I think we should recommend that people use PyQt5 in our install docs.

This sounds like an upstream bug with that backend so it would be good if you can strip down the plotting code to a minimal example and file an upstream bug. In the meantime I think we can leave this open here until we update our recommendation, and maybe add a workaround if it turns out to be doable.

cbrnr, 2020-04-07:
I'll try to isolate this issue. Let's wait until we know more before we change any recommendations.

cbrnr, 2020-04-07:
This seems to be the offending line:

https://github.com/mne-tools/mne-python/blob/487759e4b8f003311c1653f2f54592609a42c029/mne/viz/raw.py#L469

Commenting it out solves the segfault.

cbrnr, 2020-04-07:
Or rather, it's this line in `_close_event`:

https://github.com/mne-tools/mne-python/blob/487759e4b8f003311c1653f2f54592609a42c029/mne/viz/raw.py#L528

larsoner, 2020-04-07:
> Let's wait until we know more before we change any recommendations.

I'm not sure what else there is to wait for -- I've seen a handful of reports (from you mostly I think?) that end up being macOS-specific bugs not present in PyQt5. I'd also rather recommend a single backend (PyQt5) because it gives us a more uniform platform to try to support, especially since this is also what's used by our 3D plotting tools.

hoechenberger, 2020-04-07:
> that end up being macOS-specific bugs not present in PyQt5.

I always thought this was the "officially" recommended backend anyway :) Why not emit a warning if a user wants to plot using a different backend?

cbrnr, 2020-04-07:
Because `macosx` ist still the default backend on a Mac and it doesn't require an additional package. So I'd prefer if we just fix this bug.

cbrnr, 2020-04-07:
> FWIW our recommended env (and install docs, etc.) installs PyQt5, so I don't think this is a relevant consideration for what we recommend to our users. So I'm still +1 on officially recommending people not use the macOS backend...

You may be right. I'm using it anyway in MNELAB, but in an interactive session I'm too lazy to switch backends (yes I know that there is `matplotlibrc`). Even if we recommend the PyQt5 backend it would be nice if everything worked with the matplotlib defaults on all platforms. I don't even know why matplotlib doesn't make PyQt5 its default on macOS - it would be interesting to know why (maybe because of the additional package). According to this [SO answer](https://stackoverflow.com/a/58838910), they'd only need to move `macosx` back by one item.

> @cbrnr can you look into whittling down the plotting code to a minimal example and file an upstream bug? It should really be fixed there. We can work around it eventually, but it would be nice to wait to see how they officially fix it first.

I'll try.

cbrnr, 2020-04-07:
This will be a bit hard to debug, because the segfault doesn't occur when debugging.

cbrnr, 2020-04-07:
Reported upstream: https://github.com/matplotlib/matplotlib/issues/17061

larsoner, 2020-04-27:
I'll close since it looks like @cbrnr is taking care of this upstream in https://github.com/matplotlib/matplotlib/pull/17084 (thanks!)

closed 2020-04-27 by larsoner