mne-tools/mne-python #7575: Anonymized raw can't be saved: mne.io.anonymize_info() doesn't seem to work on the annotations

adam2392, 2020-04-07:
#### Describe the bug
When I instantiate a raw object, attach annotations, and I perform `mne.io.anonymize_info()` after I get an error. I believe this is due to the fact that anonymization doesn't "alter" the annotations. It works fine if I call anonymization before setting annotations.

Possible solutions:

1. better warning/error message
2. anonymization to alter the annotations - origtime as well

#### Steps to reproduce
When I run:

```
# set a measurement date to allow anonymization to run
        raw.set_meas_date(meas_date=datetime.datetime.now(tz=datetime.timezone.utc))
        mne.io.anonymize_info(raw.info)
        raw.save(tmpfpath)
```

#### Actual results
I get the following:

```
  File "/Users/adam2392/Documents/SPORF/mtsmorf/io/run_bids_conversion.py", line 56, in convert_mat_to_bids
    raw.save(tmpfpath)
  File "<decorator-gen-155>", line 21, in save
  File "/Users/adam2392/miniconda3/envs/rf/lib/python3.8/site-packages/mne/io/base.py", line 1345, in save
    _write_raw(fname, self, info, picks, fmt, data_type, reset_range,
  File "/Users/adam2392/miniconda3/envs/rf/lib/python3.8/site-packages/mne/io/base.py", line 1834, in _write_raw
    sk_onsets, sk_ends = _annotations_starts_stops(raw, 'bad_acq_skip')
  File "/Users/adam2392/miniconda3/envs/rf/lib/python3.8/site-packages/mne/annotations.py", line 521, in _annotations_starts_stops
    onsets = _sync_onset(raw, onsets)
  File "/Users/adam2392/miniconda3/envs/rf/lib/python3.8/site-packages/mne/annotations.py", line 495, in _sync_onset
    assert raw.info['meas_date'] == raw.annotations.orig_time
```

#### Additional information
Using v0.20

agramfort, 2020-04-07:
can you share a short code snippet to replicate the pb?

>

bloyl, 2020-04-07:
`anonymize_info` only anonymizes the `meas_info` object. If you want to
anonymize the `raw` (both the `meas_info` and `annotations`) you need to
call `raw.anonymize()`

closed 2020-04-08 by agramfort

adam2392, 2020-04-08:
Awesome. Thanks!

PR #7587 (2020-04-08): MRG, DOC: Document anonymization better
Closes #7575 

@drammock feel free to review and merge if you're happy.