mne-tools/mne-python #7577: RFC: Remove "spectrum_fit" from notch filter

larsoner, 2020-04-07:
I have never really gotten it to work in practice, and users on Gitter are unhappy that it takes ages to run. I wonder if we should deprecate and remove it, as I'm not sure if fixing it is even really feasible.

mmagnuski, 2020-04-08:
Maybe we should move it to sandbox? But I'm fine either way.

sappelhoff, 2020-04-10:
this is relevant to https://github.com/sappelhoff/pyprep/issues/18

@yjmantilla

yjmantilla, 2020-04-10:
What I can contribute to this is that in the tests mentioned in sappelhoff/pyprep#21 it seems that the agreement between the prep cleanline method and the mne spectrum notch is high but I haven't really tested it focusing on that part. Maybe doing some simulations would help us determining if it is worthwhile leaving it there.

larsoner, 2020-04-10:
@mmagnuski suggested that this deficiency might be due to not operating on windows of raw data, which is entirely possible. I've opened a PR to make a general overlap-add processing class in #7598 that would hopefully make it easy to do so.

christian-oreilly, 2020-07-31:
I just want to confirm, in case that can be useful, that, as it has been mentioned in #7598, at least part of the very long processing for this function comes from the fact that the algorithm is applied on the whole raw file (or at least, much too long part of it). I pinpointed for my case where it was hanging and it was at the line `dpss = _get_dpss()(N, half_nbw, Kmax)` in the function `dpss_windows(...)`. It resulted in a call equivalent to `scipy.signal.windows.dpss(100000, 3383.0, 6766)` which takes forever. It now works quite well and within an acceptable time with #7609, using filter_length='10s'.

PR #7609 (2020-08-11): MRG, ENH: Use overlap-add in spectrum_fit mode
Closes #7577.

Based on some local testing I'm not convinced it works. But @yjmantilla @sappelhoff @mmagnuski if you want to try it just pass `filter_length='10s'` or whatever to `notch_filter`. Maybe this will make it easier for people to test against `cleanline` or other implementations, and/or push fixes.
closed 2020-08-11 by drammock