mne-tools/mne-python #7595: Evoked.plot_topomap() behaves in strange ways when axes is passed

hoechenberger, 2020-04-10:
#### Describe the bug
`Evoked.plot_topomap()` produces unexpected behavior for me when I pass an `Axes` object via the `axes` keyword argument.

#### Steps to reproduce
```python
import matplotlib
matplotlib.use('Qt5Agg')

import matplotlib.pyplot as plt
from mne.datasets import sample
from mne import read_evokeds

path = sample.data_path()
fname = path + '/MEG/sample/sample_audvis-ave.fif'

condition = 'Left Auditory'
evoked = read_evokeds(fname, condition=condition, baseline=(None, 0))

times = [0.05]

for colorbar_0, colorbar_1 in zip([False, True, True],
                                  [False, True, False]):
    fig, ax = plt.subplots(2, 2)
    evoked.plot_topomap(times, ch_type='mag', time_unit='s',
                        axes=ax[0, 0], show=True, colorbar=colorbar_0)
    evoked.plot_topomap(times, ch_type='mag', time_unit='s',
                        axes=ax[1, 0], show=True, colorbar=colorbar_1)
```

#### Expected results
Not to sure, actually! But certainly not what I do get :)

#### Actual results
##### `colorbar_0 = colorbar_1 = False`
Two figures appear:
[image]
[image]

##### `colorbar_0 = colorbar_1 = True`
Two figures appear:
[image]
[image]

##### `colorbar_0 =True; colorbar_1 = False`
Two figures appear:
[image]
[image]

#### Additional information
```
Platform:      macOS-10.15.4-x86_64-i386-64bit
Python:        3.8.2 | packaged by conda-forge | (default, Mar 23 2020, 17:55:48)  [Clang 9.0.1 ]
Executable:    /Users/hoechenberger/miniconda3/envs/mne-dev-new/bin/python
CPU:           i386: 8 cores
Memory:        16.0 GB

mne:           0.21.dev0
numpy:         1.18.1 {blas=blas, lapack=blas}
scipy:         1.4.1
matplotlib:    3.2.1 {backend=Qt5Agg}

sklearn:       0.22.2.post1
numba:         Not found
nibabel:       3.0.2
cupy:          Not found
pandas:        1.0.3
dipy:          1.1.0
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.12.3}
pyvista:       0.24.0
vtk:           8.2.0
```

hoechenberger, 2020-04-10:
I think it would make sense to allow users to pass `axes` for the topomaps themselves, and `cbar_axes` to place the colorbars into!

agramfort, 2020-04-10:
yes either this or require that if colorbar=True then one need to pass
n_axes = n_times + 1 where then last one is chosen
to place the colorbar. Currently I agree it's a mess

>

drammock, 2020-04-10:
crossref #7586, maybe makes sense to tackle both of these at the same time

mmagnuski, 2020-04-16:
are you using inline backend? If so use `show=False`.

mmagnuski, 2020-04-16:
@agramfort 
I don't get missing topos (or duplicated figures) on windows in interactive backend. I was asking about inline because `show=True` shows multi-topo figure too early, so some topos are missing then.

agramfort, 2020-04-16:
to me the pb here is really about colorbar axes that is too magical now

>

PR #7706 (2020-05-02): MRG, FIX, VIZ: refactor plot_evoked_topomap
closes #7595 

Setup for testing:

```python
import numpy as np
import matplotlib.pyplot as plt
from mne.datasets import sample
from mne import read_evokeds

path = sample.data_path()
fname = path + '/MEG/sample/sample_audvis-ave.fif'
condition = 'Left Auditory'
evoked = read_evokeds(fname, condition=condition, baseline=(None, 0))
times = np.linspace(0, 0.5, 7)
```

# interactive usage

```python
evoked.plot_topomap(times='interactive')
```
### current master
[image]

### this PR
[image]

# interactive without colorbar

```python
evoked.plot_topomap( [... cut]
closed 2020-05-02 by larsoner