mne-tools/mne-python #7601: BUG: Extreme topomap valuse still too strong

larsoner, 2020-04-13:
@agramfort pointed out that the `examples/time_frequency/plot_source_power_spectrum_opm.py` example had edges that are too strong:

[image]

Doing `border='mean'` plus `extrapolate='local'` gives:

[image]

I wonder if:

1. The default extrapolation mode of `box` should actually add the local points, then also add the four points in the corners. It might be a good compromise.
2. `border='mean'` should be the default

If we did these two things, this is what the new default would look like (hacked together the box change for now and set `border='mean'`):

[image]

I know we've had some conversations about (2) at least, but I'd also like (1) because it might combine the local focality of `'local'` with the smoothness (bounded by matplotlib mask instead of interpolation grid extent) of `'box'`. Thoughts @mmagnuski ?

larsoner, 2020-04-13:
Also, using `np.median` rather than `np.mean` seems to work a tiny bit better to deal with the blob-like appearance at the edges of `local (or box+local) interp:

[image]

I tried weighting the points by inverse distance and it didn't seem to help much. But we can always improve this aspect of appearance later, the real questions are the two posed above.

agramfort, 2020-04-13:
is it possible to extrapolate a bit less far from sensors with "
border='mean' plus extrapolate='local'"
maybe divide by 2 the extra area beyond the convex hull?
I would consider changing then our defaults here.

>

larsoner, 2020-04-13:
Halving the distance fixes the blob problem, and also makes sense in terms of the "represented area" each channel gets, here shown for the new `'box'` mode (which does local then adds box; #7602)

[image]

I prefer this over pure `'local'` mode, which has jagged edges:

[image]

larsoner, 2020-04-13:
Actually there was a bug, this is what it looks like if you use half the distance for local mode (it's pretty tight):

[image]

And what it looks like for `box` on #7602: 

[image]

mmagnuski, 2020-04-13:
@larsoner 
The extrapolate='local' should have a corresponding mask, that's true.
What's the difference between the new box mode and the local?
If it's doing just what local does but just optimizes the circle mask placement, maybe `mask='circle'` or something like that would be better? Or maybe the distance in local should just be larger by default?

larsoner, 2020-04-13:
> What's the difference between the new box mode and the local?

`box` still extrapolates to a box. It just extrapolates to `local` first, then out to the box.

> The extrapolate='local' should have a corresponding mask, that's true.

I wonder if we can / should actually pass the outer edges from the triangulation. We have functions to figure out the vertices of mesh edges so it might not be so difficult. Then that could be passed to Matplotlib as a polygon mask.

If it does, maybe we kill the `box` changes I'm proposing and just let `local` be the default? In that case what we probably really want to do is keep interpolating to `distance`, but have the mask only extend `distance / 2.` beyond the sensors to keep uniform represented area.

agramfort, 2020-04-13:
local could use a clean polygon mask?

>

larsoner, 2020-04-13:
Yes, that's the idea if we continue extrapolating to 1x the median distance and mask to 0.5 the median distance I think it should work (if it still have chunks, we can extrapolate to 1.5 or 2x the median distance)

mmagnuski, 2020-04-13:
> I wonder if we can / should actually pass the outer edges from the triangulation. We have functions to figure out the vertices of mesh edges so it might not be so difficult. Then that could be passed to Matplotlib as a polygon mask.

Yes, something like this, was what I had in mind. The convex hull is computed in extrapolation, and then slightly stretched out to create local points - so the mask could be half that distance (the distance between channels and local points).

larsoner, 2020-04-13:
Okay I'll try making the `local` changes to use a polygon based on the convex hull.

One issue is that this will make the `outlines='head' | 'skirt'` stuff even worse, since we will no longer extrapolate far enough to even choose where the outlines will go.

mmagnuski, 2020-04-13:
> If it does, maybe we kill the box changes I'm proposing and just let local be the default? In that case what we probably really want to do is keep interpolating to distance, but have the mask only extend distance / 2. beyond the sensors to keep uniform represented area.

I suspect some (many?) users would still like to have circular mask. In other words: there are still cases where extrapolate='local' + mask='circle' would be useful. There could also be two masks: the circular one and the polygon mask (+ larger distance than 1 * median inter-channel?). 

> outlines='head' | 'skirt'

Do you see noticable differences between these two?

larsoner, 2020-04-13:
> I suspect some (many?) users would still like to have circular mask. In other words: there are still cases where extrapolate='local' + mask='circle' would be useful. There could also be two masks: the circular one and the polygon mask (+ larger distance than 1 * median inter-channel?).

This I don't really understand. Can you explain what you mean? The `extrapolate='local'` makes it so that the data shown extend to just past the (a slightly extended convex hull of) the electrodes. If you throw a circular mask on top of this, it's either a no-op (if it includes all electrodes) or cuts off some electrodes. In this sense there is an interaction between our extrapolation mode and the mask, in that they can both limit the visible data extent.

larsoner, 2020-04-13:
For head vs skirt so far I've been avoiding thinking about it, and don't have relevant experience messing with those. I'm just aware that it's an issue and our docs about it are wrong (#7397)

mmagnuski, 2020-04-13:
> This I don't really understand. Can you explain what you mean? The extrapolate='local' makes it so that the data shown extend to just past the (a slightly extended convex hull of) the electrodes. If [...]

Sure, my explanation was very unclear, sorry for that. It's just that circular mask will usually look nicer than convex hull based mask (its smoother). You can see in your own examples that some parts of the topos are circulary masked in the local mode already (on the left and right):
[image]

If the default distance at which extrapolation points are placed was higher - then the circular mask would mask not only sides but also the back in the example above. Then the convex hulll based mask would only be effectively masking the front. Of course it may be different for each case, so its not necessarily a general rule. But the circular mask would not cut off any channels (its created this way), and it is alread not a no-op.

larsoner, 2020-04-13:
> If the default distance at which extrapolation points are placed was higher - then the circular mask would mask not only sides but also the back in the example above. Then the convex hulll based [...]

Yes but having the top look bad is not great, and masking differently for one part (convex hull at the top) and another (circle at the bottom) is also not great.

> it's just that circular mask will usually look nicer than convex hull based mask (its smoother)

I just finished an implementation of the local at 1x + patch at 0.5x already looks pretty good and smooth. I'd prefer it to some combination of extra-circle-plus-convex-hull stuff:

[image]

If the slight angularity at the bottom really bothers people then I think we can probably fix it later, but it really does not bother me here. Basically if you use local extrapolation, I think you should only ever see data close to the channels, and we should do something to smooth the jagged edges. This seems like it would make a pretty nice default, as it:

1. Accurately allocates areas to channels
2. Accurately portrays the channel geometry
3. Does not have (big) issues at the boundaries

For EEG people that want data extrapolated and need circles, I don't think it's too much to have them set `extrapolate='box'`, and then we can make the skirt options do something again separately.

mmagnuski, 2020-04-13:
@larsoner - it looks better than I imagined, I like it the way it is. :)

larsoner, 2020-04-13:
Excellent, it was [pretty easy](https://github.com/mne-tools/mne-python/pull/7603) to implement and test because [someone already wrote nice convex hull / local code](https://github.com/mne-tools/mne-python/pull/5754) :)

PR #7603 (2020-04-21): MRG, BUG: Use auto extrap and mean borders
Probably a better solution than #7602.

WIP because:

- [x] viz examples/tutorials need to be updated
- [ ] [examples](https://19301-1301584-gh.circle-artifacts.com/0/dev/index.html) need to be checked

Closes #7601
Closes #7602
closed 2020-04-21 by larsoner