mne-tools/mne-python #7604: Can't set a list of epochs times for concatenation

rmib200, 2020-04-13:
I was trying to concatenate 2 epochs but an error appeared saying that both epochs should have the same times. I was trying to replace the array of times using "epochs_rest.times = epochs.times" but an error raised: AttributeError: can't set attribute. 
Somehow the same procedure worked here: https://www.kaggle.com/alexandrebarachant/beat-the-benchmark-0-67
Could this error be fixed?

agramfort, 2020-04-14:
do the epochs actually have different times? if so and you still want to
proceed using an EpochsArray seems the easiest.

>

hoechenberger, 2020-04-14:
I've never tried this, but you could try and see if `epochs_rest._set_times(epochs.times)` would work. However this is really a hack that could break anytime with a future release, as it's using an internal function.

hoechenberger, 2021-02-14:
Closing due to inactivity.

closed 2021-02-14 by hoechenberger