mne-tools/mne-python #7605: Could not load any valid backend / no '_Rendered' object with forward operator

DanielvdC, 2020-04-14:
#7330 ## Describe the bug
I'm trying to simulate EEG data, for which I need to compute the forward operator with `fsaverage`. This however produced the following error:

`RuntimeError: Could not load any valid 3D backend: ['mayavi', 'pyvista']`

I then successfully installed pyvista manually, but installing mayavi generated a large error log.
If I now try to compute the forward operator, I keep getting the following error:
`AttributeError: 'NoneType' object has no attribute '_Renderer'`

#### Steps to reproduce
I copied the [MNE tutorial](https://mne.tools/stable/auto_tutorials/source-modeling/plot_eeg_no_mri.html#sphx-glr-auto-tutorials-source-modeling-plot-eeg-no-mri-py), as such:
```
raw_fname, = eegbci.load_data(subject=1, runs=[6])
raw = mne.io.read_raw_edf(raw_fname, preload=True)

# Clean channel names to be able to use a standard 1005 montage
new_names = dict(
    (ch_name,
     ch_name.rstrip('.').upper().replace('Z', 'z').replace('FP', 'Fp'))
    for ch_name in raw.ch_names)
raw.rename_channels(new_names)

# Read and set the EEG electrode locations
montage = mne.channels.make_standard_montage('standard_1005')

raw.set_montage(montage)
raw.set_eeg_reference(projection=True)  # needed for inverse modeling

# Check that the locations of EEG electrodes is correct with respect to MRI
mne.viz.plot_alignment(
    raw.info, src=src, eeg=['original', 'projected'], trans=trans,
    show_axes=True, mri_fiducials=True, dig='fiducials')
```

#### Expected results
I was expecting a plot and a base to compute the forward operator.

#### Actual results
```
EDF file detected
Setting channel info structure...
Creating raw.info structure...
Reading 0 ... 19999  =      0.000 ...   124.994 secs...
Adding average EEG reference projection.
1 projection items deactivated
Average reference projection was added, but has not been applied yet. Use the apply_proj method to apply it.
Reading /Users/[USER]/mne_data/MNE-fsaverage-data/fsaverage/bem/fsaverage-ico-5-src.fif...
Using outer_skin.surf for head surface.
```
---------------------------------------------------------------------------
```AttributeError                            Traceback (most recent call last)
<ipython-input-30-c1141462bfd6> in <module>
     18 mne.viz.plot_alignment(
     19     raw.info, src=src, eeg=['original', 'projected'], trans=trans,
---> 20     show_axes=True, mri_fiducials=True, dig='fiducials')

<decorator-gen-128> in plot_alignment(info, trans, subject, subjects_dir, surfaces, coord_frame, meg, eeg, fwd, dig, ecog, src, mri_fiducials, bem, seeg, fnirs, show_axes, fig, interaction, verbose)

~/anaconda3/envs/opencv/lib/python3.7/site-packages/mne/viz/_3d.py in plot_alignment(***failed resolving arguments***)
   1034 
   1035     # initialize figure
-> 1036     renderer = _get_renderer(fig, bgcolor=(0.5, 0.5, 0.5), size=(800, 800))
   1037     if interaction == 'terrain':
   1038         renderer.set_interactive()

~/anaconda3/envs/opencv/lib/python3.7/site-packages/mne/viz/backends/renderer.py in _get_renderer(*args, **kwargs)
     31 def _get_renderer(*args, **kwargs):
     32     set_3d_backend(get_3d_backend(), verbose=False)
---> 33     return backend._Renderer(*args, **kwargs)
     34 
     35 

AttributeError: 'NoneType' object has no attribute '_Renderer'```

#### Additional information

Platform:      Darwin-19.4.0-x86_64-i386-64bit
Python:        3.7.7 (default, Mar 26 2020, 10:32:53)  [Clang 4.0.1 (tags/RELEASE_401/final)]
Executable:    /Users/[USER]/anaconda3/envs/opencv/bin/python
CPU:           i386: 4 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.20.0
numpy:         1.18.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.4.1
matplotlib:    3.2.1 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.22.1
numba:         Not found
nibabel:       Not found
cupy:          Not found
pandas:        1.0.3
dipy:          Not found
mayavi:        Not found
pyvista:       0.24.1
vtk:           8.1.2

GuillaumeFavelier, 2020-04-14:
Hello @DanielvdC and thanks for reporting this.

Indeed the 3d backend seems not to be loaded  or reloaded successfully. Let's first see if setting the backend dynamically helps with your issue. Could you add somewhere before the actual call to `plot_alignment`:

```py
...
raw.set_eeg_reference(projection=True)  # needed for inverse modeling

mne.viz.set_3d_backend("pyvista")  # <--- setting the backend here

# Check that the locations of EEG electrodes is correct with respect to MRI
mne.viz.plot_alignment(
    raw.info, src=src, eeg=['original', 'projected'], trans=trans,
    show_axes=True, mri_fiducials=True, dig='fiducials')
```

Please let me know how is the situation after this test.

It could also be helpful if you could share the result of:

`python -c "import pyvista; print(pyvista.Report())"`

DanielvdC, 2020-04-14:
Hi @GuillaumeFavelier,

`mne.viz.set_3d_backend("pyvista")  # <--- setting the backend here
`

This code snippet actually did the trick! It outputs the figure as it is supposed to. Thanks a lot for your help.

For what it's worth, this is the result of 
`python -c "import pyvista; print(pyvista.Report())"`

```Date: Tue Apr 14 15:57:59 2020 CEST

                OS : Darwin
            CPU(s) : 4
           Machine : x86_64
      Architecture : 64bit
       Environment : Python
        GPU Vendor : Intel Inc.
      GPU Renderer : Intel HD Graphics 5000 OpenGL Engine
       GPU Version : 4.1 INTEL-14.5.22

  Python 3.7.7 (default, Mar 26 2020, 10:32:53)  [Clang 4.0.1
  (tags/RELEASE_401/final)]

           pyvista : 0.24.1
               vtk : 8.1.2
             numpy : 1.18.1
           imageio : 2.8.0
           appdirs : 1.4.3
            scooby : 0.5.2
        matplotlib : 3.2.1
             PyQt5 : 5.14.2
           IPython : 7.13.0
             scipy : 1.4.1

  Intel(R) Math Kernel Library Version 2019.0.5 Product Build 20190808 for
  Intel(R) 64 architecture applications
```

agramfort, 2020-04-14:
shall we close then?

GuillaumeFavelier, 2020-04-14:
> This code snippet actually did the trick!

Glad to know that you have now a working solution :+1: 

> It outputs the figure as it is supposed to.

This shows that the backend was not loaded correctly. This operation is expected to be performed automatically. I can't reproduce this behaviour.

Can you confirm that your script works *only* with the code snippet? For example, even after restarting your working session after installing pyvista and run the script entirely?

DanielvdC, 2020-04-14:
After restarting my session the code actually works without the snippet as well. Perhaps the backend had to be set just once.

GuillaumeFavelier, 2020-04-14:
Thanks for the quick response.

>  Perhaps the backend had to be set just once.

I think so.

Feel free to close if everything works for you now :+1:

closed 2020-04-14 by DanielvdC