mne-tools/mne-python #7611: TimeViewer Play/Pause button should toggle state when tmax is reached

hoechenberger, 2020-04-16:
#### Describe the bug
Currently the Play/Pause button doesn't switch from Pause to Play when the the last time point has been reached.

#### Steps to reproduce
Open TimeViewer with an STC with multiple time points, press the play button, wait until the last time point is reached.

#### Expected results
Play/Pause button state should toggle.

#### Actual results
Play/Pause button does not toggle:
https://youtu.be/xnFgfJ1-s-M

#### Additional information
Instead of toggling automatically once tmax is reached, we could also simply loop until the user explicitly requests to pause playback?

hoechenberger, 2020-04-16:
Also needs a callback from time slider drag events.

GuillaumeFavelier, 2020-04-16:
> Also needs a callback from time slider drag events.

I am not sure to understand what you mean by that. Drag event for the time slider and matplotlib time line are 'active user event'. As opposed to the 'play/pause' button which indicates that the time points are updated automatically.

hoechenberger, 2020-04-16:
What meant is that we need to decide what happens in the following scenario:

- user has pressed the Play/Pause button
- movie is still playing
- user grabs the PyVista time slider and moves it along the tube
- should playback be paused?
- or should playback immediately continue once the slider drag event has been finalized?

larsoner, 2020-04-16:
I would say it should restart as soon as you let go of the slider, starting at the new slider location. This allows viewing the same activity over and over again just by clicking a point on the slider during playback, which is nice.

GuillaumeFavelier, 2020-04-16:
>  it should restart as soon as you let go of the slider, starting at the new slider location. 

I believe this is the current behaviour right?

[image]

larsoner, 2020-04-16:
It was last time I tested and might indeed still be true. I was just responding to @hoechenberger's question above:

> - should playback be paused?
> - or should playback immediately continue once the slider drag event has been finalized?

hoechenberger, 2020-04-16:
> It was last time I tested and might indeed still be true. I was just responding to @hoechenberger's question above:

Yes, it's still working that way! :)

closed 2020-04-16 by hoechenberger