mne-tools/mne-python #7622: mne.set_config doesn't accept Path objects

hoechenberger, 2020-04-17:
#### Describe the bug
Passing a `pathlib.Path` value to `mne.set_config()` raises an error.

#### Steps to reproduce
```python
import pathlib
import mne

path = pathlib.Path('/tmp')
mne.set_config(key='SUBJECTS_DIR', value=path)
```
#### Expected results
Configuration would be updated.

#### Actual results
```python
TypeError                                 Traceback (most recent call last)
Untitled-1 in 
      4 
      5 path = pathlib.Path('/tmp')
----> 6 mne.set_config(key='SUBJECTS_DIR', value=path)

~/Development/mne-python/mne/utils/config.py in set_config(key, value, home_dir, set_env)
    264     # While JSON allow non-string types, we allow users to override config
    265     # settings using env, which are strings, so we enforce that here
--> 266     _validate_type(value, (str, type(None)), "value",
    267                    "None or string")
    268 

~/Development/mne-python/mne/utils/check.py in _validate_type(item, types, item_name, type_name)
    377                 type_name[-1] = 'or ' + type_name[-1]
    378                 type_name = ', '.join(type_name)
--> 379         raise TypeError('%s must be an instance of %s, got %s instead'
    380                         % (item_name, type_name, type(item),))
    381 

TypeError: value must be an instance of None or string, got  instead
```

#### Additional information
As we support `Path` objects in other places in MNE, `set_config()` should not be an exception.

cbrnr, 2020-04-17:
👍 - I'd also ask why we need to restrict the types here. Somehow the data must be serialized as a string to end up in the JSON file, so I wouldn't check anything. Otherwise, we might have to add more and more types that have a suitable string representation to this check.

hoechenberger, 2020-04-17:
> Otherwise, we might have to add more and more types that have a suitable string representation to this check.

We should probably simply cast to a string of `not None`, but I suppose that's what you meant, right?

cbrnr, 2020-04-17:
> We should probably simply cast to a string if not None, but I suppose that's what you meant, right?

Basically yes.

larsoner, 2020-04-17:
They all have to be strong because we also support overriding with environment variables, so yes just cast

PR #7624 (2020-04-17): ENH: Allow path-like values passed to set_config
#### Reference issue
Fixes #7622

#### What does this implement/fix?
Allows to pass<strike> arbitrary</strike> `path-like` values to `set_config` <strike>as long as they're `None` or can be cast into a string. No further type checking is performed.</strike>

#### Additional information
<strike>Also this PR includes a small fix to the `config` test suite, as one of the tests was expecting a `TypeError`, while in fact a `RuntimeWarning` is raised.</strike>
closed 2020-04-17 by larsoner