mne-tools/mne-python #7631: 'No data channels found' with plot_psd() on misc channels

gpeyton, 2020-04-18:
**Issue:**
Error of "No Data channels found" when attempting to **plot_psd()** on **misc** channels in a dataset which contains only **misc** channels. The dataset is created from raw numpy EEG data derived from a custom made device.

**Steps:**
Create mne object from raw data
```
mne_data = create_mne_object(channels, data, samp_freq)
print(mne_data)
print(mne_data.info)
```

The output is shown below. 
```
<RawArray | 2 x 1024 (4.1 s), ~25 kB, data loaded>
<Info | 7 non-empty values
 bads: []
 ch_names: CH1, CH2
 chs: 2 MISC
 custom_ref_applied: False
 highpass: 0.0 Hz
 lowpass: 125.0 Hz
 meas_date: unspecified
 nchan: 2
 projs: []
 sfreq: 250.0 Hz
>
```
Time-domain data does plot correctly. However, when attempting to plot psd:
```
picks = mne.pick_types(mne_data.info, misc=True)
print(picks)
spect = mne_data.plot_psd(picks=picks)
```
Output:
```
[0 1]
RuntimeError: No data channels found
```

**Summary**
This issue seems similar to the bug [here](https://github.com/mne-tools/mne-python/issues/7154). Is there a workaround at this point? I have tried passing **picks=None** without success.

agramfort, 2020-04-19:
as you can see you have:

chs: 2 MISC

and MISC are not consider data channels. If you pass picks then I
think its fair to expect it to work.

if it does not please provide a full gist to replicate the pb. From
creation of data to plotting.

thanks

gpeyton, 2020-04-19:
The following code replicates the problem:

```
import numpy as np
import mne

fs = 250
channels = ['CH1', 'CH2']
data = 1e-3*np.random.rand(2, 1024)
print(data.shape)

info = mne.create_info(channels, fs)
mne_data = mne.io.RawArray(data, info)

print(mne_data.info)
plt = mne_data.plot()
picks = mne.pick_types(mne_data.info, misc=True)
print(picks)
spect = mne_data.plot_psd(picks=picks)
```

thanks

agramfort, 2020-04-19:
I can replicate the pb.

@drammock what's your feeling about this?

I am enclined to fix the behavior. If picks is None then default to what we call data channels but if you pass picks then plot whatever is requested.

drammock, 2020-04-19:
agreed, seems to me like if the user passes `picks` we should give them what they ask for.

gpeyton, 2020-04-19:
When `picks = None`, I get:
`picks (NoneNone, treated as "data") yielded no channels, consider passing picks explicitly`

agramfort, 2020-04-19:
yes we need to fix ...

>

gpeyton, 2020-04-19:
Is there a temporary workaround for this in the meantime? How would I change it to a different type?

drammock, 2020-04-19:
@gpeyton you can change the channel type with: 
https://mne.tools/dev/generated/mne.io.Raw.html#mne.io.Raw.set_channel_types

larsoner, 2020-04-19:
@drammock do you want to work on the fix? If not, I can tackle it this week

PR #7668 (2020-04-23): MRG, FIX: allow explicit non-data picks
closes #7631 

New test fails on master and passes on this PR
closed 2020-04-23 by larsoner