mne-tools/mne-python #7632: Improve documentation for set_montage to inform about usage of np.nan for coordinates

adam2392, 2020-04-20:
# Problem
The documentation for `raw.set_montage` is not very clear to inform the user that we can use `np.nan` as coordinates to get around the ValueError that is raised when there are channels w/o coordinates set in the montage. 

# Proposed Solution
This would be a short PR to adapt the documentation here:

```
    def set_montage(self, montage, match_case=True, verbose=None):
        """Set EEG sensor configuration and head digitization.

        Parameters
        ----------
        %(montage)s
        %(match_case)s
        %(verbose_meth)s

        Returns
        -------
        inst : instance of Raw | Epochs | Evoked
            The instance.

        Notes
        -----
        Operates in place.
        """
```

to add documentation to the `Notes` section saying the accepted values e.g. `np.nan` and `np.inf` if you want to set a montage, but not for specific electrodes. 

See related discussion in https://github.com/mne-tools/mne-bids/pull/393

cbrnr, 2020-04-20:
I don't quite understand. Isn't the new `on_missing` argument turning off the error? I would be confused if the documentation mentioned `np.nan` and `np.inf` for coordinates - why and how would anyone want to manually set some channel coordinates to these values?

agramfort, 2020-04-20:
to me only mne-bids would be doing this internally. End users should not
have to know about this

>

larsoner, 2020-04-27:
Sounds like we don't need this in MNE so I'll close, feel free to reopen if need be @adam2392

closed 2020-04-27 by larsoner