mne-tools/mne-python #7648: ENH: Add volumetric rendering to _Brain

larsoner, 2020-04-21:
It would be nice to make MixedSourceEstimates and VolSourceEstimates plottable with _Brain instead of just surfaces.

Random ideas:

- Slicer is probably a good resource for ideas, and [context preserving](https://www.slicer.org/wiki/Modules:VolumeRendering-Documentation-3.5) looks cool
- [VTK example](https://lorensen.github.io/VTKExamples/site/Python/Medical/MedicalDemo4/) produces:

    [image]

- A random [ray-tracing lib, MIT license](https://github.com/Gjacquenot/PyRayTracer)
- [A recent issue about it](https://discourse.vtk.org/t/volume-rendering-with-vtk-cpu-based-ok-gpu-based-not-visible/475/8)
- [5-year-old post from VTK folks](https://blog.kitware.com/volume-rendering-improvements-in-vtk/)
- [5-year-old post, using Python](https://pyscience.wordpress.com/2014/11/16/volume-rendering-with-python-and-vtk/)

Alternatively we could go with isocontours, but seems harder to control/configure

agramfort, 2020-04-21:
#nerdsniping on the way...

>

GuillaumeFavelier, 2020-04-24:
Another useful source?

https://docs.pyvista.org/examples/02-plot/volume.html?highlight=volume

larsoner, 2020-04-24:
Yes much more useful!

PR #8064 (2020-08-05): MRG, ENH: Volume rendering
Closes #7648

- [x] Vector initialization
- [x] Vector updates with time-clicking and clims
- [x] Volume initialization (should be unified / not rely on vector-ness, just happen in `set_time_point` like everything else)
- [x] Volume updates (moving scalar setting to `set_time_point` should fix this)
- [x] Volume updates with time-clicking
- [x] Actually make it look good
- [x] Make it work for non-vector `stc.plot`
- [x] Volume-only plotting
- [x] Fix bug with pos_lims for volumetric plotting
- [x] Fix bug with volumetric plotting on macOS + VTK9 (https://github.com/ [... cut]
closed 2020-08-05 by larsoner