mne-tools/mne-python #7652: get_montage and get_coordinates for Raw and Montage objects

adam2392, 2020-04-21:
#### Describe the problem
Once a user has set the montage, then getting the montage itself and data from it is not trivial.

Ideally, the users can get the actual xyz coordinates whether its an array, or dictionary via just a `get` function.

You have to do things like:
1. `ch_locs = raw.info['chs']['loc'][0:3]`
or
2. `montage = raw.get_montage(); ch_locs = montage.get_coordinates()`

In 1., it took me a bit to end up figuring out that the 12 point locations in are not relevant for the most part if I need the xyz coordinates. And the xyz are stored in the first 3 indices. This is at least for iEEG.

#### Describe your solution
An implementation at the Raw level for `get_montage()`, it simply returns the montage that is set for a Raw object, or returns an error message if no montage.

An implementation at the Montage level for `get_coordinates()`, which returns the xyz coordinates in it's 'm' form as a 3D array. 

#### Describe possible alternatives
`get_coordinates` could also be at the Raw level?

#### Additional context
See related discussion: https://github.com/mne-tools/mne-bids/pull/393

Happy to submit a PR after discussion of if this is valid.

larsoner, 2020-04-21:
I would not do `get_montage`, a `montage` is just a way to set channel positions.

I would add a `inst.get_channel_positions()` that for example returned a `dict` of the channel name: pos mapping. It would really just loop over `info['chs']` to look at `loc` but indeed a user should not have to do this.

agramfort, 2020-04-21:
Get_montage that returns a dig montage that could be fed to set_montage is to me the most consistent. I would like to get all dig points etc

larsoner, 2020-04-21:
Okay, fair enough

drammock, 2020-04-21:
I'd also note that `get_ch_positions` (rather than `get_channel_positions`) is more consistent with other methods like `inst.get_ch_types()`.  So if you end up doing both `get_montage` and `get_xxx_positions` then I think `xxx` should be `ch`, not `channel`.

cbrnr, 2020-04-22:
Although it would be useful if we could extract the name of the original montage somehow (maybe just for the standard montages shipped with MNE).

agramfort, 2020-04-22:
if you want provenance of the original montage file it's really tricky as
it means changing info
data structures.

>

cbrnr, 2020-04-22:
We could add a new info field, which contains the montage name every time the user does `raw.set_montage`. Of course, this does not mean that the name and the coordinates match (because the coordinates might have been changed after setting the montage), but something like `info["montage_name"]` would be nice to have. Maybe the value of this field could even be reset to `None` if someone tinkers with the coords, but I don't know if we have setters in place.

Anyway, if this is not feasible it's not that important.

agramfort, 2020-04-22:
I don't think it's easily feasible

>

adam2392, 2020-04-22:
Okay so:

- `inst.get_ch_positions()` -> dictionary of channels and their xyz positions. This is pretty clear to me.

> Get_montage that returns a dig montage that could be fed to set_montage is to me the most consistent. I would like to get all dig points etc

@agramfort what do you mean here? Implement a mne.get_montage() function, or an inst.get_montage()?

agramfort, 2020-04-22:
Inst.get_montage()

closed 2020-09-08 by larsoner

pchr-srf, 2024-03-24:
Sorry to dig up this PR, but what is `Inst` in your code snippet? 😅

cbrnr, 2024-03-24:
This typically refers to `Raw`, `Epochs`, or `Evoked` (depending on the context).

pchr-srf, 2024-03-24:
Ah, super quick! Yes, I just found that out, too. Thanks! Helping my partner with some analysis and super confused by everything 😅

PR #7667 (2020-09-08): [MRG] Adding get_montage for montage to BaseRaw objects
#### Reference issue
Fixes: #7652 

#### What does this implement/fix?
Adds `get_montage()` and `get_ch_positions` for `ContainsMixin` mixin class, which can be used by all instances to get the montage or channel positions.