mne-tools/mne-python #7655: 'BAD boundary' in epoch drop_log

zobm, 2020-04-22:
Hi MNE Devs, 

I am working with resting state MEG data. The data are to be used in for the spectral connectivity function. 

Since, I believe that function requires epoched data, I am epoching the resting state data by first using the "make_fixed_length_events" function and then calling the "Epochs" function. 

The drop log shows that a quarter to a third of my epochs are being dropped with the flag being 'BAD boundary'. Why would this be happening?

Thanks so much!

larsoner, 2020-04-22:
`BAD boundary` annotations should only occur when you have stopped and restarted a recording, and should be visible with `raw.plot()`. Are the epochs you're creating extremely long, and thus crossing these boundaries?

zobm, 2020-04-22:
Thanks for the fast response. I realize the issue now. The epochs are 10s in length. The scan in question had large environmental noise and discrete time points and these segments of data were removed. The 10s epochs cross these boundaries.

closed 2020-04-22 by zobm