mne-tools/mne-python #7666: installation fails on CircleCI

drammock, 2020-04-23:
```
ModuleNotFoundError: No module named 'tvtk.custom.generic_render_window_interactor'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/circleci/.local/lib/python3.7/site-packages/mayavi/mlab.py", line 16, in <module>
    from mayavi.tools.camera import view, roll, yaw, pitch, move
  File "/home/circleci/.local/lib/python3.7/site-packages/mayavi/tools/camera.py", line 24, in <module>
    from .engine_manager import get_engine
  File "/home/circleci/.local/lib/python3.7/site-packages/mayavi/tools/engine_manager.py", line 14, in <module>
    from mayavi.core.engine import Engine
  File "/home/circleci/.local/lib/python3.7/site-packages/mayavi/core/engine.py", line 28, in <module>
    from mayavi.core.base import Base
  File "/home/circleci/.local/lib/python3.7/site-packages/mayavi/core/base.py", line 19, in <module>
    from tvtk.pyface.tvtk_scene import TVTKScene
  File "/home/circleci/.local/lib/python3.7/site-packages/tvtk/pyface/tvtk_scene.py", line 44, in <module>
    class TVTKScene(HasPrivateTraits):
  File "/home/circleci/.local/lib/python3.7/site-packages/tvtk/pyface/tvtk_scene.py", line 175, in TVTKScene
    interactor = Property(Instance(tvtk.GenericRenderWindowInteractor))
  File "tvtk_classes/tvtk_helper.py", line 481, in <lambda>
  File "tvtk_classes/tvtk_helper.py", line 47, in get_class
  File "tvtk_classes/tvtk_helper.py", line 28, in get_module
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "tvtk_classes/generic_render_window_interactor.py", line 29, in <module>
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "tvtk_classes/render_window_interactor.py", line 32, in <module>
  File "tvtk_classes/render_window_interactor.py", line 446, in RenderWindowInteractor
NameError: name 'nan' is not defined
```

@GuillaumeFavelier do you know what caused this?

larsoner, 2020-04-23:
Looks suspiciously like problems I got when using PyQt5 < 5.14 on Linux with VTK9. Hopefully it's spurious...?

Can you link to the CircleCI job? I looked at the last status commit there and it was fine

GuillaumeFavelier, 2020-04-23:
I never got this one @drammock. The closest I can recall is the log @hoechenberger had some time ago:

```
========================================================== ERRORS ==========================================================
_________________________________ ERROR collecting mne/viz/backends/tests/test_renderer.py _________________________________
/Users/hoechenberger/miniconda3/envs/mne-dev-new/lib/python3.8/site-packages/tvtk/tvtk_classes.zip/tvtk_classes/tvtk_helper.py:22: in get_module
    mod = __import__('tvtk.custom.%s'%fname,
E   ModuleNotFoundError: No module named 'tvtk.custom.renderer'
        fname      = 'renderer'During handling of the above exception, another exception occurred:
test_renderer.py:13: in <module>
    from mne.viz.backends.tests._utils import (skips_if_not_mayavi,
        __builtins__ = <builtins>
        __cached__ = '/Users/hoechenberger/Development/mne-python/mne/viz/backends/tests/__pycache__/test_renderer.cpython-38.pyc'
```
which was fixed by `pip`-installing `mayavi`.

I agree with @larsoner, it looks related somehow to https://github.com/pyvista/pyvista/pull/686#issue-403215492

drammock, 2020-04-23:
Circle pipeline 1213

https://circleci.com/gh/mne-tools/mne-python/19602

larsoner, 2020-04-23:
We've had green ones since then, so maybe it's some strange spurious thing...

larsoner, 2020-04-27:
Looks like it was spurious so I'll close

closed 2020-04-27 by larsoner