mne-tools/mne-python #7669: IndexError when calling pick_channels

nschawor, 2020-04-23:
#### Describe the bug
I attempted to resave a dataset which was provided in a strange format (one file for subsets of channels), and encountered some strange behavior with the ```pick_channels``` function. I remember that this worked in mne version 0.19.2, but it does not anymore in 0.21.dev0. 

#### Steps to reproduce
```python
import mne 
import numpy as np

nr_datasets = 20
nr_channels = 10
nr_samples = 20000
fs = 1000

raws = []
for i_set in range(nr_datasets):
    ch_names = ['eeg_%i_%i' %(i_set, i) for i in range(nr_channels)]
    data = np.random.randn(nr_channels, nr_samples)
    info = mne.create_info(ch_names, fs, 'eeg')
    raw = mne.io.RawArray(data, info)
    raws.append(raw)

raw1 = raws[0].add_channels(raws[1:])
raw1.pick_channels(raw.ch_names)
```

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Expected results
A raw instance with a subset of channels should be returned.

#### Actual results
```
/home/n/.local/lib/python3.7/site-packages/mne/channels/channels.py in <listcomp>(.0)
    859             print(self._read_picks)
    860             print(idx)
--> 861             self._read_picks = [r[idx] for r in self._read_picks]
    862 
    863         if hasattr(self, '_cals'):

IndexError: index 190 is out of bounds for axis 0 with size 29
```

agramfort, 2020-04-23:
can you share the file?

does it work with 0.20.1 ?

larsoner, 2020-04-23:
Probably due to #7507 so I would expect this to work on 0.20 and 0.20.1. Indeed having the problematic file would help

nschawor, 2020-04-23:
The above minimal example should execute, since it is with some random synthetic data and should also throw an error with 0.21.dev0. It works with v.0.20.1.

agramfort, 2020-04-23:
sorry I read too quickly

I confirm it fails with current master

>

PR #7670 (2020-04-24): MRG, FIX: Fix read lims
Closes #7669.

@nschawor this allows your code snippet to run for me. Can you try it?

No need for `latest.inc` update as it's a within-release fix.
closed 2020-04-24 by agramfort