mne-tools/mne-python #7678: Add line width parameter to plot_filter

rob-luke, 2020-04-24:
#### Describe the problem
The line width in `plot_filter` is quite thick, when overlaid with other lines it looks very dominant. For an example see this [WIP](https://mne.tools/mne-nirs/auto_examples/plot_30_frequency_summary.html#discussion).

#### Describe your solution
Can I add a `linewidth` argument to the `plot_filter` function? I would make it take a single value or a list with as many values as requested properties.

#### Describe possible alternatives
Leave as is. Change default (I don't like this plan as the filtering tutorials look nice as now).

#### Additional context
This is obviously a completely subjective assessment.

larsoner, 2020-04-25:
> Describe possible alternatives

Have you considered modifying `fig.axes[ii].lines` after `fig` is returned?

rob-luke, 2020-04-25:
Yes, sorry should have listed that in alternatives.

This is a totally subjective issue, so its fine if you say this is the style that MNE has agreed on and we want it left as is. Its just that my preference is for line widths of 1 and single argument changes rather than modifying `axes.lines`.

So I can leave as is if you prefer. But also happy to make the code changes and submit a PR.

larsoner, 2020-04-25:
Generally we try not to expose all the mpl parameters in our API but instead let users change them after the fact. So I'd rather not add an argument, but I'm open to changing the default so it looks better and/or makes the function more broadly applicable

larsoner, 2020-04-25:
(usually thick lines are not a good choice, so it's entirely possible something better could be done. If you go this route please touch the plot_background_filtering.py tutorial since that's where the changes will be most obvious)

hoechenberger, 2020-05-08:
@larsoner 
> I'm open to changing the default so it looks better and/or makes the function more broadly applicable

+1 on making the default less thick

rob-luke, 2020-05-10:
Ok. I’ll open a PR and then we can see what it looks like.

teanijarv, 2022-11-19:
Would be nice to change the linewidth with the plot_filter function as if I use the rcParams then time-domain linewidth changes, but not frequency-domain or delay plots because they are being overwritten in the function. Currently I created a custom function from MNE plot_filter function for myself, but I'm following this thread to see if the function gets updated.

larsoner, 2022-11-19:
@teanijarv would you be up for making a PR to fix our internal `linewidth`? Basically tweak the `linewidth` we use internally until you are aesthetically happy with it, and we can look at it and merge if things look better!

FWIW [as above](https://github.com/mne-tools/mne-python/issues/7678#issuecomment-619298438) I don't think we should add a `line_width` property as in the top comment, but rather just change our internal default to be better. People who want to modify after the fact can pretty easily modify `fig.axes.lines` directly afterward.

teanijarv, 2022-11-20:
@larsoner I added the minor edits. Hope you are happy with it! MNE is such an amazing package, hopefully can help some time again! :)

larsoner, 2023-10-01:
Looks like this was fixed, closing

closed 2023-10-01 by larsoner