mne-tools/mne-python #7680: Unable to read EDF file

tayyabafaisal, 2020-04-26:
Hello
I'm working on an EEG signal but even after trying several ways, I'm unable to read the dataset.
There is an error  of null values in the header while reading with mne
I have around 200 edf files in the dataset all have the same issue. Can you please identify the issue.

cbrnr, 2020-04-26:
Can you share one of the problematic files with us?

tayyabafaisal, 2020-04-26:
Zipped two files.Please check.Hope you can figure out
Thankyou 
[edffiles.zip](https://github.com/mne-tools/mne-python/files/4535089/edffiles.zip)

cbrnr, 2020-04-26:
Both files work for me with the latest 0.20.1 release - please update MNE and try again.

tayyabafaisal, 2020-04-27:
Thankyou for your reply 
Im using same version but get this error (check attached)
My script is
import mne
f='0000002.edf'
edf_file = mne.io.read_raw_edf(f, preload = True)

If you can please check my script 
Thankyou
[image]
[image]
[image]

[image]

cbrnr, 2020-04-27:
Got it, you're using the current dev version and I can reproduce the error with the current master. I'll try to figure out what change introduced the regression.

cbrnr, 2020-04-27:
Wait, I made a stupid mistake. In fact, current master still works and correctly loads your example file. Same for 0.20.1. I guess you are referring to the issue reported and fixed in #7641. Prior to 0.20.1, I can indeed reproduce the error.

You are on an outdated master branch. Please either pull the latest changes from `upstream` (your remote might also be called `origin` depending on how you cloned the repo):

```
git pull upstream
```

However, I'd recommend to work with a release version which you can install with `pip`:

```
pip install -U mne
```

larsoner, 2020-04-27:
Thanks for investigating and explaining @cbrnr

closed 2020-04-27 by larsoner

tayyabafaisal, 2020-04-28:
@cbrnr can you please share script with me