mne-tools/mne-python #7683: viz.get_3d_backend() should return None if no 3d backend is found, instead of raising an exeption

hoechenberger, 2020-04-27:
#### Describe the problem
Currently, `mne.viz.get_3d_backend()` will raise a `RuntimeError` if no usable backend has been found:

https://github.com/mne-tools/mne-python/blob/d0b8bcf5201d7b5c278e7410e4035dc5aead30b8/mne/viz/backends/renderer.py#L119-L147

#### Describe your solution
I believe it should return `None` instead: if there is currently no 3d backend, then, well, there is none ;)

#### Describe possible alternatives
I currently have to be ready to catch a `RuntimeError` when I simple want to query whether I can produce a 3d visualization or not. Background: I'm plotting STCs in environments that sometimes don't have working `pyvista` or `mayavi` installations. I want to find out when that's the case, and switch to using matplotlib instead.

#### Additional context
I checked how `get_3d_backend()` is currently used, and without exception, any use is a comparison like if get_3d_backend() != 'pyvista':`; therefore, returning `None` if no working backend was found should not break any existing code.

cc @GuillaumeFavelier

agramfort, 2020-04-27:
wdyt @GuillaumeFavelier ?

GuillaumeFavelier, 2020-04-28:
`get_3d_backend()` was not designed for such a use. All the functions of `mne/viz/backends/renderer` expect it to return a valid 3d backend. So it's possible to make it return `None` instead of raising an error but I can't evaluate the impact of this change to the rest of those functions, which are used extensively in `_3d.py`. What would be the default behaviour of those functions if no 3d backend is found?

If the goal is to check if you can use a 3d backend, I could refactor a function that tests just that and return `None` otherwise. What do you think?

hoechenberger, 2020-04-28:
> `get_3d_backend()` was not designed for such a use. All the functions of `mne/viz/backends/renderer` expect it to return a valid 3d backend. So it's possible to make it return `None` instead of [...]

Yes I was concerned about this too, bc then these functions all would have to check if `None` was returned or not, which seems cumbersome. However like I said, I've searched the sources for calls of this function, and there's always a comparison like `if get_3d_backend() == 'pyvista'`, so it should be safe to make such a change, but …

> 
> If the goal is to check if you can use a 3d backend, I could refactor a function that tests just that and return `None` otherwise. What do you think?

… this sounds like a cleaner approach. What would the function be called? `have_3d_backend()`?

GuillaumeFavelier, 2020-04-28:
>  there's always a comparison like 

Not always, you can look in `_get_renderer()` which is used in every function of `_3d.py` and `_use_test_3d_backend()` which is used indirectly in `test_3d.py` through the `renderer` fixture.

> What would the function be called?

`have_3d_backend()` sounds good to me. So this function would return `True` is there is a valid 3d backend and `False` otherwise, right?

hoechenberger, 2020-04-28:
> `have_3d_backend()` sounds good to me. So this function would return `True` is there is a valid 3d backend and `False` otherwise, right?

That would be the idea, yes :)

GuillaumeFavelier, 2020-04-28:
I'll open a PR to take care of it then :)

agramfort, 2020-04-28:
ok for have_3d_backend function

>

closed 2020-04-30 by larsoner