mne-tools/mne-python #7691: Coordinate Frames of Montage Converted from Unknown to Head Automatically?

adam2392, 2020-04-28:
#### Describe the bug
When I pass in a montage with `coord_frame = 'unknown'', and check the actual montage, it is all of a sudden in the head coordinate. I believe it is due to this line:

mne-python/mne/channels/montage.py:664

> mnt_head = _get_montage_in_head(montage)

Inside function: mne.channels.montage._set_montage. As a result, raw.set_montage(montage) now has all coordinate frames in head frame. 

#### Steps to reproduce
```
def _make_toy_raw(n_channels):
    return RawArray(
        data=np.empty([n_channels, 1]),
        info=create_info(
            ch_names=list(ascii_lowercase[:n_channels]),
            sfreq=1, ch_types='eeg'
        )
    )

  raw = _make_toy_raw(4)
    raw.set_channel_types({ch: 'ecog' for ch in raw.ch_names})
    ch_names = raw.ch_names
    n_channels = len(ch_names)
    ch_coords = np.arange(n_channels * 3).reshape(n_channels, 3)

    # create mne.DigMontage
    ch_pos = dict(zip(ch_names_raw, ch_locs))
    montage = mne.channels.make_dig_montage(ch_pos=ch_pos, coord_frame='unknown')
raw.set_montage(montage)

print(raw.info['dig'])
```

#### Expected results
A simple fix I presume would be to add a `compute_head = True` to the `set_montage` function, so that way you can turn it on or off?

#### Additional information
See: https://github.com/mne-tools/mne-bids/pull/390#discussion_r416942332 for related problems when trying to use mne-python to read/write data.

adam2392, 2020-04-28:
Actually for that matter, I'm also having a really hard time making the montages agree after `set_montage` and `get_montage` in this PR: https://github.com/mne-tools/mne-python/pull/7667

I believe the only change is this `_get_montage_in_head()` transformation that is applied.

hoechenberger, 2020-04-29:
> I believe it is due to this line:
> 
> mne-python/mne/channels/montage.py:664
> 
> > mnt_head = _get_montage_in_head(montage)
> 
> Inside function: mne.channels.montage._set_montage.

Hey @adam2392, this can make your life easier:
https://help.github.com/en/github/managing-your-work-on-github/creating-a-permanent-link-to-a-code-snippet

Also you can get syntax highlighting in code blocks if you start them with
"```python" :)

agramfort, 2020-04-29:
@larsoner we need to discuss this again.

chs loc "must" be in head according to MNE convention
but are we currently too strict with dig?

>

larsoner, 2020-04-29:
To me dig does *not* require all points in head coords, correct.

`ch['loc']` we could in principle even relax because there is `ch['coord_frame']`, but it would require a lot of careful changes to get right (e.g., in forward calculation).

adam2392, 2020-04-29:
Hmm seems that the transformation would definitely impact EEG and MEG. Is it possible to at least as a simple fix, add a flag `transform_head = True`  or something to just allow iEEG data setting montage to get rid of this transformation?

agramfort, 2020-04-29:
>
> To me dig does *not* require all points in head coords, correct.
>
> ch['loc'] we could in principle even relax because there is
> ch['coord_frame'], but it would require a lot of careful changes to get
> right (e.g., in forward calculation).
>

i feel we should consider this as currenlty sEEG/ECoG folks will read in MRI
fake it head and than silently use trans Head<->MRI as Identity to get it
to work.

thoughts?

jasmainak, 2020-04-29:
could we allow `ch['loc']` to be not `'head'` based on the channel type? Because there is no forward calculation to be done for ECoG data (yet) in MNE. Then you avoid complexity and only check this when setting.

larsoner, 2020-04-30:
> i feel we should consider this as currenlty sEEG/ECoG folks will read in MRI fake it head and than silently use trans Head<->MRI as Identity to get it to work.

The question in my mind is, what should these people see for example with a topomap plot? Do we allow them to be silently in MRI coords when they do their plot? It breaks the uniformity of our topomap transformation, which might be worth it but we need to consider it.

> Because there is no forward calculation to be done for ECoG data (yet) in MNE. Then you avoid complexity and only check this when setting.

If you're going to allow non-head for some sensor types (sEEG, ECoG), it's unclear why you wouldn't allow it in general, e.g., for EEG electrodes. Then you need to decide how to handle those cases in forward calculation (which IIRC is always done in head coords).

And unfortunately it's more than just the forward we need to consider. Any function that plots sensor locations or uses sensor locations needs to be checked that things are in the correct coordinate frame or that we aren't (still) implicitly assuming a head coordinate frame. So at a minimum the big things I foresee being problems are:

- `plot_alignment` -- need to be more careful about coordinate frames (actually easiest to fix)
- `topomap` -- do we allow non-head coords? do we use the same transformation we do for head? do will still draw a head outline?
- `forward` -- do we allow computing in the MRI frame?
- `DigMontage` -- do we still require fiducials and/or being in the head coordinate frame, or do we allow people to work with `FIFFV_COORD_UNKNOWN` for EEG?

jasmainak, 2020-04-30:
> If you're going to allow non-head for some sensor types (sEEG, ECoG), it's unclear why you wouldn't allow it in general, e.g., for EEG electrodes.

I am not an intracranial expert but I think it's not kosher in the sEEG/ECoG world to deal with a head space. You have a CT scan with the device coordinates and you transform them to MRI coordinates by aligning with the anatomical MRI so that everything is in MR coordinates.

I agree that supporting a full pipeline for ECoG is going to be challenging. I understood that we needed this only for MNE-BIDS to be able to help convert existing data to BIDS format @agramfort @adam2392 ?

larsoner, 2020-04-30:
> You have a CT scan with the device coordinates and you transform them to MRI coordinates by aligning with the anatomical MRI so that everything is in MR coordinates.

Something similar can happen with EEG+fMRI, so I think we have to think more broadly than special-casing sEEG and ECoG

adam2392, 2020-04-30:
> I am not an intracranial expert but I think it's not kosher in the sEEG/ECoG world to deal with a head space. You have a CT scan with the device coordinates and you transform them to MRI coordinates [...]

Idk what a full pipeline would look like for ECoG/SEEG in mne, but honestly, I'm okay with just coordinate frame support when `set_montage` is called. Transforming to head seems weird to me. Just having a unified data structure that doesn't get arbitrarily modified/transformed would be a good start. E.g. you create a digMontage from coordinates and you know its in a specific T1 MRI coordinate system, then I think your downstream analysis can happen outside of  mne (for now?)

Because of that, to maintain backwards compatibility, why can't we just add a `transform_to_head=True` parameter to `set_montage`? This then lets iEEG users just turn it off and stick w/ the mne supported "ras", "mri", "mri_voxel", "fs_tal", "mni_tal" coord systems. If iEEG users want to use the above supported pipelines to do some visualizations, just document that currently you need to "transform_to_head"?

larsoner, 2023-10-01:
I think we have addressed this in MNE-BIDS and MNE-Python but @adam2392 feel free to reopen if I'm mistaken

closed 2023-10-01 by larsoner