mne-tools/mne-python #7694: Adding new dataset (ground-truth for source localization)

ezemikulan, 2020-04-29:
Hi all, we have just published a [dataset](https://www.nature.com/articles/s41597-020-0467-x) that contains simultaneous intracerebral stimulation and HD-EEG. I think it would be great to add it to the datasets in MNE. What do you think?

agramfort, 2020-04-29:
that would be awesome !

you know where to start @ezemikulan ?

ezemikulan, 2020-04-29:
Great. I've been looking a bit at the datasets module and in particular  at the mne.datasets.utils._data_path function. I think I can have a closer look and let you know if I have questions (I need to finish a couple of things before being able to work on this). In any case, if you have specific instructions os suggestions I'll be more than happy to hear them.

agramfort, 2020-04-29:
great

ask us if you need help

>

larsoner, 2020-04-29:
@ezemikulan you can look at this PR to get a sense of all of the files you should change to add a new dataset

https://github.com/mne-tools/mne-python/pull/7018/files

ezemikulan, 2020-04-29:
Perfect, thanks

larsoner, 2021-02-19:
@ezemikulan still interested in this? There is now a more recent dataset going through this process so it should outline what changes are needed nowadays: https://github.com/mne-tools/mne-python/pull/8867/files

agramfort, 2021-02-20:
having such a dataset would be great !

how big would it be? any chance it can be made available in BIDS format?

>

ezemikulan, 2021-02-20:
Hi, yes, it's still on my to-do list (that as you can see has some items that are old enough to start going to kindergarten). The data are already in BIDS format, but as derivatives, and is around 12GB. I'll try to work on this in the upcoming weeks, and let you know if I find any issues.

agramfort, 2021-02-21:
can you point us to the data online? is it on openneuro?

ezemikulan, 2021-02-22:
Sure, the data is on g-node [here](https://gin.g-node.org/ezemikulan/Localize-MI) and the paper with the description [here](https://www.nature.com/articles/s41597-020-0467-x).

agramfort, 2021-02-25:
having a tutorial with maybe one subject would be the way to start

I am thinking more and more that we should promote executable scripts
as mne side projects where CircleCI is used to render the tutorials.
Otherwise we'll hit very quickly the cache size limit for one project.

>

ezemikulan, 2021-02-25:
Do you think something like [this](https://github.com/iTCf/mikulan_et_al_2020/blob/master/_00_getting_started.ipynb) would work?

agramfort, 2021-02-28:
yes that's great

I would add a data fetcher to make it run out of the box for people

I created this repo https://github.com/mne-tools/awesome-mne

inspired by common practice to have a curated list of satellite projects

ezemikulan, 2021-03-15:
Hi, I've been been trying to add the data fetcher but unfortunately g-node does not allow to fetch a subset of the data, do you have any suggestions on how should I do it? I think the two options are: 1) putting a subset of the data of one subject on another repository or 2) add a fetcher for the whole dataset from g-node.

larsoner, 2021-03-15:
+1 for adding one subject's data to MNE's OSF.io repo (maybe whichever is the cleanest / highest SNR?), and linking to the complete dataset in the fetcher + tutorial docs

agramfort, 2021-10-30:
I think we'll just have to wait then. thanks for sharing anyway

>

ezemikulan, 2021-10-30:
Ok, I will take care of this once the full dataset is available