mne-tools/mne-python #7710: Set all types to False in pick_types

cbrnr, 2020-05-01:
Would it make sense to set all arguments in `pick_types` to `False` by default? In other words, currently everything is `False` except for `meg`. I know where this is coming from, but for people working with EEG, NIRS, or other non-MEG signals, we always have to include `meg=False` in addition to e.g. `eeg=True`.

agramfort, 2020-05-01:
deprecation cycle will be a huge pain. I feel fully guilty for this but
it's trade off.
If you have only EEG data it's not a pb though no?

>

larsoner, 2020-05-01:
+1 for doing it. I wish we had done it 5 (10?) years ago but it still seems worth it in the long run. I don't think actually it will be too terrible for users (many probably will have set it explicitly anyway), but even forcing them to add `True/False` (when MEG channels are present) will be more explicit in their uses anyway. A potential benefit (aside from consistency) is that it could actually help them uncover some silent bugs in their code.

drammock, 2020-05-01:
So then `raw.pick_types()` with all default arguments would be an error?  Or a raw object with zero channels?  :)

larsoner, 2020-05-01:
Yeah, I suppose so. Kind of a weird side effect. But if you don't ask to pick anything it's probably better that it throws an error than it silently pick one channel type...

drammock, 2020-05-01:
> if you don't ask to pick anything it's probably better that it throws an error than it silently pick one channel type...

Agreed. +1 for embarking on the painful deprecation effort.

hoechenberger, 2020-05-02:
+1 for this from my end, the current behavior always felt weird to me (an EEG user)

hoechenberger, 2020-05-02:
> If you have only EEG data it's not a pb though no?

Valid comment though. But since simultaneous recordings with different methods are gaining in popularity, I believe addressing this issue will become more and more important in the future.

cbrnr, 2020-05-02:
Alright, so it looks like we agree that this must be done. I'll submit a PR.

dengemann, 2020-05-02:
I think this is a good idea. +1 from my side.

agramfort, 2020-05-02:
before fully diving into this let's have a clear plan as it's a lot of work.

In Master:

If you have EEG/MEG:
raw.pick_types(meg=False, eeg=True) -> EEG only
raw.pick_types() -> MEG only

If you have EEG:
raw.pick_types() -> Error?

If you have MEG:
raw.pick_types() -> MEG only

what behavior do you expect in the next release and the next next one?

cbrnr, 2020-05-02:
The next release (0.21) should behave identically to 0.20, but a `DeprecationWarning` should be raised (something like "The defaults of pick_types will change in the next release. All signal types will be set to False by default (currently, meg=True by default).").

Release 0.22 should implement the new behavior. Your examples will be:

If you have EEG/MEG:
```python
raw.pick_types(meg=False, eeg=True)  # EEG only, will work as before
raw.pick_types()  # error (no signals picked)
raw.pick_types(eeg=True)  # new option to pick EEG only
```

If you have EEG:
```python
raw.pick_types()  # error (no signals picked)
raw.pick_types(eeg=True)  # need to select EEG
```

If you have MEG:
```python
raw.pick_types()  # error (no signals picked)
raw.pick_types(meg=True)  # need to select MEG
```

hoechenberger, 2020-05-02:
I've converted @agramfort's list to a table, which we can potentially amend as the discussion progresses.

| ﻿Type      | pick_types() call               | master behavior | proposed new behavior |
|-----------|---------------------------------|-----------------|-----------------------|
| EEG & MEG | pick_types(meg=False, eeg=True) | EEG only        |                       |
| EEG & MEG | pick_types()                    | MEG only        |                       |
| EEG       | pick_types()                    | Error?          |                       |
| MEG       | pick_types()                    | MEG only        |                       |

cbrnr, 2020-05-02:
What is the current behavior when I pick types that don't exist? Does it raise an error? Or pick nothing (i.e. return a zero-channel object)? Or is this a no-op?

hoechenberger, 2020-05-02:
Updated table based on @cbrnr's and @agramfort's comments – hope I copied everything correctly.

| ﻿Type      | pick_types() call               | master behavior | proposed new behavior |
|-----------|---------------------------------|-----------------|-----------------------|
| EEG & MEG | pick_types(meg=False, eeg=True) | EEG only        |                       |
| EEG & MEG | pick_types()                    | MEG only        |                       |
| EEG       | pick_types()                    | Error?          |                       |
| MEG       | pick_types()                    | MEG only        |                       |

agramfort, 2020-05-02:
to be tested. I would really start by writing tests here.

>

PR #7823 (2020-05-29): Deprecate meg=True in pick_types
Fixes #7710 (change the default `meg=True` to `meg=False` in a deprecation cycle.

- [x] Fix/prepare examples/tutorials/etc.
- [x] Change also in `pick_types_forward`?
- [x] ~~Any additional MEG bias we should remove in this PR?~~
- [x] ~~Make arguments kw-only?~~
closed 2020-05-29 by drammock