mne-tools/mne-python #7713: Change description on homepage

cbrnr, 2020-05-03:
Currently, the short description on the [MNE homepage](https://mne.tools/stable/index.html) says:

"Open-source Python software for exploring, visualizing, and analyzing human neurophysiological data: MEG, EEG, sEEG, ECoG, and more."

Maybe I'm too ignorant, but I don't even know what sEEG is (I always thought sEEG and ECoG are synonyms). Furthermore, NIRS support is probably more interesting for more users, so I'd change this sentence to:

"Open-source Python toolbox for exploring, visualizing, and analyzing human neurophysiological data: MEG, EEG, ECoG, NIRS, and more."

I'd also change "software" to "toolbox". WDYT?

agramfort, 2020-05-03:
>
> Maybe I'm too ignorant, but I don't even know what sEEG is (I always
> thought sEEG and ECoG are synonyms).
>

sEEG is stereotactic EEG (implanted electrodes). Sometime it's just called
iEEG

ECoG means a grid

> Furthermore, NIRS support is probably more interesting for more users, so
> I'd change this sentence to:
>

+1 to add NIRS

> "Open-source Python toolbox for exploring, visualizing, and analyzing
> human neurophysiological data: MEG, EEG, ECoG, NIRS, and more."
>
> I'd also change "software" to "toolbox". WDYT?
>
I don't like much the word "toolbox" that reminds me of my matlab time :)

maybe package? But I prefer software

"Open-source Python software for exploring, visualizing, and analyzing
human neurophysiological data: MEG, EEG, sEEG, ECoG, NIRS, and more."

in our ch_type we have sEEG different from ECoG (due to my background on
forward model that is different)

cbrnr, 2020-05-04:
Thanks for the clarification. Software is a very general term and it sounds awkward if you use it with an article, e.g. "MNE is a Python software" doesn't sound right and "MNE is Python software" sounds too generic, at least for me. Therefore, I'd prefer Python package, which is also the correct technical term for what MNE really is.

"Open-source Python package for exploring, visualizing, and analyzing
human neurophysiological data: MEG, EEG, sEEG, ECoG, NIRS, and more."

?

agramfort, 2020-05-04:
deal for me

>

larsoner, 2020-05-04:
LGTM

PR #7724 (2020-05-04): Minor MNE description improvement
Fixes #7713.
closed 2020-05-04 by larsoner